fwupd/fwupd #1328: Document development/debug proccess

FFY00, 2019-08-30:
Hello, would it be possible for someone to add a wiki page describing the development process for fwupd contributors? It might help people who want to start contributing.

hughsie, 2019-08-30:
Sure, that's a great idea. What kind of questions did you need answering? If you do me a quick list I can draft something next week.

FFY00, 2019-08-30:
Mainly a quick way to build, test and debug your patches. I think that's the main thing people will be looking for.

superm1, 2019-08-30:
Like a "Getting involved" page on the Wiki?  I agree it's a great idea.
Here's my ideas:

* I would recommend linking to the existing [compilation page](https://github.com/fwupd/fwupd/wiki/Compilation) in this wiki page 

* Also I would recommend referencing https://fwupd.github.io for documentation.

* Linking to the CI [README.md](https://github.com/fwupd/fwupd/tree/master/contrib/ci)

* Talking about [distro helper scripts](https://github.com/fwupd/fwupd/tree/master/contrib)

* Talking about how the debug flags `--daemon-verbose` and `--verbose` work - how to figure out which domains to use.

* Identifying strings to use to build GUIDs (probably mentioning to use sudo and --verbose to cover including the changes in 1.3.1)

superm1, 2019-09-29:
I started it here, https://github.com/fwupd/fwupd/wiki/Getting-involved feel free to add to it or modify it.

closed 2019-10-18 by superm1