fwupd/fwupd #1329: Rename the `unifying` plugin to `logitech`

FFY00, 2019-08-30:
The DFU protocol implemented by the `unifying` plugin is used by all Logitech products (at least HID products, not sure about cameras and such) that support firmware upgrades. For this reason I think we should rename the plugin to `logitech`, or `logitech-hid` (I don't like this one because it might need another rename in the future).

hughsie, 2019-08-30:
There were two reasons we didn't use `logitech` as a name originally:

* Logitech is a vendor trademark, and it would have been more tricky to get the legal team to approve it (this is also why the 8bitdo plugin is called `ebitdo`...)

* Typically mega-vendors have more than one update protocol, e.g. `synaptics-rmi` and `synapticsmst`

FFY00, 2019-08-30:
I understand, but I think those issues can be overcome.

> Logitech is a vendor trademark, and it would have been more tricky to get the legal team to approve it (this is also why the 8bitdo plugin is called `ebitdo`...)

We could try to contact them. No harm would be done, right?

> Typically mega-vendors have more than one update protocol, e.g. `synaptics-rmi` and `synapticsmst`

That might be the case. But I am sure all HID devices use the implemented protocols, so `logitech-hid` would be suitable.

superm1, 2019-09-23:
Given how the nomenclature has been adopted for some others (synaptics-prometheus, wacom-*, dell-*) I think that `logitech-hid` makes good sense.  It's worth asking them if that's OK in my opinion.

hughsie, 2019-09-23:
I'm not a huge fan of the vendor name there and keep flip flopping between including it for vendor specific protocols and just using the name used in other places. Should `cxaudio` really be `conexant-audio`? Should `synaptics-rmi` be `rmi4`?

superm1, 2019-09-23:
I think when it comes to supporting a vendor's devices and they're not used by any other vendor (ever) it makes fine sense to adopt `vendor-$something` or `vendor` if it's okay with the vendor.

hughsie, 2019-09-23:
I'm hesitant to do `vendor` as large vendors often have more than one update protocol, cough, synaptics, cough.

superm1, 2019-09-23:
Right, I'm saying `vendor-$something` when they have a bunch of protocols, or `vendor` when  they're really just the one.

superm1, 2019-09-23:
And even if they start out as one and adopt another one later, a `git mv vendor vendor-something` and a few sed lines isn't a "super" big deal.

superm1, 2019-11-01:
@FFY00 and I discussed this some more and given non-unifying logitech devices use the same protocol, we probably should revisit.  Maybe once another non-unifying device is added is the right time?

hughsie, 2019-11-01:
I'm okay with `logitech-unifying` fwiw. For 1.3.4 tho..

hughsie, 2019-11-01:
Which non-unifying device is being supported? Would another plugin be appropriate?

superm1, 2019-11-01:
Yes, agree 1.3.4 stuff.  @FFY00 can fill in more details about the other devices.

FFY00, 2019-11-01:
> Which non-unifying device is being supported? Would another plugin be appropriate?

We want to add support for gaming receivers (Lightspeed) and gaming devices. All receivers use the same update protocol (nordic/texas logitech update protocol) and for the devices is also the same (they all speak HID++).

I don't think a new plugin would be appropriate as everything uses the same protocol.

closed 2019-11-04 by superm1

ourichermath, 2022-05-19:
> We want to add support for gaming receivers (Lightspeed) and gaming devices. All receivers use the same update protocol (nordic/texas logitech update protocol) and for the devices is also the same [...]

Long closed (almost 3 yrs later), but why did support for gaming receivers (Lightspeed) not get folded in?

FFY00, 2022-05-19:
It's non-trivial due to the way the driver is currently implemented. I still have this on my TODO list but is not a priority and I have been swamped with other stuff.

FFY00, 2022-05-19:
Unfortunately, I don't remember the details anymore. But when I looked into it, it was non-trivial, otherwise I would have done it. I am not saying is too complicated, just it's not trivial.

commit ec15b5f9 (2019-11-01): Rename unifying plugin to logitech-hid (Fixes: #1329)
commit cfc9383c (2019-11-04): Rename unifying plugin to logitech-hid (Fixes: #1329)
commit bd5fceee (2019-11-04): Rename unifying plugin to logitech-hid (Fixes: #1329)
commit 99452e0b (2019-11-04): Rename unifying plugin to logitech-hidpp (Fixes: #1329)