fwupd/fwupd #1330: readlink of /sys/block/nvme0n1/device/driver failed

nnovikov, 2019-09-01:
I try to update BIOS on my Lenovo X1 Carbon laptop, but ```fwupdmgr install  N23ET65W.cab``` fails with error ```{error #0} linux.c:406 device_get(): readlink of /sys/block/nvme0n1/device/driver failed: No such file or directory	{error #1}```

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

- Operating system and version: Debian GNU/Linux 10 (buster)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get 
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-09-03:
What kernel version are you running?
Can you give the output of `# tree /sys/block/nvme0n1` so we can see what's actually missing?
That failure is coming from right here: https://github.com/rhboot/efivar/blob/master/src/linux.c#L406

rugubara, 2019-09-30:
I also experience this on my Thinkpad P52.
```
PF16W6Y2 /boot/efi/EFI # tree /sys/block/nvme0n1
/sys/block/nvme0n1
├── alignment_offset
├── bdi -> ../../../../../../virtual/bdi/259:0
├── capability
├── dev
├── device -> ../../nvme1
├── discard_alignment
├── eui
├── events
├── events_async
├── events_poll_msecs
├── ext_range
├── hidden
├── holders
├── inflight
├── mq
│   ├── 0
│   │   ├── cpu0
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 1
│   │   ├── cpu1
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 10
│   │   ├── cpu10
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 11
│   │   ├── cpu11
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 2
│   │   ├── cpu2
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 3
│   │   ├── cpu3
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 4
│   │   ├── cpu4
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 5
│   │   ├── cpu5
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 6
│   │   ├── cpu6
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 7
│   │   ├── cpu7
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 8
│   │   ├── cpu8
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   └── 9
│       ├── cpu9
│       ├── cpu_list
│       ├── nr_reserved_tags
│       └── nr_tags
├── nsid
├── nvme0n1p1
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_time
│   │   ├── runtime_status
│   │   └── runtime_suspended_time
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── nvme0n1p2
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_time
│   │   ├── runtime_status
│   │   └── runtime_suspended_time
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── nvme0n1p3
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_time
│   │   ├── runtime_status
│   │   └── runtime_suspended_time
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── power
│   ├── autosuspend_delay_ms
│   ├── control
│   ├── runtime_active_time
│   ├── runtime_status
│   └── runtime_suspended_time
├── queue
│   ├── add_random
│   ├── chunk_sectors
│   ├── dax
│   ├── discard_granularity
│   ├── discard_max_bytes
│   ├── discard_max_hw_bytes
│   ├── discard_zeroes_data
│   ├── fua
│   ├── hw_sector_size
│   ├── io_poll
│   ├── io_poll_delay
│   ├── iostats
│   ├── io_timeout
│   ├── logical_block_size
│   ├── max_discard_segments
│   ├── max_hw_sectors_kb
│   ├── max_integrity_segments
│   ├── max_sectors_kb
│   ├── max_segments
│   ├── max_segment_size
│   ├── minimum_io_size
│   ├── nomerges
│   ├── nr_requests
│   ├── nr_zones
│   ├── optimal_io_size
│   ├── physical_block_size
│   ├── read_ahead_kb
│   ├── rotational
│   ├── rq_affinity
│   ├── scheduler
│   ├── wbt_lat_usec
│   ├── write_cache
│   ├── write_same_max_bytes
│   ├── write_zeroes_max_bytes
│   └── zoned
├── range
├── removable
├── ro
├── size
├── slaves
├── stat
├── subsystem -> ../../../../../../../class/block
├── trace
│   ├── act_mask
│   ├── enable
│   ├── end_lba
│   ├── pid
│   └── start_lba
├── uevent
└── wwid

```

nnovikov, 2019-09-30:
Linux labor 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

```
/sys/block/nvme0n1
├── alignment_offset
├── bdi -> ../../../../../../virtual/bdi/259:0
├── capability
├── dev
├── device -> ../../nvme0
├── discard_alignment
├── eui
├── ext_range
├── hidden
├── holders
├── inflight
├── integrity
│   ├── device_is_integrity_capable
│   ├── format
│   ├── protection_interval_bytes
│   ├── read_verify
│   ├── tag_size
│   └── write_generate
├── mq
│   ├── 0
│   │   ├── cpu0
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 1
│   │   ├── cpu1
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 2
│   │   ├── cpu2
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 3
│   │   ├── cpu3
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 4
│   │   ├── cpu4
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 5
│   │   ├── cpu5
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   ├── 6
│   │   ├── cpu6
│   │   ├── cpu_list
│   │   ├── nr_reserved_tags
│   │   └── nr_tags
│   └── 7
│       ├── cpu7
│       ├── cpu_list
│       ├── nr_reserved_tags
│       └── nr_tags
├── nsid
├── nvme0n1p1
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── async
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_kids
│   │   ├── runtime_active_time
│   │   ├── runtime_enabled
│   │   ├── runtime_status
│   │   ├── runtime_suspended_time
│   │   └── runtime_usage
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── nvme0n1p2
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── async
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_kids
│   │   ├── runtime_active_time
│   │   ├── runtime_enabled
│   │   ├── runtime_status
│   │   ├── runtime_suspended_time
│   │   └── runtime_usage
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── nvme0n1p3
│   ├── alignment_offset
│   ├── dev
│   ├── discard_alignment
│   ├── holders
│   │   └── dm-0 -> ../../../../../../../../virtual/block/dm-0
│   ├── inflight
│   ├── partition
│   ├── power
│   │   ├── async
│   │   ├── autosuspend_delay_ms
│   │   ├── control
│   │   ├── runtime_active_kids
│   │   ├── runtime_active_time
│   │   ├── runtime_enabled
│   │   ├── runtime_status
│   │   ├── runtime_suspended_time
│   │   └── runtime_usage
│   ├── ro
│   ├── size
│   ├── start
│   ├── stat
│   ├── subsystem -> ../../../../../../../../class/block
│   ├── trace
│   │   ├── act_mask
│   │   ├── enable
│   │   ├── end_lba
│   │   ├── pid
│   │   └── start_lba
│   └── uevent
├── power
│   ├── async
│   ├── autosuspend_delay_ms
│   ├── control
│   ├── runtime_active_kids
│   ├── runtime_active_time
│   ├── runtime_enabled
│   ├── runtime_status
│   ├── runtime_suspended_time
│   └── runtime_usage
├── queue
│   ├── add_random
│   ├── chunk_sectors
│   ├── dax
│   ├── discard_granularity
│   ├── discard_max_bytes
│   ├── discard_max_hw_bytes
│   ├── discard_zeroes_data
│   ├── fua
│   ├── hw_sector_size
│   ├── io_poll
│   ├── io_poll_delay
│   ├── iostats
│   ├── logical_block_size
│   ├── max_discard_segments
│   ├── max_hw_sectors_kb
│   ├── max_integrity_segments
│   ├── max_sectors_kb
│   ├── max_segments
│   ├── max_segment_size
│   ├── minimum_io_size
│   ├── nomerges
│   ├── nr_requests
│   ├── optimal_io_size
│   ├── physical_block_size
│   ├── read_ahead_kb
│   ├── rotational
│   ├── rq_affinity
│   ├── scheduler
│   ├── wbt_lat_usec
│   ├── write_cache
│   ├── write_same_max_bytes
│   ├── write_zeroes_max_bytes
│   └── zoned
├── range
├── removable
├── ro
├── size
├── slaves
├── stat
├── subsystem -> ../../../../../../../class/block
├── trace
│   ├── act_mask
│   ├── enable
│   ├── end_lba
│   ├── pid
│   └── start_lba
├── uevent
└── wwid
```

superm1, 2019-10-01:
I've reported it upstream to efivar in https://github.com/rhboot/efivar/issues/138.  
It looks like `tree` didn't follow the symlinks.  Can you please check for specificially `/sys/block/nvme0n1/device/device/driver` and `/sys/block/nvme0n1/device/driver`

On my local 5.3 test machine it looks that I have `/sys/block/nvme0n1/device/device/driver` which I've not seen problems with efivar 37 myself.

superm1, 2019-10-01:
that's different than the first ones - you are seeing `Error writing to file descriptor: No space left on device` at the end.  Other people on this thread, did you see multiple errors including that last one?

rugubara, 2019-10-04:
I do have "no space left on device" error too.

superm1, 2019-10-04:
I would find it highly unlikely that the storage fill up for the backing of efivarfs.

Those of you affected - do you have some sort of locking set in you firmware setup?  Boot variable locking, or NVRAM locking, something like that?

rugubara, 2019-10-05:
I have pstore in efivars configured and dmesg from kernel oopses fill it
quite regularly.

I cleared the pstore and immediately I have succeeded with update.

I wish the error message suggested that efivars was out of space.

superm1, 2019-10-06:
Well glad to hear that's what the problem is.
I've made a PR that will make this error message clearer in the case of efivarfs being full.

rugubara, 2019-10-06:
Can you reference a PR here so I could patch and test too?

superm1, 2019-10-06:
https://github.com/fwupd/fwupd/pull/1435

superm1, 2019-10-16:
Since #1435 is merged, i'm going to close this. If you can reproduce with master please re-open.

closed 2019-10-16 by superm1

Knusper, 2021-08-08:
> I have pstore in efivars configured and dmesg from kernel oopses fill it quite regularly. I cleared the pstore and immediately I have succeeded with update. I wish the error message suggested that efivars was out of space. 

I appologise for making noise in a closed bug report.  But can you detail a bit how this is achieved (clearing pstore)?

rugubara, 2021-08-09:
@Knusper, once you have your pstore mounted, it is enough to rm files from there.