fwupd/fwupd #1336: Add a "Generic" TPM2 plugin

superm1, 2019-09-05:
Just as a follow up to #1331 so this information isn't lost.

> There is a documented upgrade process, see the TPM 2.0 specification, Part 3, Section 27. I don't have any experience with these commands, but it seems you need to supply a signature of the new firmware to TPM2_FieldUpgradeStart, after that is validated successfully you can upload the firmware with TPM2_FieldUpgradeData.

https://trustedcomputinggroup.org/wp-content/uploads/TPM-Rev-2.0-Part-3-Commands-01.38.pdf#page=321

For the basic enumeration stuff:
* Ideally GUID strings would be built like this:
`TPM-$MODE-$MANUFACTURER-$VENDORSTR`

* Version strings can be read from fixed properties and are 4 parts.

* Updatability should be determined by TPM Ownership

superm1, 2019-09-05:
{CC @diabonas}

I had some internal discussion and it's not likely that any TPM vendor will support using a generic fwupd plugin due to certification requirements.

At least based on that, I would say a generic TPM2 plugin would likely be "information only".

hughsie, 2019-09-06:
> I would say a generic TPM2 plugin would likely be "information only".

I still think that would be a useful thing to have...

superm1, 2019-09-06:
OK, i'll re-open.

I think if we land something like this it will need to have something similar to how synaptics mst and dell-dock work where they build the device ID's the same but dell-dock is preferred.

hughsie, 2019-12-05:
Fixed with addc0211352c2193d5d7b3fbb224bf978cbecba7

closed 2019-12-05 by hughsie