fwupd/fwupd #1342: Regression in uefi plugin for MinnowBoardMAX hardware

hughsie, 2019-09-10:
A MinnowBoardMAX running MNW2MAX1.X64.0101.R01.1908091048.bin is failing to `fu_plugin_startup()` in the uefi plugin as this test is failing:

    	if (!(data[0x13] & (1 << 3))) {
    		g_set_error_literal (error,
    				     FWUPD_ERROR,
    				     FWUPD_ERROR_NOT_SUPPORTED,
    				     "System does not support UEFI mode");
    		return FALSE;
    	}

It clearly supports UEFI (and even supplies capsules for updates) so perhaps this test is too strict for all hardware perhaps?

superm1, 2019-09-10:
Can you please provide a dump of `dmidecode` output for one of these systems?

hughsie, 2019-09-10:
Sure thing, attached:

[dmidecode.txt](https://github.com/fwupd/fwupd/files/3596418/dmidecode.txt)

superm1, 2019-09-10:
Hmm, definitely a high enough smbios version.  This looks like a FW bug to me.

superm1, 2019-09-10:
At least until that FW bug should be fixed, I would think this should just be quirked.

hughsie, 2019-09-10:
I'm confused what the `0x13` offset is actually doing. Do you have some docs handy?

hughsie, 2019-09-10:
Public docs FTW. It's `UEFI Specification is supported.` so you're exactly right.

closed 2019-09-10 by hughsie