fwupd/fwupd #1346: Missing header lines (msgfmt gives warnings)

scootergrisen, 2019-09-11:
If i run msgfmt on da.po i get these warnings:

    fwupd.master.da.po:9: warning: header field 'PO-Revision-Date' missing in header
    fwupd.master.da.po:9: warning: header field 'Last-Translator' missing in header

Seems like the da.po file is missing there lines in the header for consistency:

    "POT-Creation-Date: 0000-00-00 00:00+0000\n"
    "PO-Revision-Date: YEAR-MO-DA HO:MI+ZONE\n"
    "Last-Translator: FULL NAME <EMAIL@ADDRESS>\n"

superm1, 2019-09-11:
I recall this is actually intended behavior:
https://github.com/fwupd/fwupd/commit/7ccba7afba9fbd35179f4975f7fa20e0eb319ea3

scootergrisen, 2019-09-11:
Are they removed because they change in each release?
Why would that be good to remove?

I like to be able to run msgfmt on all po files without getting warnings.
The dates can help translators see how old the translation is and when it was last updated and who did it.

hughsie, 2019-09-11:
>  The dates can help translators see how old the translation is and when it was last updated and who did it.

That breaks when the translations are managed by transifex and the .po files are autogenerated server side. Every time I synced about ~20 files would change for no reason.

scootergrisen, 2019-09-11:
Why is it a problem those lines changes each time you sync ?
Are they not just changing to show the latest dato and user that made the last change etc.?

hughsie, 2019-09-11:
> Why is it a problem those lines changes each time you sync ?

You get lots of noise in the git history:

    --- a/po/ast.po
    +++ b/po/ast.po
    @@ -8,8 +8,8 @@ msgid ""
     msgstr ""
     "Project-Id-Version: fwupd\n"
     "Report-Msgid-Bugs-To: \n"
    -"POT-Creation-Date: 2018-03-12 10:37+0000\n"
    -"PO-Revision-Date: 2018-03-12 10:37+0000\n"
    +"POT-Creation-Date: 2018-06-06 14:54+0100\n"
    +"PO-Revision-Date: 2018-06-06 14:00+0000\n"
     "Last-Translator: Richard Hughes <richard@hughsie.com>\n"
     "Language-Team: Asturian (http://www.transifex.com/freedesktop/fwupd/language/ast/)\n"
     "MIME-Version: 1.0\n"

scootergrisen, 2019-09-11:
I guess it might be a little annoying but seems it would be more consistent to keep the lines.
Dont know how other projects do it.

Maybe they only download the po files from Transifex before before release or something like that.
Not that i would like that option because then i have to wait a long time to see the changes.

Or maybe you can ignore changes in po/*.po file some how so you dont see them but i dont know about that.

I just like to be able to test all translation files in like GNOME at once with software like msgfmt and hopefully see no errors/warnings.

hughsie, 2019-09-11:
Sorry, we created the script for a reason. fwupd isn't a GNOME project and doesn't even use the GNOME translation teams. Maybe you can filter out these msgfmt validation problems from your test script? Thanks.

closed 2019-09-11 by hughsie