fwupd/fwupd #1347: have a batch mode for get-devices - not batch mode if there exist upgrade history.

ycheng, 2019-09-12:
For the machine I use, I have upgrade history. For tooling requirement, I want to run

"fwupdmgr get-devices --show-all-devices"

in batch mode. However it asks me whether to upload history, and that makes batch mode impossible. I believe a "-n" option will be sufficient for this use case.

  -----

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Please answer the following questions:

- Operating system and version: ubuntu 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): as ubunt install.

hughsie, 2019-09-12:
We have `--no-unreported-check` which you can use to suppress the report checking. That should have existed since 1.0.4 -- does that work?

superm1, 2019-09-12:
Add --no-unreported-check to your command line. You probably also want --no-metadata-check.