fwupd/fwupd #1351: Dell TB16: Not all components can be updated

paulmenzel, 2019-09-13:
The [ArchWiki page](https://wiki.archlinux.org/index.php/Dell_TB16) claims, not all components of the docking station Dell TB16 can be updated, and a Microsoft Windows system is needed.

> The firmware updates provided for Windows at Dell's TB16 Support: Drivers page include
> firmware that cannot be updated in Linux, such as the ASMedia USB Host Controller firmware
> which fixes numerous instability USB issues and Synaptics controller (DisplayPort).

hughsie, 2019-09-13:
@superm1 is that right? If so, we should get ASMedia onboard.

superm1, 2019-09-13:
Yes asmedia can not be updated from Linux, but my understanding is that updates are not relevant for Linux because of the kernel code level changes. Have you observed an actual difference in USB performance between two docks that one is updated asmedia and one isn't?

Asmedia isn't actually connected to synaptics (different buses) so I don't see how there can be a connection there. Synaptics improvements would come specifically from MST firmware.

paulmenzel, 2019-09-14:
Using Linux 5.2.9 from Debian Sid/unstable, USB keyboard and mouse do not work, when Thunderbolt security is not set to *no security* in the Dell XPS 13 9380 laptop firmware. The ArchWiki article suggests to update the Dell TB16 firmware.

superm1, 2019-09-14:
Can you please share kernel messages from boot all the way to when you first plug it in?

paulmenzel, 2019-10-23:
> Can you please share kernel messages from boot all the way to when you first plug it in?

The user reports more problems with dock. As it’s off-topic here, what mailing list or bug tracker should be used?

superm1, 2019-10-23:
If it's problems with usptream Linux, you can file kernel bugs for relevant components.  If it's problems with a Dell pre-loaded Linux image, then you should contact Dell support.

superm1, 2019-10-30:
It's worth mentioning there is this out there today: https://github.com/smx-smx/ASMTool.  It uses low level IO stuff directly to addresses though.  Probably wouldn't work too well in a locked down system?

> @superm1 is that right? If so, we should get ASMedia onboard.

I don't disagree.

hughsie, 2019-10-30:
I don't think that port IO is going to work when the kernel lockdown stuff hits distros.

superm1, 2019-10-30:
> I don't think that port IO is going to work when the kernel lockdown stuff hits distros.

superio 😭 

I wonder if something like `libpci` and just working directly with PCIe configuration space is a workable solution then for ASMedia.  For superio I guess the only option is to move it into the EFI app really.

paulmenzel, 2020-01-24:
After not only getting a device with Microsoft Windows installed, but a *Dell* device with Microsoft Windows to make the [Dell TB16 update program](https://www.dell.com/support/home/de/de/debsdt1/drivers/driversdetails?driverid=7tj1h&oscode=w732&productcode=dell-thunderbolt-dock-tb16) work and not abort reading the *laptop* firmware, the firmware of the TB16 components were updated.

```
MST1            3.10.002    3.12.002
MST2            3.10.002    3.12.002
ASM USB 3.0 Cntlr    10.11.23    10.11.A9
Dock NVM        00.00.16    00.00.27 
```

The user reports, that the USB input device issues are fixed now.

So, it is really important, that fwupd supports updating all the Dell TB16 firmware, or at least notifies about these components. The Linux kernel should probably also log something. This would save users and the Dell support a lot of time.

superm1, 2020-01-24:
Well thanks for checking that.

paulmenzel, 2020-02-07:
The (German) Dell technical support refuses to take on the issue. They claim it works with Ubuntu 16.04 (which it does not), and that Ubuntu is (in their language) freeware, and so they do not give support, and refuse to escalate this. (They cannot fix the Linux drivers (despite me saying it’s likely a firmware issue).)

hughsie, 2020-08-19:
I think we should close this; there's nothing fwupd can do, and I know the right people are already aware of this issue. Sorry that's not more helpful.

closed 2020-08-19 by hughsie