fwupd/fwupd #1353: Crashing after 1.3.1 update

FFY00, 2019-09-13:
fwupd is failing to start after the update to 1.3.1.

```
Sep 13 16:32:09 saetre systemd[1]: Starting Firmware update daemon...
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 Fwupd                looking for os-release at /etc/os-release
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 FuConfig             loading /usr/share/fwupd/metainfo
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 XbSilo               file: 451419ad-0e77-5491-b388-e2a624cad17e, current:451419ad-0e77-5491-b388-e2a624cad17e, cached: (null)
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 XbSilo               loading silo with file contents
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 FuConfig             using config path of /etc/fwupd
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0021 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd-tests.conf
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             using config path of /usr/lib/fwupd
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             ordering lvfs-testing=lvfs+1
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0022 FuConfig             ignoring unfound remote fwupd
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuEngine             client certificate exists and working
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHistory            got schema version of 5
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuIdle               setting timeout to 7200s
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property Manufacturer=LENOVO
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property EnclosureKind=a
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property Family=ThinkPad P52s
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property ProductName=20LBCTO1WW
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property ProductSku=LENOVO_MT_20LB_BU_Think_FM_ThinkPad P52s
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BiosVendor=LENOVO
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BiosVersion=N27ET32W (1.18 )
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BiosMajorRelease=01
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BiosMinorRelease=12
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BaseboardManufacturer=LENOVO
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuHwids              smbios property BaseboardProduct=20LBCTO1WW
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0028 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ata.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/colorhug.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell-dock.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0029 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dfu.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ebitdo.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/fastboot.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nitrokey.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nvme.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hid.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hub.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/solokey.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/steelseries.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/superio.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synaptics-prometheus.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synapticsmst.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi-recovery.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/unifying.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0030 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-raw.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-usb.quirk
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuQuirks             now 189 quirk entries
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuEngine             remote lvfs-testing not enabled, so skipping
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuEngine             remote fwupd-tests not enabled, so skipping
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuEngine             remote vendor not enabled, so skipping
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 FuEngine             remote vendor-directory not enabled, so skipping
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               0.02ms      parse blob
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               0.04ms      loaded file
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               file: 008006cd-b548-53e4-9eb7-f3d87e5ca547, current:008006cd-b548-53e4-9eb7-f3d87e5ca547, cached: (null)
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               loading silo with file contents
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0031 XbSilo               0.00ms      parse blob
[... 100 lines cut ...]
Sep 13 16:32:10 saetre fwupd[2685213]: 15:32:10:0110 FuPlugin             performing fu_plugin_device_registered() on thunderbolt_power
[... cut ...]
Sep 13 16:32:10 saetre systemd[1]: fwupd.service: Main process exited, code=dumped, status=5/TRAP
Sep 13 16:32:10 saetre systemd[1]: fwupd.service: Failed with result 'core-dump'.
Sep 13 16:32:10 saetre systemd[1]: Failed to start Firmware update daemon.
```

Please answer the following questions:

- Operating system and version: Archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): I updated the system package to 1.3.1
- Have you tried rebooting?  Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-09-13:
Can you get a backtrace please?

FFY00, 2019-09-13:
```
#0  0x00007ffff7a14176 in  () at /usr/lib/libglib-2.0.so.0
#1  0x00007ffff7a143c8 in g_logv () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff7a145b0 in g_log () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff0acf27a in Tss2_Tcti_Tabrmd_Init () at /usr/lib/libtss2-tcti-tabrmd.so
#4  0x00007ffff1aaecbf in  () at /usr/lib/libtss2-esys.so.0
#5  0x00007ffff1aaf14a in get_tcti_default () at /usr/lib/libtss2-esys.so.0
#6  0x00007ffff1aa5cd2 in Esys_Initialize () at /usr/lib/libtss2-esys.so.0
#7  0x00007ffff1ba1c4d in fu_uefi_pcrs_setup_tpm20 (self=0x55555571fa00, error=0x7fffffffe5c0) at ../plugins/uefi/fu-uefi-pcrs.c:118
#8  0x00007ffff1ba211d in fu_uefi_pcrs_setup (self=0x55555571fa00, error=0x7fffffffe5c0) at ../plugins/uefi/fu-uefi-pcrs.c:197
#9  0x00007ffff1ba06c4 in fu_uefi_device_add_system_checksum (device=0x555555715230, error=0x7fffffffe638) at ../plugins/uefi/fu-uefi-device.c:475
#10 0x00007ffff1ba0a02 in fu_uefi_device_probe (device=0x555555715230, error=0x7fffffffe7c0) at ../plugins/uefi/fu-uefi-device.c:554
#11 0x0000555555572ce4 in fu_device_probe ()
#12 0x00007ffff1b9ab3f in fu_plugin_uefi_coldplug_device (plugin=0x5555556e7210, dev=0x555555715230, error=0x7fffffffe7c0) at ../plugins/uefi/fu-plugin-uefi.c:526
#13 0x00007ffff1b9bc78 in fu_plugin_coldplug (plugin=0x5555556e7210, error=0x7fffffffe7c0) at ../plugins/uefi/fu-plugin-uefi.c:887
#14 0x000055555557c7bb in fu_plugin_runner_coldplug ()
#15 0x00005555555897bd in  ()
#16 0x00005555555944e2 in fu_engine_load ()
#17 0x00005555555672d5 in main ()
```

FFY00, 2019-09-13:
Hum, looks like our tpm2-tss is a bit outdated (2.2.1). That's probably the issue.

hughsie, 2019-09-13:
If that's the case, can you find the lowest version that doesn't crash (!) and we can depend on that in meson.build -- thanks!

FFY00, 2019-09-13:
Sure, I am already looking into it :)

FFY00, 2019-09-13:
Updating to the latest version (2.3.1) gives me this.

```
ERROR:tcti:src/tss2-tcti/tcti-device.c:439:Tss2_Tcti_Device_Init() Failed to open device file /dev/tpm0: Device or resource busy
WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7ffff7f68ec0 failed with a000a
WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-device
ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-default.so
```

```
#0  0x00007ffff7a14176 in  () at /usr/lib/libglib-2.0.so.0
#1  0x00007ffff7a143c8 in g_logv () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff7a145b0 in g_log () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff0b6327a in Tss2_Tcti_Tabrmd_Init () at /usr/lib/libtss2-tcti-tabrmd.so.0
#4  0x00007ffff1aab153 in  () at /usr/lib/libtss2-esys.so.0
#5  0x00007ffff1aab41a in  () at /usr/lib/libtss2-esys.so.0
#6  0x00007ffff1aac7b9 in  () at /usr/lib/libtss2-esys.so.0
#7  0x00007ffff1aac98e in  () at /usr/lib/libtss2-esys.so.0
#8  0x00007ffff1aac19f in  () at /usr/lib/libtss2-esys.so.0
#9  0x00007ffff1aac3ad in  () at /usr/lib/libtss2-esys.so.0
#10 0x00007ffff1a9d4ad in Esys_Initialize () at /usr/lib/libtss2-esys.so.0
#11 0x00007ffff2c5ac4d in fu_uefi_pcrs_setup_tpm20 (self=0x555555dd64c0, error=0x7fffffffe5c0) at ../plugins/uefi/fu-uefi-pcrs.c:118
#12 0x00007ffff2c5b11d in fu_uefi_pcrs_setup (self=0x555555dd64c0, error=0x7fffffffe5c0) at ../plugins/uefi/fu-uefi-pcrs.c:197
#13 0x00007ffff2c596c4 in fu_uefi_device_add_system_checksum (device=0x555555e60230, error=0x7fffffffe638) at ../plugins/uefi/fu-uefi-device.c:475
#14 0x00007ffff2c59a02 in fu_uefi_device_probe (device=0x555555e60230, error=0x7fffffffe7c0) at ../plugins/uefi/fu-uefi-device.c:554
#15 0x0000555555572ce4 in fu_device_probe ()
#16 0x00007ffff2c53b3f in fu_plugin_uefi_coldplug_device (plugin=0x5555558c6a40, dev=0x555555e60230, error=0x7fffffffe7c0) at ../plugins/uefi/fu-plugin-uefi.c:526
#17 0x00007ffff2c54c78 in fu_plugin_coldplug (plugin=0x5555558c6a40, error=0x7fffffffe7c0) at ../plugins/uefi/fu-plugin-uefi.c:887
#18 0x000055555557c7bb in fu_plugin_runner_coldplug ()
#19 0x00005555555897bd in  ()
#20 0x00005555555944e2 in fu_engine_load ()
#21 0x00005555555672d5 in main ()
```

Do you have any idea what's happening?

I can confirm that removing tpm2-tss fixed my issue.

hughsie, 2019-09-13:
@diabonas any ideas?

FFY00, 2019-09-13:
Restarting `tpm2-abrmd.service` fixed `/dev/tpm0: Device or resource busy` and the daemon started fine. I think we should investigate more so that we can understand what happened.

diabonas, 2019-09-13:
I'm pretty sure that is indeed a problem with tpm2-abrmd - maybe the service failed to start correctly, blocking the access to `/dev/tpm0`, but not communicating via D-Bus? If you're able to reproduce this somehow, it's definitely worth filing a bug report [upstream](https://github.com/tpm2-software/tpm2-abrmd).

Regarding the crash with tpm2-tss 2.2.1, I don't understand how that happens because tpm2-abrmd 2.2.0 is the latest version, but shouldn't explicitly depend on tpm2-tss 2.3. Again, this would be an upstream bug regarding the interaction of tpm2-tss and tpm2-abrmd, fwupd itself should only require tpm2-tss >=2.0, which we [already check](https://github.com/fwupd/fwupd/blob/ff1d7efb9ed0382f4a3f263d45adbae1a5f5d630/meson.build#L237).

To verify my analysis, could you uninstall tpm2-abrmd and run fwupd with tpm2-tss 2.2.1? Hopefully this should work without issues.

diabonas, 2019-09-16:
The problem seems to be that GLib aborts even if tpm2-abrmd only uses `g_debug`, which is obviously not desirable. @hughsie, can you shed some light on this? Probably fwupd does something with `g_log_set_fatal_mask` or `g_log_set_handler` that is incompatible with tpm2-abrmd, it doesn't appear to be a problem in tpm2-abrmd per se. If I do something like `g_log_set_fatal_mask (NULL, G_LOG_LEVEL_ERROR | G_LOG_LEVEL_CRITICAL)` in tpm2-abrmd, everything works as expected.

hughsie, 2019-09-16:
Good hunting! Commit 38027e696 added a call to `g_log_set_fatal_mask (NULL, G_LOG_LEVEL_MASK);` which looks wrong indeed. I'll do a PR to strip this out, as it's clearly wrong.

closed 2019-09-16 by hughsie