fwupd/fwupd #1357: Updating Thunderbolt firmware on X280 required too much --force

hifi, 2019-09-16:
Acting upon the Lenovo Thunderbolt bricking issue (https://pcsupport.lenovo.com/fi/en/solutions/ht508988) I upgraded fwupd from snap to be able to flash the new firmware with fwupd as the one with Ubuntu 18.04 was way too old for that.

Getting the updates worked fine, my listed image version of the firmware was stock 12 as expected and the target was 20 as expected. The downloaded cab file was this: https://fwupd.org/downloads/c44e55a529826b54dea6372a65fd406f29a5c8a1-Lenovo-ThinkPad-X280-Thunderbolt-Firmware-N20TF19W-Secured.cab

I checked and checked it textually matched the update package on the Lenovo site and proceeded to upgrade but fwupd said it wasn't compatible with my device. Unfortunately I don't have a cap from the shell anymore so I don't remember the exact message but it wasn't able to verify it was the correct one.

Anyway, against any best practice I proceeded with `--force` which started the flashing process. It took a good while and it was likely expected that I would lose my external USB keyboard and mouse connected to my dock while doing that but it did worry and surprise me. It ended up with an error message of the device not coming back up after the upgrade and I was fairly sure I just bricked my TB controller. :cold_sweat: 

After rebooting the system using internal input devices the firmware version was properly bumped to 20 and I believe everything is fine as things seem work. :tada: 

One thing I did notice that the warning about NOT forcing the firmware in I found later on the web page: https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN20TF.firmware

> DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.

And the suggestion of doing some BIOS magic:

> Before update, set the Thunderbolt BIOS Assist Mode to "Disable" in BIOS setup under Config > Thunderbolt 3.

I did not see those lines in any obvious way presented to me when running fwupdmgr *before* I proceeded to `--force`. They may have been visible somewhere but at least they weren't in the change notes as they were abruptly cut short or completely missing in the update list so I wasn't really aware of the extra danger in the operation.

So I guess I'm a happy customer but maybe a few things could be improved:
- if there's a special warning on fwupd.org or instructions, those probably should be clearly visible and require manual approval by the user before proceeding with any upgrade in a way it's impossible to miss them
- X280 Thunderbolt upgrade compatibility check seems either to be broken or missing
- X280 Thunderbolt upgrade kills TB until reboot, hardware issue? if so, a warning would be nice
- in general if a firmware upgrade happens "live" a warning about possible side effects would be nice

Disclaimer: I wasn't very careful and I wouldn't hold fwupd responsible for bricking my device with `--force` so it's all good but could be better :+1: 

```shell
$ fwupdmgr --version
client version: 1.3.1
checkout info:  1.3.1
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.1
```

```shell
$ fwupdmgr get-devices
20KF001GMX
?
??ThinkPad X280 Thunderbolt Controller:
?     Device ID:           ae22f9b9e65093d39c6fc66fcf01268c8be2894a
?     Summary:             Unmatched performance for high-speed I/O
?     Current version:     20.00
?     Vendor:              Lenovo (TBT:0x0109)
?     Flags:               internal|updatable|require-ac|supported|registered
?     GUID:                4808eca4-fd4a-50e6-9e8d-bfd813f063da ? TBT-01091704-native
?   
??SAMSUNG MZVLB512HAJQ-000L7:
?     Device ID:           e11623b2caa18fee292058a5c09ca4e6152f7ecf
?     Summary:             NVM Express Solid State Drive
?     Current version:     3L2QEXA7
?     Vendor:              NVME:0x144D
?     Serial Number:       [scrubbed]
?     Flags:               internal|updatable|require-ac|registered|needs-reboot
?     GUIDs:               47335265-a509-51f7-841e-1c94911af66b ? NVME\VEN_144D&DEV_A808
?                          79d6cfae-a5a2-5936-9248-5aebd23480f7 ? SAMSUNG MZVLB512HAJQ-000L7
?   
??System Firmware:
?     Device ID:           5a566863d357fb728a620cdf235632fb9bc99f5f
?     Current version:     0.1.30
?     Minimum Version:     0.0.1
?     Vendor:              LENOVO
?     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
?     GUID:                508f7539-1ad6-48b9-8680-38377535009d
?   
??UEFI Device Firmware:
?     Device ID:           093ef0be8328a2c4ed2fe55cd36aae3171b92ade
?     Current version:     184.65.3590
?     Minimum Version:     0.0.1
?     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
?     GUID:                6d28cd9f-7bcd-4fb9-9f10-0372e2962fc4
?   
??UEFI Device Firmware:
?     Device ID:           ca368aebcf7da847029e9f2520ec55fb7a036b31
?     Current version:     0.1.4
?     Minimum Version:     0.1.4
?     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
?     GUID:                3f4a527b-6588-45b8-b2d3-dc61189b63cb
?   
??VMM5322:
      Device ID:           585c346c009da3cc19a400fb59a4ed8d0cbd0151
      Summary:             Multi-Stream Transport Device
      Current version:     5.02.011
      Vendor:              Synaptics
      Flags:               updatable|registered
      GUIDs:               f15aa55c-9cd5-5942-85ae-a6bf8740b96c ? MST-panamera
                           c083122a-aae0-5ce2-9f80-c65a1aa1151b ? MST-panamera-vmm5322-593
                           d65b3310-c47a-5e0c-bee0-df7f130126ec ? MST-panamera-593

```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes (it was required)
- Are you using an NVMe disk? Yes

superm1, 2019-09-19:
> DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.

This seems precarious to me.  @hughsie can you give more context?  I don't think anyone should be force updating controllers - if that's needed probably a quirk is needed, why is this even here?

> Before update, set the Thunderbolt BIOS Assist Mode to "Disable" in BIOS setup under Config > Thunderbolt 3.

@hughsie Is it not possible to automatically do this in Linux with any lenovo manageability interfaces?  It seems like it's something that should be possible to put together a lenovo plugin that could do this before certain updates if it's needed.

> if there's a special warning on fwupd.org or instructions, those probably should be clearly visible and require manual approval by the user before proceeding with any upgrade in a way it's impossible to miss them

In general I think this is a pretty good idea, and in many instances can tell people things like "Don't unplug the power before updating" or "don't unplug the Thunderbolt device while updating".

superm1, 2019-09-19:
I do have an educated guess at what's going on here; it's related to native mode versus non native mode for the firmware.  If you flash firmware that defaults to native mode while TBT controller and platform firmware is in BIOS assist mode it's undefined behavior.  The TBT controller will probably need a cold boot to come back to life.

The GUID that fwupd creates for native mode has a `-native` appended to the instance ID and hence expects the firmware to be flashed to it to default to native mode as well.

If that's a correct hypothesis, then this is absolutely right, you shouldn't be flashing the TBT firmware onto this device until you've switched BIOS modes.

hughsie, 2019-09-19:
> tell people things like "Don't unplug the power before updating" 

We have about a dozen different ways vendors are phrasing this now on the lvfs. If you file an issue there we can work out a set of phrases that can be translated client side and not just pasted into every update description.

superm1, 2019-09-19:
I'm not convinced it should go into the update description.  I was think it should actually be a client message that is created based upon the flags of the device, like a prompt that you acknowledge after you call `# fwupdgmr update` (but of course that can be scripted around).

Example:
* If the device doesn't have `internal`: "The update will take approximately %u seconds to complete, please don't unplug %s until it is complete or the device may be damaged."
* If the device has `require-ac` but not `reboot-required`: "The update will take approximately %u seconds to complete, please be sure that your %s is plugged into a power source for the duration of the update."

closed 2019-10-26 by stale[bot]