fwupd/fwupd #1362: Multiple errors about 'TCTI'

LinuxOnTheDesktop, 2019-09-16:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.3.1
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             eded407096031a81370bc25ae01650697a7c72b6
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 thunderbolt
  Created:              2019-09-16

LENSE20512GMSP34MEAT2TA
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
  Guid:                 ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  VendorId:             NVME:0x17AA
  Version:              2.8.8341
  Icon:                 drive-harddisk
  Created:              2019-09-16

System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               LENOVO
  Version:              0.1.40
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-09-16

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-09-16

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
[... 807 lines cut ...]
Sep 16 22:36:47 x1 systemd[1]: Mounted Mount unit for fwupd, revision 1023.
Sep 16 22:36:47 x1 systemd[1]: Started Activate fwupd updates.
Sep 16 22:36:48 x1 systemd[1]: Started Service for snap application fwupd.fwupd.
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tcti-device.c:440:Tss2_Tcti_Device_Init() Failed to open device file /dev/tpm0: No such file or directory
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7f618a42bcd0 failed with a000a
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-device
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-default.so
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tcti-device.c:440:Tss2_Tcti_Device_Init() Failed to open device file /dev/tpmrm0: No such file or directory
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7f618a42bcd0 failed with a000a
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-device
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-device.so.0
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tcti-device.c:440:Tss2_Tcti_Device_Init() Failed to open device file /dev/tpm0: No such file or directory
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7f618a42bcd0 failed with a000a
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-device
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-device.so.0
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/util/io.c:252:socket_connect() Failed to connect to host 127.0.0.1, port 2321: errno 111: Connection refused
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7f618a42b230 failed with a000a
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-socket
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-mssim.so.0
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr-dl.c:248:tctildr_get_default() No standard TCTI could be loaded
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:tcti:src/tss2-tcti/tctildr.c:418:Tss2_TctiLdr_Initialize_Ex() Failed to instantiate TCTI
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Sep 16 22:36:54 x1 fwupd.fwupd[1675]: 21:36:54:0712 FuPluginUefi         Failed to get PCR0s: failed to initialize TPM library
Sep 16 22:37:06 x1 snapd[1223]: storehelpers.go:436: cannot refresh: snap has no updates available: "core", "fwupd"
```

superm1, 2019-09-17:
Do you not physically have a TPM? Or at least a firmware TPM?

LinuxOnTheDesktop, 2019-09-17:
I think I do physically have a TPM. For, there's an option to enable or disable it in the BIOS. My current setting there is: disabled.

I discover also that at present when I run `fwupdmgr get-devices' every line of output starts with a question mark.

superm1, 2019-09-17:
If you turn it on do the tcti errors go away?

The question marks seem to be a problem with how the snap handles Unicode.

superm1, 2019-09-17:
CC @diabonas for additional comments.

So previously before fwupd the TPM was completely optional, and you just wouldn't have checksums if it wasn't found.  With fwupd 1.3.1 this changed and if the TPM library fails to initialize then the UEFI plugin will fail to initialize too (`FWUPD_ERROR_NOT_FOUND` instead of `G_IO_ERROR_NOT_SUPPORTED`).

Regarding the question marks fwupd 1.3.1 uses unicode more heavily, so it's likely that they were present in commands from earlier fwupd too, but not as apparent.  I believe it's this bug:
https://bugs.launchpad.net/ubuntu/+source/snapd/+bug/1576411

diabonas, 2019-09-17:
Regarding the early bailout with `FWUPD_ERROR_NOT_FOUND`, that seems to have been changed in https://github.com/fwupd/fwupd/commit/0f490525f0f235da3539daf768bdab595e8f2671. If this is not desirable, a less fatal error needs to be thrown (and caught in the self-test accordingly).

Regarding the error messages, you could reduce their verbosity a bit by setting [`TSS2_LOG`](https://github.com/fwupd/fwupd/blob/0ad59cbc03522d068e4c6888a8d4e83283930938/plugins/uefi/fu-uefi-pcrs.c#L115) to `esys+error,tcti+none`. This would get rid of the list of TCTI warnings and errors, while keeping the final
```
ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
```
to see what is going on on the TCTI side. If you don't want any logging from tpm2-tss at all, set `TSS2_LOG` to `all+none` instead, that will hide all logging messages unless verbose mode is turned on.

superm1, 2019-09-17:
Yeah - so more than anything this turns into a policy decision, do we start to require TPMs to allow UEFI updates and be able to record PCR values and compare to metadata?
@hughsie my thought is in general we want to be pushing the ecosystem forward, and once more golden values for PCR0 end up in LVFS this would just be holding it back.  Curious your thoughts though.

hughsie, 2019-09-17:
I think we have to allow UEFI updates wven when the TPM is turned off. It's irresponsible to leave the user with an insecure system just because they've turned off something we'd quite like.

diabonas, 2019-09-17:
IMHO turning off the TPM doesn't make any sense, this doesn't improve privacy or security, but instead stops applications from making use of the TPM e.g. to protect your secret keys against exfiltration by malware, therefore worsening security.

On the other hand, fwupd doesn't make use of the core capabilities of the TPM, it only uses it to check the success of the UEFI firmware update. Not every system has a TPM, so I think the check must be optional (even if it is strongly desirable to have this verification, of course).

superm1, 2019-09-17:
>Not every system has a TPM, so I think the check must be optional (even if it is strongly desirable to have this verification, of course).

Right, at least machines that support `UpdateCapsule` are much more likely to have one though.

I guess we need to catch `FWUPD_ERROR_NOT_FOUND` when the pcr stuff is being initialized in the function.  I'll do a PR tomorrow.

closed 2019-09-17 by superm1

superm1, 2019-09-23:
FYI @LinuxOnTheDesktop  the question marks are fixed in the current edge snap by switching core18.  The 1.3.2 stable release will have the fix.

LinuxOnTheDesktop, 2019-09-23:
@superm1 

Thanks. By the way, you might want to link your page on snap to the frontpage of this git repo, and add to that page that (1) if installing the 'edge' snap one should remove the normal snap, (2) to install the edge snap one needs `sudo snap install --edge --classic`. EDIT: Actually, the process didn't work:
```
$ sudo snap remove fwupd
fwupd removed
$ sudo snap install fwupd --edge
error: This revision of snap "fwupd" was published using classic confinement and thus may perform
       arbitrary system changes outside of the security sandbox that snaps are usually confined to,
       which may put your system at risk.

       If you understand and want to proceed, repeat the command including --classic.
$ sudo snap install fwupd --edge --classic
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/1063/etc/dbus-1/system.d/org.freedesktop.fwupd.conf': No such file or directory)
$ 

```

superm1, 2019-09-23:
Thanks for checking.
Yeah looks like a bug specifically to fresh install, opening a PR shortly.

commit 78987fb3 (2019-09-17): uefi: make the TPM optional for runtime again (Fixes: #1362)

It's intentional to use a different error message for library failure
so that we can tell if there is a regression in CI related to TPM usage.