fwupd/fwupd #1365: fwupd report only looks in /etc/machine-id

juanitotc, 2019-09-17:
$ fwupdmgr --version
client version: 1.3.1
compile-time dependency versions
        gusb:   0.2.9
        efivar: 37
daemon version: 1.3.
```

$ fwupdmgr get-devices
Latitude E7240
?
??Unifying Receiver:
      Device ID:           53d907a89b8cf6db282e318dcb99ad85850fb789
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.06_B0030
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Update State:        success
      Flags:               updatable|supported|registered
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4
                           c033c930-2f19-5148-bf82-5f232464d5d7
                           87fd7145-3913-50c8-bfcb-86f85006d7d1
```

After a successful firmware update, fwupd cannot report back as it only looks for /etc/machine-id:

$ fwupdmgr get-updates
No upgrades for Unifying Receiver, current is RQR24.06_B0030: RQR24.06_B0030=same, RQR24.05_B0029=older
________________________________________________

Devices that have been updated successfully:

 ? Unifying Receiver (RQR24.01_B0023 ? RQR24.06_B0030)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: Y
Failed to open file ?/etc/machine-id?: No such file or directory

On non-systemd machines, it should also look for /var/lib/dbus/machine-id

hughsie, 2019-09-17:
Can you do a PR to fix this please? It should be pretty straightforward.

juanitotc, 2019-09-17:
Sorry - I can see the issue is around here:

/tmp/fwupd-1.3.1/src/fu-self-test.c:    payload = fu_common_get_contents_bytes ("/etc/machine-id", &error);
/tmp/fwupd-1.3.1/libfwupd/fwupd-self-test.c:    if (!g_file_test ("/etc/machine-id", G_FILE_TEST_EXISTS)) {
/tmp/fwupd-1.3.1/libfwupd/fwupd-self-test.c:            g_test_skip ("Missing /etc/machine-id");
/tmp/fwupd-1.3.1/libfwupd/fwupd-common.c: * Gets a salted hash of the /etc/machine-id contents. This can be used to
/tmp/fwupd-1.3.1/libfwupd/fwupd-common.c: * machine-id from the machine-hash.
/tmp/fwupd-1.3.1/libfwupd/fwupd-common.c:       if (!g_file_get_contents ("/etc/machine-id", &buf, &sz, error))
/tmp/fwupd-1.3.1/libfwupd/fwupd-common.c:                                    "The machine-id is present but unset");

..but I lack the ability to fix it :(

superm1, 2019-09-17:
On my machine (with systemd) it looks like `/var/lib/dbus/machine-id` is actually a symlink to `/etc/machine-id`.  Is that nominal?  If so, maybe we should just look at the `/var/lib/dbus/machine-id` to cover both cases.

juanitotc, 2019-09-18:
I don't have a systemd box to check, but I believe /var/lib/dbus/machine-id is the default for dbus.

hughsie, 2019-09-18:
From https://www.freedesktop.org/software/systemd/man/machine-id.html

> The simple configuration file format of /etc/machine-id originates in the /var/lib/dbus/machine-id file introduced by D-Bus. In fact, this latter file might be a symlink to /etc/machine-id.

I think it's a little dangerous to assume that /var/lib/dbus/machine-id is symlinked to /etc/machine-id and think a fallback is a more suitable fix.

closed 2019-09-23 by hughsie