fwupd/fwupd #1366: fwupd does not give details for dell laptop

juanitotc, 2019-09-17:
```shell
$ fwupdmgr --version
client version: 1.3.1
compile-time dependency versions
        gusb:   0.2.9
        efivar: 37
daemon version: 1.3.1

```

```shell
$ fwupdmgr get-devices
Latitude E7240
?
??Unifying Receiver:
      Device ID:           53d907a89b8cf6db282e318dcb99ad85850fb789
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.06_B0030
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Update State:        success
      Flags:               updatable|supported|registered
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4
                           c033c930-2f19-5148-bf82-5f232464d5d7
                           87fd7145-3913-50c8-bfcb-86f85006d7d1

```

Is there some way to check why no details are given for the dell latitude e7240?

superm1, 2019-09-17:
You mean no BIOS details?  Can you check the daemon log for more information?
Also, do you have your TPM disabled?

juanitotc, 2019-09-17:
I meant no details like those shown above for the mouse unifying receiver -  I presume the presence of those would allow fwupd to check for a firmware update for the dell laptop.

The details are available:

$ sudo smbios-sys-info-lite
Libsmbios:    2.4.2
System ID:    0x05CA
Service Tag:  2FDPF12
Express Service Code: 5283578198
Asset Tag:  Not Specified
Product Name: Latitude E7240
BIOS Version: A27
Vendor:       Dell Inc.
Is Dell:      1
OEM String 1: Dell System
OEM String 2: 1[05CA]
OEM String 3: 3[1.0]
OEM String 4: 12[www.dell.com]
OEM String 5: 14[1]
OEM String 6: 15[0]
OEM String 7:  

All the daemon log shows is:

19:42:54:0508 FuMain               Called GetDevices()
19:42:54:0511 FuMain               Called GetHistory()
19:42:54:0512 FuCommon             not checking RQR24.01_B0023 as no version format set
19:42:54:0513 XbSilo               1.17ms      query on / with `components/component/releases/release/checksum[@target='container'][text()='fe037400f0977a13c215d94e427f6be14ae0be3e']/../../../../custom/value[@key='fwupd::RemoteId']` limit=1 -> 1 results

How can I confirm what fwupd believes the tpm status is?

superm1, 2019-09-17:
What sort of details are you expecting for the mouse?  Regarding host firmware and the TPM question, don't mind it .  It looks like it's in a success updated state now.  I suspect that machine is a little too old, it doesn't support UEFI capsule updates.

juanitotc, 2019-09-17:
The problem is not with the mouse - fwupd recognised it, downloaded and updated its firmware.

The problem is with the laptop - fwupd doesn't seem to fully recognise it or check if a firmware update exists.

superm1, 2019-09-17:
I really don't believe that model supports UEFI Capsule updates.  Looking at your SMBIOS tool output it's from before the BIOS versions were converted to triplets (XX.YY.ZZ).

juanitotc, 2019-09-18:
The latitude e7240 is 4-5 years old so maybe it is not supported - it does seem stange that fwupd gives so few details in comparison to the mouse though..

superm1, 2019-09-18:
Is your expectation that fwupd shows the version for the system firmware even if it can't update it?

I guess let's check a few things:
1. Are you booted in UEFI mode?
2. Assuming 1 is yes, can you check `fwupdmgr get-devices --show-all-devices` output?

juanitotc, 2019-09-19:
Yes, my expectation would be that details are shown for the laptop, even if it couldn't be updated.

I'm booted in uefi mode.

$ fwupdmgr get-devices --show-all-devices
Latitude E7240
?
??Unknown Device:
?     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
?     Current version:     0b
?     Vendor:              PCI:0x8086
?     Flags:               internal|registered
?     GUIDs:               3e4de07e-29ea-5976-8f81-b426702e4d28
?                          c544adab-efd9-5019-b90d-5eddd1612ddf
?   
??Intel AMT [unprovisioned]:
?     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
?     Summary:             Hardware and firmware technology for remote out-of-band management
?     Current version:     9.5.65.3000
?     Bootloader Version:  9.5.65.3000
?     Vendor:              Intel Corporation
?     Flags:               internal|registered
?     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
?   
??Unifying Receiver:
      Device ID:           53d907a89b8cf6db282e318dcb99ad85850fb789
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.06_B0030
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Flags:               updatable|registered
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4

superm1, 2019-09-19:
That fix above was merged and should help show you what you want for your system.  Since you were using the snap, feel free to switch to the `--edge` channel if you want to test it.  Otherwise it will be in fwupd 1.3.2.

closed 2019-09-19 by superm1

juanitotc, 2019-09-20:
That seems to work - thanks

$ fwupdmgr get-devices --show-all-devices
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Latitude E7240
?
??Unknown Device:
?     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
?     Current version:     0b
?     Vendor:              PCI:0x8086
?     Flags:               internal|registered
?     GUIDs:               3e4de07e-29ea-5976-8f81-b426702e4d28
?                          c544adab-efd9-5019-b90d-5eddd1612ddf
?   
??System Firmware:
? ?   Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
? ?   Current version:     A27
? ?   Vendor:              Dell Inc.
? ?   Update Error:        UEFI Capsule updates not available or enabled
? ?   Flags:               internal|require-ac|registered|needs-reboot
? ?   GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
? ? 
? ??Intel AMT [unprovisioned]:
?       Device ID:         e2623122c99d58220498aacbfcfdb1baebbae3c5
?       Summary:           Hardware and firmware technology for remote out-of-band management
?       Current version:   9.5.65.3000
?       Bootloader Version:9.5.65.3000
?       Vendor:            Intel Corporation
?       Flags:             internal|registered
?       GUID:              2800f812-b7b4-2d4b-aca8-46e0ff65814c
?     
??Unifying Receiver:
      Device ID:           53d907a89b8cf6db282e318dcb99ad85850fb789
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.06_B0030
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Flags:               updatable|supported|registered
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4

PR #1376 (2019-09-19): uefi, dell-esrt: Show firmware version when capsule updates unavailable
When the system doesn't support UEFI capsule updates no firmware version
is displayed for the BIOS.  Fix this by creating a dummy device:
```
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     1.7.0
│     Vendor:              Dell Inc.
│     Update Error:        UEFI Capsule updates not available or enabled
│     Flags:               internal|require-ac|registered|needs-reboot
```

If the dell-esrt plugin determines that capsule updates can be enabled
however,  [... cut]
commit 01a7958c (2019-09-19): uefi, dell-esrt: Show firmware version when capsule updates unavailable

When the system doesn't support UEFI capsule updates no firmware version
is displayed for the BIOS.  Fix this by creating a dummy device:
```
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     1.7.0
│     Vendor:              Dell Inc.
│     Update Error:        UEFI Capsule updates not available or enabled
│     Flags:               internal|require-ac|registered|needs-reboot
```

If the dell-esrt plugin determines that capsule updates can be enabled
however, [... cut]