fwupd/fwupd #1369: Docking station not removed when disconnecting

BjoernDaase, 2019-09-18:
I have a Dell XPS 9370 and a Dell WD15.
When booting the XPS with the WD15 connected, the WD15 dock shows up as a device when running `fwupdmgr get-devices` --> expected.  
But after disconnecting the dock, it does not get removed from the devices. It is still listed when running `fwupdmgr get-devices` --> unexpected.

Furthermore I observe the following behaviour:
When booting the XPS without the WD15 connected, the WD15 dock does not show up as a device when running `fwupdmgr get-devices` --> expected.  
But when you then connect the dock, it does not get added to the devices. It is still not listed when running `fwupdmgr get-devices` --> unexpected.

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
XPS 13 9370 Thunderbolt Controller
  DeviceId:             8108df885fc4985d7fc9ad2f600b661cced3d00e
  Guid:                 4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-18

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(c666009482077cfd42e0675a34f9f76c17ad8bb6)
  Checksum:             SHA256(f76f740407390389bec7e254c9f20579b619502c2aa7861f01b530ab9e27081b)
  Version:              0.1.11.1
  VersionLowest:        0.1.11.1
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-09-18

XPS 13 9370 TPM 2.0
  DeviceId:             370e10407b1f04ade798a9f1d3e1fa57c67750c3
  Guid:                 cb8da68d-cd80-5f5b-8fef-038383adbb83
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              7.2.0.1
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-09-18

Synaptics VMM3332 inside Dell WD15/TB16/TB18 wired Dock
  DeviceId:             d83854e8c81f5d47b0b99c8a3b42ae5127c75eb8
  Guid:                 78c73924-821a-5e6f-af82-86df6f67b182
  Guid:                 02b77ccb-03ab-521f-b98e-ef05395e9fa2
  Guid:                 a6f00ca7-54b7-5d51-9d6e-543d8df85376
  Summary:              Multi-Stream Transport Device
  Plugin:               synapticsmst
  Flags:                updatable|registered
  Vendor:               Synaptics
  Version:              3.10.002
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-09-18

PM981 NVMe Samsung 512GB
  DeviceId:             f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 ad4ad443-6f4d-5735-ba86-17a1583a2403
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              EXA73D1Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-18

Unifying Receiver
  DeviceId:             f2a4275eddaca5782ce76f18b227b11cb965658e
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.06_B0030
  VersionBootloader:    BOT03.01_B0008
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-09-18
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001
Boot0000* Fedora	HD(1,GPT,6f04ded3-0f10-4e00-ba56-401890ed82ff,0xffff,0xffff0)/File(\EFI\fedora\shimx64.efi)
Boot0001* UEFI: PM981 NVMe Samsung 512GB, Partition 1	HD(1,GPT,6f04ded3-0f10-4e00-ba56-401890ed82ff,0xffff,0xffff0)/File(EFI\boot\bootx64.efi)..BO
```

```shell
$ efivar -l | grep fw
-none-
```

```shell
$ tree /boot
/boot
├── 0-rescue
├── config-5.2.14-200.fc30.x86_64
├── config-5.2.9-200.fc30.x86_64
├── config-5.3.0-0.rc6.git0.1.fc31.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-32e7c023a79043fe93cf61b18871b289.img
├── initramfs-5.2.14-200.fc30.x86_64.img
├── initramfs-5.2.9-200.fc30.x86_64.img
├── initramfs-5.3.0-0.rc6.git0.1.fc31.x86_64.img
├── loader
│   └── entries [error opening dir]
├── memtest86+-5.01
├── System.map-5.2.14-200.fc30.x86_64
├── System.map-5.2.9-200.fc30.x86_64
├── System.map-5.3.0-0.rc6.git0.1.fc31.x86_64
├── vmlinuz-0-rescue-32e7c023a79043fe93cf61b18871b289
├── vmlinuz-5.2.14-200.fc30.x86_64
├── vmlinuz-5.2.9-200.fc30.x86_64
└── vmlinuz-5.3.0-0.rc6.git0.1.fc31.x86_64

6 directories, 77 files
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `dnf`
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2019-09-18:
Yes, this is a side effect of how the `synapticsmst` used to work.  In fwupd 1.3.1 it's been adjusted and won't behave this way anymore.

closed 2019-09-18 by superm1

BjoernDaase, 2019-09-18:
@superm1 Ah, sounds good. Then I am looking forward to see it in the Fedora repositories :)