fwupd/fwupd #1371: How do I see the changelog again of my last update?

kaihendry, 2019-09-18:
Use case: I updated. But I didn't keep a record of the update notes. How do I see a history with a changelog / release notes of the last applied update?

I was hoping `fwupdmgr get-history` would help, but it doesn't.

https://s.natalian.org/2019-09-18/fwupd.txt

Btw great work otherwise.

superm1, 2019-09-18:
I would think get-releases should show what you're looking for.

kaihendry, 2019-09-18:
There is no date, so I am not sure how to tell which is which.

superm1, 2019-09-18:
`get-history` should show you the version you updated to/from and `get-releases` should hopefully show those releases.

superm1, 2019-09-19:
OK so here's how to do it.  By looking at `get-history` you can see which releases were installed:
```
# fwupdmgr get-history 
XPS 13 9380
│
├─XPS 13 9380 System Firmware:
│     Device ID:           6c24a747f97668873b761558e322398a91dbf394
│     Current version:     0.1.6.0
│     Update State:        success
│     Flags:               internal|updatable|require-ac|supported|needs-reboot|reported
│   
└─XPS 13 9380 System Firmware:
      Device ID:           6c24a747f97668873b761558e322398a91dbf394
      Current version:     0.1.7.0
      Update State:        success
      Flags:               internal|updatable|require-ac|supported|needs-reboot|reported
 ```

Followed by running `get-releases` to see which releases were available (with changelogs):
```
# fwupdmgr get-releases
Choose a device:
0.	Cancel
1.	8357922f7f208e1f3b7537141ce17db04bc219d9 (Thunderbolt controller in Dell dock)
2.	1eca9eabb0c992c136e1deb1f89e3f70c465aa1c (Package level of Dell dock)
3.	ca36279eeac13463fd5d974c88794859294fbfb1 (RTS5413 in Dell dock)
4.	acb404019656654d44f80922d94735e831d9bb40 (RTS5487 in Dell dock)
5.	58931e8c9ef5eead9c007563814f96fcadb1b993 (VMM5331 in Dell dock)
6.	a455a3a21752e5d31c228f1f5b51761dad88a432 (WD19TB)
7.	6c24a747f97668873b761558e322398a91dbf394 (System Firmware)
7
XPS 13 9380
│
├─XPS 13 9380 System Update:
│     Version:             0.1.7.0
│     Remote ID:           lvfs
│     Summary:             Firmware for the Dell XPS 13 9380
│     License:             proprietary
│     Size:                22.4 MB
│     Vendor:              Dell Inc.
│     Flags:               none
│     Description:         This stable release fixes the following issues:
│   
│   �� Fixed the issue where the Dell Power Manager displays an error when a 130W Type-C adapter is connected to the system.
│   
│   me new functionality has also been added:
│   
│   �� Added a new feature to automatically suspend BitLocker before upgrading the firmware. After the firmware upgrade is complete, BitLocker is automatically enabled.
│   
├─XPS 13 9380 System Update:
│     Version:             0.1.6.0
│     Remote ID:           lvfs
│     Summary:             Firmware for the Dell XPS 13 9380
│     License:             proprietary
│     Size:                22.4 MB
│     Vendor:              Dell Inc.
│     Flags:               is-downgrade|blocked-version
│     Description:         This stable release fixes the following issues:
│   
│   �� Fixed the issue where a USB drive cannot be used as a bootable device when connected to a Type-C port.
│   
│   me new functionality has also been added:
│   
│   �� Updated the alert message that is displayed when Dell Thunderbolt Dock WD19TB is connected to systems with Non-Thunderbolt port.
│   �� Added BIOS option to bypass the Dock warning messages.
│   
├─XPS 13 9380 System Update:
│     Version:             0.1.5.0
│     Remote ID:           lvfs
│     Summary:             Firmware for the Dell XPS 13 9380
│     License:             proprietary
│     Size:                22.4 MB
│     Vendor:              Dell Inc.
│     Flags:               is-downgrade|blocked-version
│     Description:         This stable release fixes the following issues:
│   
│   �� Fixed the issue where BitLocker may prompt for recovery if Auto Power On option in BIOS setup is set to wake up system at a specific time
│   
│   me new functionality has also been added:
│   
│   �� This update integrates the BIOSConnect feature into Dell SupportAssist OS Recovery. It connects the system to the Dell image server to download and recover the operating system.
│   �� Reduces Deferred Procedure Call (DPC) latency for real-time task performance enhancement, such as audio playback.
│   �� Enhanced the compatibility of Jabra office headset when connected to Dell Docking Station WD19.
│   
├─XPS 13 9380 System Update:
│     Version:             0.1.4.0
│     Remote ID:           lvfs
│     Summary:             Firmware for the Dell XPS 13 9380
│     License:             proprietary
│     Size:                20.0 MB
│     Vendor:              Dell Inc.
│     Flags:               is-downgrade|blocked-version
│     Description:         This stable release fixes the following issues:
│   
│   �� Firmware updates to address security advisories INTEL-SA-00213 and INTEL-SA-00233
│   
│   curity issues fixed:
│   
│   �� CVE-2019-0090
│   �� CVE-2019-0086
│   �� CVE-2019-0091
│   �� CVE-2019-0093
│   �� CVE-2019-0098
│   �� CVE-2018-12126
│   �� CVE-2018-12127
│   �� CVE-2018-12130
│   �� CVE-2019-11091
│   
└─XPS 13 9380 System Update:
      Version:             0.1.3.2
      Remote ID:           lvfs
      Summary:             Firmware for the Dell XPS 13 9380
      License:             proprietary
      Size:                19.9 MB
      Vendor:              Dell Inc.
      Flags:               is-downgrade|blocked-version
      Description:         This stable release fixes the following issues:
    
    �� Fixed an issue with Secure Boot Option ROM Signature Verification.
    �� Firmware updates to address security advisory INTEL-SA-00185 (CVE-2018-12188 CVE-2018-12190 CVE-2018-12191 CVE-2018-12192 CVE-2018-12199 CVE-2018-12198 CVE-2018-12200 CVE-2018-12187 CVE-2018-12196 CVE-2018-12185).
    �� Firmware updates to address security advisories INTEL-SA-00191(CVE-2018-12201, CVE-2018- 12202, CVE-2018-12203, CVE-2018-12205).
    
    me new functionality has also been added:
    
    �� Removed the option Always Allow Dell Docks from BIOS settings. Dell dock connection and the port behavior will be controlled via the USB and Thunderbolt Adapter configuration settings under operating system environment.
    �� Replaces Absolute Software service from Computrace to Absolute Persistence Module service.
    �� Supports Dell Dock Station WD19.
```

kaihendry, 2019-09-20:
Without dates, I still don't quite know what the previous state was and when. And if I don't know that, how am I supposed to bisect and work out what caused a bad update?

Can I please ask for dates in these commands, as a feature request?

https://s.natalian.org/2019-09-20/feedback.mp4

Studying https://s.natalian.org/2019-09-20/history.txt
Can I assume I updated from 0.1.25 (last entry in history) to 0.1.33 (last entry in get-releases)? I'm rather confused.

superm1, 2019-09-20:
If you can try that PR above, I think it will better message what you're looking for in `get-history`.  You'll still need to cross reference `get-releases` though for the full changelog.

superm1, 2019-09-20:
FYI, versions before fwupd 1.3.2 are not recording the modified date to the database, so you won't be able to cross reference it until you perform updates from 1.3.2.  However you will be able to see which versions you started from and went to along with changelogs now.

Sample output:
```
├─XPS 13 9380 System Firmware:
│ │   Device ID:           6c24a747f97668873b761558e322398a91dbf394
│ │   Previous version:    0.1.6.0
│ │   Update State:        success
│ │   Flags:               internal|updatable|require-ac|supported|needs-reboot|reported|historical
│ │   GUID:                ce945437-7358-49f1-95d8-6b694a10a755
│ │ 
│ └─XPS 13 9380 System Update:
│       New version:       0.1.7.0
│       Remote ID:         lvfs
│       Summary:           Firmware for the Dell XPS 13 9380
│       License:           proprietary
│       Size:              22.4 MB
│       Vendor:            Dell Inc.
│       Description:       This stable release fixes the following issues:
│     
│      Fixed the issue where the Dell Power Manager displays an error when a 130W Type-C adapter is connected to the system.
│     
│      new functionality has also been added:
│     
│      Added a new feature to automatically suspend BitLocker before upgrading the firmware. After the firmware upgrade is complete, BitLocker is automatically enabled.
```

closed 2019-09-20 by superm1