fwupd/fwupd #1383: We need a way to notify the clients that a replug is needed

FFY00, 2019-09-20:
Currently we are relying on the release notes to have the instructions for the update, like when to replug a device. This works but is not optimal, we know when the replug needs to happen in the code, so we could easily notify the client of this. The replug is the most common example, but there might be other type of instructions that would benefit from this.

The issue is how to implement this? @superm1 any ideas?

superm1, 2019-09-20:
There is functionalities to allow showing an update message in the frontend after the update. 

The only plugin that uses this today is dell-dock and only frontend support is fwupdmgr IIRC.

I think that makes sense for this behavior though.

superm1, 2019-09-20:
Regarding needing a replug, is that really true? Can't you do a USB reset ioctl to accomplish the same result too?

FFY00, 2019-09-20:
I am not sure. I will need to test.

superm1, 2019-09-20:
btw, here is the commit that added `UpdateMessage` stuff if you want to look at it: https://github.com/fwupd/fwupd/commit/32241f4256bfbc6d1042e7ae725edb7d7906dc59

Any CAB file can just include the key `LVFS::UpdateMessage` and `fwupdmgr`/`fwupdtool` will show the message after it's done installing (fwupd 1.2.4+).
AFAIK `gnome-software`, `gnome-firmware`, and `kde-discover` don't have this functionality yet.

FFY00, 2019-09-20:
Nevermind, an ioctl won't work for wireless devices because the user still needs to turn off and on the device itself.

superm1, 2019-09-20:
Then yeah I think that `UpdateMessage` is a good avenue.  The only problem is it will mean re-spinning any XML in the CABs.

We probably also need to make a concerted effort to get the other frontends plumbed.

superm1, 2019-09-20:
I've filed bugs with the other frontends I know about:
* KDE Discover: https://bugs.kde.org/show_bug.cgi?id=412126
* Gnome Firmware: https://gitlab.gnome.org/hughsie/gnome-firmware-updater/issues/10
* Gnome Software: https://gitlab.gnome.org/GNOME/gnome-software/issues/806
* Pop OS firmware manager: https://github.com/pop-os/firmware-manager/issues/73

hughsie, 2019-09-20:
Do we set UpdateMessage before the replug check? I thought the property would be set after the update had completed.

superm1, 2019-09-20:
My understanding of this was a need to have the user manually power cycle, but not that the daemon would actually be "waiting" for it.  Is that wrong?

FFY00, 2019-09-23:
Actually `UpdateMessage` doesn't solve our issue. The issue is that in some Logitech devices a replug is needed in order to enter bootloader mode (this avoids someone flashing a malicious firmware without you knowing), so we need to show this instruction in the middle of the update.

Firmware update process (HID++ 2.0 - c1 feature):
- Tell the device to enter bootloader mode
(now the device will enter the bootloader on the next boot)
- Ask the user to replug the device
- Flash the firmware
- Wait for the device to come back
- Verify if the update was successful

cc @xnlcasad

superm1, 2019-09-23:
This flow is similar to how 8bitdo controller updates work, you have to do something manual to enter bootloader mode after you tell fwupd that it's going to happen.

FFY00, 2019-09-23:
What are we doing thee?

superm1, 2019-09-23:
Basically when calling update if the device isn't the bootloader, an error is returned (https://github.com/fwupd/fwupd/blob/master/plugins/ebitdo/fu-ebitdo-device.c#L405)

So whatever client requested it will return the error message.

The daemon internally will wait up to 40 seconds for the user to plug again:
https://github.com/fwupd/fwupd/blob/d8aaa21cde3dbd4dbc6343cff7463e5e7732d485/src/fu-device.h#L81

If they do, it re-uses that device and they run the update again it will do it.

It's a little clumsy, but doesn't block the dbus connection the whole time and does "work".

FFY00, 2019-09-23:
Yeah, it's the same as `unifying` then. This is bad because if the user doesn't replug device (and it's not obvious) the update will fail. Logitech was actually getting a bunch of failed update reports and the firmware was demoted to embargo. This is definitely and issue, not just a UI problem.

superm1, 2019-09-23:
Can you detect which devices are needing this flow?  I guess if they support HID++ 2.0 - c1?

FFY00, 2019-09-23:
Yes, devices that use the `c1` feature of HID++ 2.0.

superm1, 2019-10-18:
@FFY00 I was looking at unifying code today, and it seems like some of this infrastructure is started and just needs some minor cleanups.

https://github.com/fwupd/fwupd/blob/master/plugins/unifying/fu-unifying-peripheral.c#L620
Seems to check if the c1 feature is available and actually puts the device into replug.

I think the error just needs to be written in this case and `return FALSE` to tell the user that the device needs to be re-plugged instead.  I've filed a PR for this, see if that helps.

commit 2baf0353 (2019-10-18): unifying: notify users to replug when using dfu control (Fixes: #1383)
commit 94d35c50 (2019-10-20): unifying: notify users to replug when using dfu control (Fixes: #1383)
commit b2209a94 (2019-10-31): unifying: notify users to replug when using dfu control (Fixes: #1383)
commit 0c7ca0b4 (2019-10-31): unifying: notify users to replug when using dfu control (Fixes: #1383)
closed 2019-11-01 by hughsie