fwupd/fwupd #1384: We need to listen to connection notifications in logitech

FFY00, 2019-09-20:
Logitech receivers have a enumerator interface called DJ (see [this](https://lekensteyn.nl/files/logitech/Unifying_receiver_DJ_collection_specification_draft.pdf)). We have a kernel driver that will create hidraw nodes for each device, and these are the hidraw nodes we use in fwupd to update the devices. However the kernel driver behaves a bit differently than normal HID devices, when a device is disconnected the hidraw node doesn't go away. For this reason we fail to detect a replug if the user turns off and on the device itself, he will need to replug the receiver instead.

We need to listen to HID++ connection events ([here](https://github.com/fwupd/fwupd/blob/e091427993122add0bd45349b957f2df0946b7c3/plugins/unifying/fu-unifying-runtime.c#L95)) and export this information internally so that we can properly detect a device replug.

superm1, 2019-09-20:
I think it's worth discussing this with the driver maintainers. That behavior sounds like it could be a kernel bug.

FFY00, 2019-09-20:
It is not. It's a technical limitation of some devices. I already discussed this with @bentiss.

bentiss, 2019-09-20:
Just to explain *why* we do not kill/recreate the devices on disconnect/connect. Contrary to Bluetooth or plain USB mouse, the DJ code has no buffer where it would stores the incoming events if the device is still not ready. `hid-logitech-dj` is merely a router of events.

So to prevent losing a few key presses during the initial connect, we have to create the virtual HID device beforehand and we can't really destroy it or every time a keyboards goes to sleep, or you'll lose the first few keys...

I'm open to any kernel fix however if it is required for fwupd

FFY00, 2020-02-15:
FYI, we started a discussion upstream, I forgot to CC you.

https://www.spinics.net/lists/linux-input/msg65615.html

superm1, 2020-02-18:
> FYI, we started a discussion upstream, I forgot to CC you.
> 
> https://www.spinics.net/lists/linux-input/msg65615.html

It is difficult to follow the discussions when not followed along in the mailbox, just feel free to post the conclusions if they're reached.

bentiss, 2020-02-19:
> It is difficult to follow the discussions when not followed along in the mailbox, just feel free to post the conclusions if they're reached.

End result is the kernel will emit a udev event `KOBJ_CHANGE` (`kobject_uevent(&hdev->dev.kobj, KOBJ_CHANGE);`) whenever the device connects to notify user-space about a new connection.

That should sort all of our user space problems (hopefully).

closed 2021-01-27 by superm1