fwupd/fwupd #1395: Latest System Firmware (Version 0.1.34) for Lenovo ThinkPad T480s does not show up

maaattes, 2019-09-24:
Latest System Firmware ([Version 0.1.34](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/ds502226) released on 18.09.2019) for Lenovo ThinkPad T480s does not show up. It was [uploaded to LVFS](https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN22ET.firmware).

Maybe the same problem as [last time](https://github.com/fwupd/fwupd/issues/1280)?

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             24dd1466bf80723529c067bcfd0d35ab5bea2f38
  Guid:                 3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-24

20L8S02D00 System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(0bb526eed55dbd2fd21a31e917db53d56d228b5f)
  Checksum:             SHA256(ae7d0fb558d40ad95ba6c1d1a15b6be0edc5ce48cf59d1eb8750eca105599d9e)
  Version:              0.1.33
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-09-24
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-09-24

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.16
  VersionLowest:        0.1.16
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-09-24
  UpdateState:          success

INTEL SSDPEKKF512G8L
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              L08P
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-24

________________________________________________

Erfolgreich aktualisierte Geräte:

 • 20L8S02D00 System Firmware (0.1.31 → 0.1.33)
 • UEFI Device Firmware (0.1.13 → 0.1.16)
 • UEFI Device Firmware (184.60.3561 → 184.65.3590)
```

Please answer the following questions:

- Operating system and version: Fedora 30
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): via dnf install
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-09-24:
Not the same problem, I see it in the metadata (in /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz -- search for ThinkPadN22ET using VIM) -- what does "get-updates" say for you?

maaattes, 2019-09-24:
```shell-session
$ fwupdmgr get-updates
No upgrades for 20L8S02D00 System Firmware, current is 0.1.33: 0.1.33=same, 0.1.31=older, 0.1.30=older, 0.1.29=older, 0.1.27=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.16: 0.1.16=same, 0.1.13=older
________________________________________________

Erfolgreich aktualisierte Geräte:

 • 20L8S02D00 System Firmware (0.1.31 → 0.1.33)
 • UEFI Device Firmware (0.1.13 → 0.1.16)
 • UEFI Device Firmware (184.60.3561 → 184.65.3590)
```

If I search `/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz` for `ThinkPadN22ET` I only find 0.1.33 :(

[image]

hughsie, 2019-09-24:
Do you still only see that old version after a "fwupdmgr refresh"?

maaattes, 2019-09-24:
Just updated successfully `fwupdmgr refresh` did the trick.

closed 2019-09-24 by FFY00