fwupd/fwupd #1396: dell-self-test failing on Fedora 31

hughsie, 2019-09-24:
I get:

    $ ./plugins/dell/dell-self-test 
    # random seed: R02S338148a3e249e0769dd23561bd8028e3
    1..2
    # Start of fwupd tests
    # FuPlugin-DEBUG: performing init() on /home/hughsie/Code/fwupd/build/plugins/dell/../uefi/libfu_plugin_uefi.so
    # FuPlugin-DEBUG: performing startup() on uefi
    # FuPlugin-DEBUG: performing init() on /home/hughsie/Code/fwupd/build/plugins/dell/libfu_plugin_dell.so
    # FuPluginDell-DEBUG: Using libsmbios 2.4
    # FuPlugin-DEBUG: performing startup() on dell
    # FuPluginDell-DEBUG: Called with fake SMBIOS implementation. We're ignoring test for SBMIOS table and ESRT. Individual calls will need to be properly staged.
    # FuPlugin-DEBUG: performing coldplug() on dell
    # FuPluginDell-DEBUG: TPM HW version: 0x0
    # FuPluginDell-DEBUG: TPM Status: 0x0
    # FuPluginDell-DEBUG: TPM not enabled (0)
    # FuPluginDell-DEBUG: No switchable TPM detected
    # FuPluginDell-DEBUG: Failed to query system for TPM information: (4294967294)
    # FuPluginDell-DEBUG: TPM HW version: 0x0
    # FuPluginDell-DEBUG: TPM Status: 0x101
    # FuPluginDell-DEBUG: Creating primary TPM GUID 0000-1.2 and secondary TPM GUID 0000-2.0
    # FuDevice-DEBUG: using 3ca929adb22a017654a24ce341b35205d2ec45cd for DELL-c8926cae-cee4-5d3b-80e5-70bd5ff1c101lu
    ERROR:tcti:src/tss2-tcti/tcti-device.c:439:Tss2_Tcti_Device_Init() Failed to open device file /dev/tpm0: Permission denied 
    WARNING:tcti:src/tss2-tcti/tctildr.c:62:tcti_from_init() TCTI init for function 0x7f14a312ffc0 failed with a000a 
    WARNING:tcti:src/tss2-tcti/tctildr.c:92:tcti_from_info() Could not initialize TCTI named: tcti-device 
    ERROR:tcti:src/tss2-tcti/tctildr-dl.c:150:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-default.so 
    Bail out! FATAL-WARNING: Failed to create connection with service: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected

    ** (./plugins/dell/dell-self-test:27987): WARNING **: 12:06:57.995: Failed to create connection with service: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected
    Trace/breakpoint trap (core dumped)

and as root, the more helpful:

    # FuPluginDell-DEBUG: Creating primary TPM GUID 0000-1.2 and secondary TPM GUID 0000-2.0
    # FuDevice-DEBUG: using 3ca929adb22a017654a24ce341b35205d2ec45cd for DELL-c8926cae-cee4-5d3b-80e5-70bd5ff1c101lu
    ERROR:sys:src/tss2-sys/api/Tss2_Sys_Execute.c:114:Tss2_Sys_ExecuteFinish() Unsupported device. The device is a TPM 1.2 
    ERROR:esys:src/tss2-esys/api/Esys_Startup.c:216:Esys_Startup_Finish() Received a non-TPM Error 
    ERROR:esys:src/tss2-esys/api/Esys_Startup.c:78:Esys_Startup() Esys Finish ErrorCode (0x00080001) 
    **
    FuPluginDell:ERROR:../plugins/dell/fu-self-test.c:137:fu_plugin_dell_tpm_func: 'ret' should be TRUE
    Bail out! FuPluginDell:ERROR:../plugins/dell/fu-self-test.c:137:fu_plugin_dell_tpm_func: 'ret' should be TRUE
    Aborted

This is on a Lenovo P50 if that helps.

superm1, 2019-09-24:
I guess it's specifically because of using a real TPM 1.2 in this hardware and the tcti finding it?

hughsie, 2019-09-24:
Yes, I think the TPM is in 1.2 mode.

commit 9a57fde1 (2019-09-24): trivial: dell: make failures to add GUIDs non-fatal (Fixes: #1396)

Avoids situations that the TSS doesn't handle well such as TPM1.2.
commit 9cc48694 (2019-09-24): trivial: dell: make failures to add GUIDs non-fatal (Fixes: #1396)

Avoids situations that the TSS doesn't handle well such as TPM1.2.
commit 9297d8ec (2019-09-24): trivial: dell: make failures to add GUIDs non-fatal (Fixes: #1396)

Avoids situations that the TSS doesn't handle well such as TPM1.2.
commit eab3809d (2019-09-25): trivial: uefi, dell: skip self tests with TPM if non-root (Fixes: #1396)

There are some packaging problems in some distributions that lead
to TSS stack emitting warnings that will fail self tests.

These don't occur as root, and furthermore those distributions run
CI as root already.

And in the dell plugin make it non-fatal to have TPM register read failures
in case the system has TPM1.2 not TPM2.0
closed 2019-09-25 by superm1