fwupd/fwupd #1400: "No reports require uploading" although there are reports

der-eismann, 2019-09-24:
Hey guys, once again: Thanks for this awesome project!

However I'm encountering a quite annoying bug. Everytime I execute `fwupdmgr get-updates` or similar the program tells me that there are reports to upload. But when I confim with `Y` I get the error that there are no reports that require uploading. Here's the verbose log:

```
fwupdmgr -v get-updates
No upgrades for 20L8S02D00 System Firmware, current is 0.1.34: 0.1.34=same, 0.1.33=older, 0.1.31=older, 0.1.30=older, 0.1.29=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.16=older, 0.1.13=older
________________________________________________

Devices that have been updated successfully:

 ? 20L8S02D00 System Firmware (0.1.33 ? 0.1.34)
 ? UEFI Device Firmware (0.1.16 ? 0.1.17)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: 
(fwupdmgr:3402): Fwupd-DEBUG: 22:49:50.502: looking for os-release at /etc/os-release
(fwupdmgr:3402): FuMain-DEBUG: 22:49:50.519: lvfs has no RemoteURI
(fwupdmgr:3402): FuMain-DEBUG: 22:49:50.519: lvfs has no RemoteURI
(fwupdmgr:3402): FuMain-DEBUG: 22:49:50.519: ignoring e7cc234e82d08b6a14e246beb6b5f8142091e570 with UpdateState needs-reboot
(fwupdmgr:3402): FuMain-DEBUG: 22:49:50.519: ignoring 323bda90831520867f8c02544efb9711dfca9fb6 with UpdateState needs-reboot
No reports require uploading
``` 

This also persisted after updating again. Not sure if this is distro-specific or what could be the cause. Let me know if you need additional information!

```shell
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

```shell
$ fwupdmgr get-devices    
Thinkpad T480s Thunderbolt Controller
  DeviceId:             2c9b2f711997d861656a6083f26195d25c1f18b0
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              17.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-24

20L8S02D00 System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.34
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-09-24
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-09-24

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.17
  VersionLowest:        0.1.17
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-09-24
  UpdateState:          success

KXG5AZNV512G TOSHIBA
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
  Guid:                 7acac8b6-ad80-5f1a-90c5-165e7938f773
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              5109AALA
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-24
```

Please answer the following questions:

- Operating system and version: Ubuntu 19.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2019-09-24:
Can you check `/etc/fwupd/remotes.d/lvfs.conf`?  It sounds like you're missing `ReportUri` from there, perhaps you manually removed it?

superm1, 2019-09-24:
And if you have that field, can you please post your `/var/lib/fwupd/pending.db` somewhere to look at?

der-eismann, 2019-09-24:
The field was indeed commented out... but I don't remember editing it and according to the timestamps I also didn't edit it:
```
➜  ~ cat /etc/fwupd/remotes.d/lvfs.conf
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
Keyring=gpg
MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.gz
#ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=fwupd
ApprovalRequired=false
➜  ~ ll /etc/fwupd/remotes.d 
insgesamt 24K
-rw-r--r-- 1 root root 247 Jul  4 00:51 dell-esrt.conf
-rw-r--r-- 1 root root 181 Apr 10 23:20 fwupd.conf
-rw-r--r-- 1 root root 306 Jul  4 00:51 lvfs.conf
-rw-r--r-- 1 root root 351 Jul  4 00:51 lvfs-testing.conf
-rw-r--r-- 1 root root 317 Jul  4 00:51 vendor.conf
-rw-r--r-- 1 root root 296 Jul  4 00:51 vendor-directory.conf
```
Not sure if the Ubuntu version is delivered like this... I'll do some additional testing tomorrow, now that I have a clue where to start :)

superm1, 2019-09-24:
I'm a little confused you have 1.2.9. 19.04 has 1.2.5 and 19.10 has 1.2.10. 

Is this some other distro? Or a manually backported package? Could a downstream packager have caused this?

superm1, 2019-09-25:
To me it looks like this is likely something done by popos. That's exactly what their file looks like.

https://github.com/pop-os/fwupd/blob/master/data/remotes.d/lvfs.conf

der-eismann, 2019-09-25:
Oh my, this is embarrassing. I think I added the popos PPA for their GTK theme... didn't think I would get their own fwupd.
After disabling their repo and installing the version from the Ubuntu repo uploading the reports worked fine. Sorry for causing trouble :see_no_evil:

hughsie, 2019-09-25:
I still think this is a bug; if we comment out the report uri we should just not ask the user to upload.

der-eismann, 2019-09-25:
Fair point

closed 2019-09-25 by superm1