fwupd/fwupd #1401: Can fwupd update the BIOS?

kaihendry, 2019-09-25:
I was performing a BIOS update for my NUC5CPYB since it was having some fan noise, by placing the `PY0078.bio` on a vfat formatted USB stick from https://downloadcenter.intel.com/download/29056/BIOS-Update-PYBSWCEL-86A-?product=85254

What I was wondering after doing it, could **fwupd** perform this operation instead?

hughsie, 2019-09-25:
I get asked about NUC firmware on a weekly basis. I talked to Intel recently about it and there's some appetite for supporting these but quite a few internal challenges to actually do it. I'll ping my Intel contact about it again and see if we can get some kind of official statement.

superm1, 2019-09-25:
When the file is dropped on the USB stick do you actually have to interact with BIOS setup to install it?  Or is it just "picked up" (and maybe flashed as a capsule already?)

IE if it's just a lack of ESRT support, you could at least use `/usr/libexec/fwupd/fwupdate` to flash the images.

hughsie, 2019-09-25:
I've pinged Brian at Intel about this on Twitter. We talked in person a bit about this at OSFC.io but it just needs a bit more _consumer pressure_ I think.

closed 2019-10-30 by hughsie

hughsie, 2024-08-29:
No, sorry. Intel initially said they would upload firmware for the NUC, and after a lot of legal back and forwards (which we solved) they then decided it wasn't going to happen. For me it was super disappointing and a huge waste of time.