fwupd/fwupd #1405: uefi: Need support to auto-mount partition

superm1, 2019-09-25:
Clear Linux doesn't currently enable fwupd by default because of some problems they encountered a while back.  As part of an effort to re-enable it by default (https://github.com/clearlinux/distribution/issues/792) I noticed that they have a different behavior from most other linux distributions - they don't mount their EFI system partition at runtime other than when copying stuff to it.

This seems like a perfectly good use case and actually not a bad idea outside of clear.  
* `vfat` can get corrupted if not cleanly unmounted
* it's probably a fraction quicker to startup if you don't bother mounting the partition

This matches their design behavior and I think we should update fwupd to support it too.
https://clearlinux.org/news-blogs/where-etcfstab-clear-linux

Basically if our runtime heuristics discover ESP isn't mounted, we should look for `c12a7328-f81f-11d2-ba4b-00a0c93ec93b` to discover it's location.  Then when flashing firmware mount it momentarily and unmount it when done.
https://www.freedesktop.org/wiki/Specifications/DiscoverablePartitionsSpec/

superm1, 2019-09-25:
I was thinking and something else this "potentially" opens up is being able to install UEFI firmware updates that come from LVFS from live media - staging using the ESP already on the disk.

superm1, 2019-09-25:
Some more thoughts on this:
* Using udisks dbus interface is the way to go for that.  Loop the block devices and there is a property `Type` that shows the right value.
* If multiple ESPs are found try to match the GUID used in the EFI variable pointed to by `BootCurrent` to the value `UUID` (seems to match up for me)
* If multiple ESPs are found and it can't match (say live media), try to guess from one of the `BootXXXX` entries, or show an error

hughsie, 2019-09-25:
Not against it, but not sure I want to write the code... Also, is multiple ESPs common? I'd err on multiple -> error message myself.

superm1, 2019-09-25:
That sure would make it a lot simpler.  I think what clear has done has a lot of value, once we have something like this in fwupd (even if you or me don't write it) I think we should push for more distros to adopt it.  There really is not a lot of reason to mount the ESP (really I like the whole idea of no fstab at installation too, but that's another topic)

commit 7e818575 (2019-09-26): uefi: detect unmounted ESP partitions (Fixes: #1405)

Mount these while writing out updates.
Leave them mounted when done (you gotta reboot anyway).
commit 48d7f3db (2019-09-26): uefi: detect unmounted ESP partitions (Fixes: #1405)

Mount these while writing out updates.
Leave them mounted when done (you gotta reboot anyway).
commit 4062a356 (2019-09-26): uefi: detect unmounted ESP partitions (Fixes: #1405)

Mount these while writing out updates.
Leave them mounted when done (you gotta reboot anyway).
commit 5bd649ef (2019-09-26): uefi: detect unmounted ESP partitions (Fixes: #1405)

Mount these while writing out updates.
Leave them mounted when done (you gotta reboot anyway).
closed 2019-09-26 by superm1