fwupd/fwupd #1408: fwupd without consolekit/systemd ?

joakim-tjernlund, 2019-09-26:
Is there any way one could build and use fwupd without consolekit/systemd ?
Our use is a PXE booted linux installer in which I would like to manually run fwupd
to update BIOS etc. before installing out Linux img
This installer does not have systemd/consolekit as it supposed to be small and lean.

hughsie, 2019-09-26:
I think in theory you can compile without systemd and consolekit, although I don't think anyone has tested it and it's not being run as a CI target. Could you give it a go?

superm1, 2019-09-26:
Which plugin will you be updating the BIOS with?  Keep in mind the way the UEFI plugin works it will need to have a stateful location like an EFI system partition to the store the payload for access from a different EFI bootloader (fwupdx64.efi).

You will also need to then PXE boot fwupdx64.efi after you have staged the files.

joakim-tjernlund, 2019-09-26:
> I think in theory you can compile without systemd and consolekit, although I don't think anyone has tested it and it's not being run as a CI target. Could you give it a go?

Did a quick try and got suck on vala, libgudev, dev-libs/libgusb etc.
Seems like I have to install quite a lot more ...  also fwupd  wants polkit too? polkit wants systemd ...

joakim-tjernlund, 2019-09-26:
> Which plugin will you be updating the BIOS with? Keep in mind the way the UEFI plugin works it will need to have a stateful location like an EFI system partition to the store the payload for access [...]

I have PXE boot with both BIOS and UEFI. Any user needing new  FW will have to run
UEFI. But I didn't think of the EFI system partition so maybe this idea isn't going to work.

superm1, 2019-09-26:
> For your purpose you should not use fwupd daemon, fwupdtool is plenty for you. It doesn't need polkit.

Specifically `-Dbuild=standalone`

superm1, 2019-09-26:
I've got another idea for you that might help with the stateful partition problem. This is an installer you say right? So it will be making an ESP for you?

You can just mount it in the right place after install is "done" and use fwupdtool at that point. The first boot after you're all done will be a firmware update then before you boot into the actual OS you installed.

joakim-tjernlund, 2019-09-26:
These are good ideas, need to spend some time to figure out what to do.
Thanks!

superm1, 2019-09-26:
Cool.  If you come up with something workable, feel free to post something to the wiki about it for others.

closed 2019-09-26 by superm1

ckrzen, 2020-02-07:
Success! No ``systemd`` or ``elogind`` using a ``build=all``[default] of ``fwupd`` on CRUX Linux 3.5.

```
# Description: A simple daemon to allow session software to update firmware
# URL: https://github.com/fwupd/fwupd
# Maintainer:
# Packager: Chris Rainey, ckrzen at tuta dot io
# Depends on: valgrind libxmlb git libsoup vala help2man bash_completion tpm2-tss libsmbios libgusb gcab autoconf-archive

name=fwupd
version=1.3.7
release=1
source=(https://github.com/$name/$name/archive/$version.tar.gz)

build() 
{
        cd $name-$version

                meson build \
                -Dlibexecdir=/usr/lib/ \
                -Dprefix=/usr/ \
                -Ddocs=false \
                -Dnls=false \
                -Dgtkdoc=false \
                -Dstemmer=true \
                -Dvapi=false \
                -Dconsolekit=false \
                -Dsystemd=false \
                -Dgcab:docs=false \
                -Dgcab:vapi=false \
                -Dgusb:docs=false \
                -Dgusb:vapi=false

                ninja -C build

                DESTDIR=$PKG ninja -C build install
}
```

[image]

jedahan, 2020-09-11:
does this still work on 1.4.6?

superm1, 2020-09-11:
I don't see why it wouldn't.  If you're having problems, please open up a new issue with all the details.