fwupd/fwupd #1409: get-updates produces perplexing output

LinuxOnTheDesktop, 2019-09-26:
$  fwupdmgr --version
client version:	1.3.3
checkout info:	1.3.2-5-g506ef63a
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.3

fwupd installed by snap, on Linux Mint 19.2 x64 Cinnamon.

I had to restore my system partition from backup. Then I ran fwup. Here is what I did and what happened.
```

$ fwupdmgr get-updates
internal error, please report: running "fwupd.fwupdmgr" failed: cannot find installed snap "fwupd" at revision 1088: missing file /snap/fwupd/1088/meta/snap.yaml
nicholas@x1 ~ $ snap list
Name   Version  Rev   Tracking  Publisher       Notes
core   16-2.41  7713  stable    canonical✓      core
fwupd  1.3.1    1023  stable    richard-hughes  classic
nicholas@x1 ~ $ snap remove fwupd
fwupd removed
nicholas@x1 ~ $ snap install fwupd --classic
fwupd 1.3.2-5-g506ef63a from Richard Hughes installed
nicholas@x1 ~ $ fwupdmgr get-updates
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

No upgrades for System Firmware, current is 0.1.41: 0.1.41=same, 0.1.40=older, 0.1.38=older, 0.1.37=older, 0.1.36=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.15=older, 0.1.13=older, 0.1.12=older
20KHCTO1WW
│
└─Unifying Receiver:
  │   Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR24.06_B0030
  │   Bootloader Version:  BOT03.01_B0008
  │   Vendor:              Logitech (USB:0x046D)
  │   Install Duration:    30 seconds
  │   Flags:               updatable|supported|registered
  │   GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  │                        279ed287-3607-549e-bacc-f873bb9838c4
  │ 
  └─Unifying Receiver Device Update:
        New version:       RQR24.10_B0036
        Remote ID:         lvfs
        Summary:           Firmware for the Logitech Unifying Receiver (RQR24.xx)
        License:           LicenseRef-proprietary
        Size:              74.9 kB
        Vendor:            Logitech
        Duration:          30 seconds
        Flags:             is-upgrade
        Description:       This release addresses an encryption-key extraction vulnerability.
        
         • Wireless Mouse M335
         • Zone Touch Mouse T400
         • Wireless Mouse M545
         • Wireless Mouse M560
         • Touch Mouse M600
         • Touch Mouse T620
         • Wireless Rechargeable Touchpad T650
      
nicholas@x1 ~ $ fwupdmgr get-updates
No upgrades for System Firmware, current is 0.1.41: 0.1.41=same, 0.1.40=older, 0.1.38=older, 0.1.37=older, 0.1.36=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.15=older, 0.1.13=older, 0.1.12=older
20KHCTO1WW
│
└─Unifying Receiver:
  │   Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR24.06_B0030
  │   Bootloader Version:  BOT03.01_B0008
  │   Vendor:              Logitech (USB:0x046D)
  │   Install Duration:    30 seconds
  │   Flags:               updatable|supported|registered
  │   GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  │                        279ed287-3607-549e-bacc-f873bb9838c4
  │ 
  └─Unifying Receiver Device Update:
        New version:       RQR24.10_B0036
        Remote ID:         lvfs
        Summary:           Firmware for the Logitech Unifying Receiver (RQR24.xx)
        License:           LicenseRef-proprietary
        Size:              74.9 kB
        Vendor:            Logitech
        Duration:          30 seconds
        Flags:             is-upgrade
        Description:       This release addresses an encryption-key extraction vulnerability.
        
         • Wireless Mouse M335
         • Zone Touch Mouse T400
         • Wireless Mouse M545
         • Wireless Mouse M560
         • Touch Mouse M600
         • Touch Mouse T620
         • Wireless Rechargeable Touchpad T650
```

I can understand that there were problems after I restored my system partition. But that last bit of output, what does that mean? That an update is available? Why can't your program be clear? Your program lists components for which, it seems to say, there are no updates. Then it (your program) draws a line, seemingly, from one of those components to - what? another component? - and seems to say that there is an upgrade available for that component?

hughsie, 2019-09-27:
> Some sort of 'update successful' message would be good

I've heard other people say that too. I'll work on that now.

superm1, 2019-09-27:
> I can understand that there were problems after I restored my system partition. But that last bit of output, what does that mean? That an update is available? Why can't your program be clear? Your [...]

When you run `get-updates` you end up with:
1.  (on stderr) A list of devices that don't have updates
2. A tree view of devices that do have updates with a list of the updates that apply to them as they're connected to your system.

So you see:
```
Computer at the top
|
Device
|- Firmware release
```

From that you can see your current firmware on the device, the firmware you're installing, installation time, all the relevant details that can be exposed.

The SMBIOS product name chosen by Lenovo isn't exactly a friendly string describing the machine, but on Dell it's clearer.
````
XPS 13 9380
|
Logitech device
|- Firmware release
```

The tree view is new to 1.3.2.  The idea is supposed to be that when you have a variety of updates you can show logical information of the firmware release and map it back to a device.  

> I've heard other people say that too. I'll work on that now.

the next release will add those success messages as you can probably tell from the above commits.

superm1, 2019-10-16:
Since we added completion messages to 1.3.3 i'm going to close this issue.  If you have separate recommendations to make the tree output clearer happy to listen though.

closed 2019-10-16 by superm1