fwupd/fwupd #1410: No supported devices found

MountainX, 2019-09-27:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.2
```

```shell
$ fwupdmgr get-devices
20KHCTO1WW
│
├─X1 Carbon Thunderbolt Controller:
│     Device ID:           af0931ae5a019de856a5c4fc815e6928d5616dc7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:               xxxx  ← TBT-01091706
│   
└─Samsung SSD 970 EVO 2TB:
      Device ID:           xxxx
      Summary:             NVM Express Solid State Drive
      Current version:     1B2QEXE7
      Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
      Serial Number:       xxxxxxx
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               xxx ← NVME\VEN_144D&DEV_A808
                           xxxx ← Samsung SSD 970 EVO 2TB

```

```shell
$ efibootmgr -v
BootCurrent: 001A
Timeout: 0 seconds
BootOrder: 001C,0018,0019,001A,0000,001B,001D,001E,001F,0024
Boot0000* Windows Boot Manager  HD(xxxxxxx)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...B................
Boot0010  Setup FvFile(xxxxxxx)
Boot0011  Boot Menu     FvFile(xxxxxxx)
Boot0012  Diagnostic Splash Screen      FvFile(xxxxxxx)
Boot0013  Lenovo Diagnostics    FvFile(xxxxxxx)
Boot0014  Regulatory Information        FvFile(xxxxxxx)
Boot0015  Startup Interrupt Menu        FvFile(xxxxxxx)
Boot0016  Rescue and Recovery   FvFile(xxxxxxx)
Boot0017  MEBx Hot Key  FvFile(xxxxxxx)
Boot0018* USB CD        VenMsg(xxxxxxx)
Boot0019* USB FDD       VenMsg(xxxxxxx)
Boot001A* NVMe0 VenMsg(xxxxxxx)
Boot001B* ATA HDD0      VenMsg(xxxxxxx)
Boot001C* USB HDD       VenMsg(xxxxxxx)
Boot001D  PCI LAN       VenMsg(xxxxxxx)
Boot001E  Other CD      VenMsg(xxxxxxx)
Boot001F  Other HDD     VenMsg(xxxxxxx)
Boot0020* USBR BOOT CDROM       PciRoot(xxxxxxx)
Boot0021* USBR BOOT Floppy      PciRoot(xxxxxxx)
Boot0022* ATA HDD       VenMsg(xxxxxxx)
Boot0023* ATAPI CD      VenMsg(xxxxxxx)
Boot0024* PCI LAN       VenMsg(xxxxxxx)

```

```shell
$ efivar -l | grep fw
EMPTY (no output)
echo $?
1
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-fallback.conf
│   │   └── arch-lts.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts

```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd: `pacman`
- Have you tried rebooting? YES
- Are you using an NVMe disk? YES
- Is secure boot enabled (only for the UEFI plugin)? NO

Here are my problems:

```shell
fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

fwupdmgr get-updates
No updatable devices
```
Since that did not work, I next tried this method:

```shell
wget https://fwupd.org/downloads/c04107f51e405d4921b2c933d5305c0adbca7116-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.41.cab
fwupdmgr install c04107f51e405d4921b2c933d5305c0adbca7116-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.41.cab
Decompressing…           [***************************************]
No supported devices found
```
I have the following BIOS settings:
Flash BIOS Updating by End-Users [Enabled]
Secure RollBack Prevention [Enabled]
Windows UEFI Firmware Update [Enabled]

hughsie, 2019-09-27:
What does `fwupdmgr get-devices --show-all-devices` say?

exploide, 2019-09-28:
Think I see the same here, just that it's not the X1 but the T490 for me.

I'm using fwupd on Fedora 30.

```
> fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```
> fwupdmgr get-devices
T490/T590/P53s/P43s Thunderbolt Controller
  DeviceId:             934297aea936899e588ad4bf075cc02ee5b1272b
  Guid:                 52265728-359a-574c-9a6a-a23d3b1f952d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              18.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-28

KXG6AZNV512G TOSHIBA
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 94eae9cd-5ee6-5b80-9a3e-ffa2db88cd87
  Guid:                 7907bb58-84c2-5473-bcca-255d8f33a347
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              5105AGLA
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-28
```

```
> fwupdmgr get-devices --show-all-devices
T490/T590/P53s/P43s Thunderbolt Controller
  DeviceId:             934297aea936899e588ad4bf075cc02ee5b1272b
  Guid:                 52265728-359a-574c-9a6a-a23d3b1f952d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              18.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-28

UHD Graphics 620 (Whiskey Lake)
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 52aa8ac7-c837-561f-af32-b7979cd2f6ed
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2019-09-28

KXG6AZNV512G TOSHIBA
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 94eae9cd-5ee6-5b80-9a3e-ffa2db88cd87
  Guid:                 7907bb58-84c2-5473-bcca-255d8f33a347
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              5105AGLA
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-28
```

Despite the LVFS lists a few firmwares for the T490 in stable state.

superm1, 2019-10-03:
This looks firmly in the camp of firmware isn't populating ESRT to me.  I would try to reset BIOS defaults and then turn back on the related options for capsule updates.