fwupd/fwupd #1414: thunderbolt-self-test failing

lovesegfault, 2019-09-28:
I'm trying to package fwupd 1.3.2 for Gentoo, but running into the following issue during tests:
```
meson test -C /var/tmp/portage/sys-apps/fwupd-1.3.2/work/fwupd-1.3.2-build --num-processes 12
ninja: Entering directory `/var/tmp/portage/sys-apps/fwupd-1.3.2/work/fwupd-1.3.2-build'
ninja: no work to do.
 1/13 fwupd-exported-api                      OK       0.01 s
 2/13 fwupd-self-test                         OK       0.04 s
 3/13 fu-self-test                            OK       7.07 s
 4/13 ata-self-test                           OK       0.02 s
 5/13 dfu-self-test                           OK       0.03 s
 6/13 nitrokey-self-test                      OK       0.02 s
 7/13 synaptics-prometheus-self-test          OK       0.04 s
 8/13 udev-self-test                          OK       0.02 s
 9/13 unifying-self-test                      OK       0.04 s
10/13 wacom-usb-self-test                     SKIP     0.03 s
11/13 nvme-self-test                          OK       0.02 s
12/13 thunderbolt-self-test                   TIMEOUT 120.01 s
13/13 uefi-self-test                          OK       0.05 s

Ok:                   11
Expected Fail:         0
Fail:                  0
Unexpected Pass:       0
Skipped:               1
Timeout:               1
```
Here's the full log for the test: [gist](https://gist.github.com/a8467e3e8986916d2354ebfc1fdc5b18)

hughsie, 2019-09-28:
If you run the failing test manually what do you get? If nothing, can you strace it please.

lovesegfault, 2019-09-28:
Alright, I can't reproduce the failure outside of Portage's env; could this be due to portage using cgroups/namespaces for building?

lovesegfault, 2019-09-28:
I'm thinking that might preclude `umockdev` from working.

hughsie, 2019-09-28:
Yup, failing umockdev would certainly break the tests.

lovesegfault, 2019-09-28:
Ah, well, is there a simple way to remove/disable that particular test, @hughsie?

Right now I just restricted tests as a whole, which isn't great :)

FFY00, 2019-09-28:
Just sed it out, like I do in Archlinux.

https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/fwupd#n37

lovesegfault, 2019-09-28:
Awesome, got it working :)

Grato, @FFY00 :)

closed 2019-09-28 by lovesegfault

superm1, 2019-09-28:
Is it possible to detect cgroups/namespaces? A cleaner way would be to skip tests that use umockdev when those are in use.

We do stuff to detect build environments that will cause failures already elsewhere and skip tests sometimes.