fwupd/fwupd #1418: (Optionally) remember removable devices

cassidyjames, 2019-09-29:
I have an 8bitdo controller that get updated via fwupd, but I don't frequently connect it to the computer via USB—I use Bluetooth and I use it more frequently with other devices/OSes.

However, once fwupd has detected that device once, it would be handy to be (maybe optionally) informed of new firmware, so I know to pull out the USB cable and actually connect things that way to get the latest firmware.

I could see this also being useful for devices like Wacom tablets or even Logitech devices that support both the unifying receiver and Bluetooth, as the mouse might usually be connected via Bluetooth but the user would want to keep the most secure firmware on the USB dongle as well.

hughsie, 2019-09-30:
I did ponder this, and there's even skeleton support for it in the code. What I'm not sure about is how you'd do the UX to say "this device is no longer in my possession or is lost" -- and also "please find the device as we think it _might_ need an update"

superm1, 2019-09-30:
I think you would have to bury it deeper in the options. Something like open GF(U) pick your device and check a box to notify for updates even if not plugged in.

But of course the problem then is GF(U) needs to provide those types of update notifications which gets confusing as GS shows "other" update notifications.

cassidyjames, 2019-09-30:
@hughsie that's fair. I think the former is pretty easy UX-wise: any time there's an update for a removable device, you could have a switch to “:ballot_box_with_check: Notify of future updates even when disconnected” or something. I'm not sure how that translates to the CLI, but I can picture it in GNOME Software or another firmware updating GUI. Then if an update comes through and the user no longer has it, they can just uncheck that and it won't bother them again—unless/until the device has been reconnected.

The latter is trickier; I think you could phrase it like, "New firmware since the last update" though. I wonder how annoying that would be if someone does use a device with multiple machines, though, and thus gets essentially double the notifications.

closed 2020-12-08 by superm1