fwupd/fwupd #1421: Cannot update new dell 7390's firmware

fredoche, 2019-09-30:
Hi,

I just got the latest dell xps 13 developper edition 7390 with the core i7 10th gen cpu -not 2in1) . I removed Ubuntu and installed fedora 31.

I cannot update its firmware with gnome-software. After clicking update, and rebooting, supposedly to run the update program, it show four lines of logs in the upper left of the screen, too fast to be read (i think it says something about a path being added), and then reboot again, this time normally and goes back to GDM and the usual user session.

```shell
$ fwupdmgr --version
fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11

```
```shell
$ fwupdmgr get-devices
XPS 7390 Thunderbolt Controller
  DeviceId:             bdf5329a388f2b37c11ab55433343b83ee6fbd6c
  Guid:                 7d538854-204d-51b2-8f9d-1fe881c70200
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              41.02
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-09-30

XPS 13 7390 System Firmware
  DeviceId:             b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  Guid:                 cbe49cca-492b-93e8-b0f4-f693501f271b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.1.0.0
  VersionLowest:        0.1.0.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-09-30
  UpdateState:          success

Micron 2200S NVMe 512GB
  DeviceId:             c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
  Guid:                 b29d3ccf-500b-5b69-9bb9-e46b4832225f
  Guid:                 d63070ee-eeed-11e8-8eb2-f2801f1b9fd1
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot|no-auto-instance-ids
  Vendor:               Micron Technology Inc
  VendorId:             NVME:0x1344
  Version:              22001010
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-30

```

```shell
$ efibootmgr -v
BootCurrent: 0005
Timeout: 0 seconds
BootOrder: 0005,0001,0002,0000
Boot0000* UEFI Micron 2200S NVMe 512GB 193022D9A71A 1	PciRoot(0x0)/Pci(0x1d,0x4)/Pci(0x0,0x0)/NVMe(0x1,00-A0-75-01-22-D9-A7-1A)/HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0001* ubuntu	HD(1,GPT,f4874422-1a07-42dd-a9a6-4dfd6b68a838,0x800,0x186000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux Firmware Updater	HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0005* Fedora	HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)

```

```shell
$ efivar -l | grep fw
** nothing **
```

```shell
$ tree /boot
/boot
├── config-5.3.0-0.rc6.git0.1.fc31.x86_64
├── config-5.3.1-300.fc31.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-01d1fd354a0e48b7ac999d422c01d3a1.img
├── initramfs-5.3.0-0.rc6.git0.1.fc31.x86_64.img
├── initramfs-5.3.1-300.fc31.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-5.3.0-0.rc6.git0.1.fc31.x86_64
├── System.map-5.3.1-300.fc31.x86_64
├── vmlinuz-0-rescue-01d1fd354a0e48b7ac999d422c01d3a1
├── vmlinuz-5.3.0-0.rc6.git0.1.fc31.x86_64
└── vmlinuz-5.3.1-300.fc31.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 31 beta
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): its not installed, but fwupdmgr is
- Have you tried rebooting? yes 6 or 7 times
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)?
 yes
 mokutil --sb-state
SecureBoot enabled

superm1, 2019-10-01:
You didn't even see a Dell logo pop up with a progress bar?

My _gut_ feeling is secure boot causing problems, but we need to get some debug logging turned on to see what's going on.

1. Run `/usr/libexec/fwupd/fwupdate -d` to turn on `fwupdx64.efi` application debugging
2. Re-run your update as normal
3. Run `/usr/libexec/fwupd/fwupdate -L` to look at the debugging logs captured from last attempt

If there are no logs, please try turning off secure boot and try again.

When complete you should run `/usr/libexec/fwupd/fuwpdate -D` to disable EFI application debugging again.

fredoche, 2019-10-02:
Yes, I do see the dell logo. Logs are printing on overlay.

``
/usr/libexec/fwupd/fwupdate -L
This program may only work correctly as root
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b-0../plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)/\EFI\fedora\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x0100000000A0750122D9A71A84012C00)/HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E) and PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x0100000000A0750122D9A71A84012C00)/HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E) and Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x0100000000A0750122D9A71A84012C00)/HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E) and Pci(0x0,0x0)/Path(131,23,0x0100000000A0750122D9A71A84012C00)/HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E) and Path(131,23,0x0100000000A0750122D9A71A84012C00)/HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E) and HD(Part1,Sig9183E0E0-7491-40F7-B911-763ACFD8758E)../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 8897055 bytes../plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: CBE49CCA-492B-93E8-B0F4-F693501F271B../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: CBE49CCA-492B-93E8-B0F4-F693501F271B../plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 1../plugins/uefi/efi/fwupdate.c:596:fwup_apply_capsules(): QueryCapsuleCapabilities: Success max: 104857600 reset:1../plugins/uefi/efi/fwupdate.c:598:fwup_apply_capsules(): Capsules: 1[fred@localhost ~]

superm1, 2019-10-02:
Hmm, that's odd - everything looks good there and all this stuff has been rock solid for a while.
You get all the way up to the point of calling the UEFI RunTimeService `UpdateCapsule()`.  After that - just power cycles?  You had AC adapter plugged in for the duration of the update attempt?

I just updated a 7390 myself with master recently.  I don't think there is anything notably that has changed related to this stuff in master though.

@hughsie  any ideas?  I guess I can downgrade back to 1.0.0 firmware and see if I can repro on 1.2.10.

hughsie, 2019-10-03:
The log looks like it got all the way to RT->UpdateCapsule which is where we either hard over to the capsule or fail and return. Because there is no `Reset System` in the logs we have to conclude it ran the capsule and failed in the firmware update itself I think.

superm1, 2019-10-03:
Well just tried with master and the new reinstall command w/ 1.3.3 on top of itself and no problem.  I'll have to get a copy of 1.0.0 to downgrade and try to repro again.

As an extra data point - please try turning off secure boot and issuing the upgrade again.  It's a notable difference between our setups (other than mine is on Ubuntu and master).

fredoche, 2019-10-03:
Hi, Thanks for the help.
I tried just now and the update went fine with SB disabled.

superm1, 2019-10-04:
Well from a fwupd perspective that's good to hear, but I suppose there is still a bug here somewhere.  From the new firmware could you try to issue a reinstall with secure boot enabled?  If that issue still happens on the new FW I think this needs investigation by the firmware team.

superm1, 2019-10-30:
#1421 is a duplicate of #1504 (although I agree #1421 came first - #1504 has more information)

superm1, 2019-11-13:
I've reproduced this on Ubuntu's fwupd 1.2.10 as well now with secure boot enabled.  So the Fedora 31 issue is the same as Ubuntu issue.  The Fedora 30 issue is a signing issue for the bootloader.

On Ubuntu the failure shows up pretty much the same, I see same logs except that I have BGRT enabled so I see the UX capsule attempt to load too.  Trying to disable with a hand crafted `no-ux-capsule` quirk doesn't resolve it, so it's not directly related to the use of the ux capsule.
```
..
/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule..
/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b-0..
/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule..
/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)/\EFI\ubuntu\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap..
..
./plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\ubuntu\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap..
/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device..
/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume..
/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file..
/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 57218 bytes..
/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697..
/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697..
/plugins/uefi/efi/fwupdate.c:564:fwup_add_update_capsule(): Checking GOP for ux capsule..
/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule..
/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)/\EFI\ubuntu\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap..
..
./plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Pci(0x0,0x0)/Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and Path(131,23,0x01000000ACE42E00901E31DC84012C00)/HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3) and HD(Part1,SigE5EB007D-B3BD-427A-8EA1-57636CAC6CB3)..
/plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\ubuntu\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap..
/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device..
/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume..
/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file..
/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 8900369 bytes..
/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: CBE49CCA-492B-93E8-B0F4-F693501F271B..
/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: CBE49CCA-492B-93E8-B0F4-F693501F271B..
/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 2..
/plugins/uefi/efi/fwupdate.c:596:fwup_apply_capsules(): QueryCapsuleCapabilities: Success max: 104857600 reset:1..
/plugins/uefi/efi/fwupdate.c:598:fwup_apply_capsules(): Capsules: 2
```

I'm thinking this looks like a BIOS issue.  I'll have some internal discussion about it.

fredoche, 2019-11-26:
Hi,
I see that there is another BIOS update from Dell, yet I stille hit the same bug with secure boot activated. Do you have additional details on what's going on?
Cheers

superm1, 2019-12-02:
Sorry nothing to share about this yet.  I do still agree it looks like a BIOS issue though with secure boot on this system.

arnobl, 2019-12-10:
I add the same issue. As a workaround, I overcome the problem by disabling temporarily the "secure boot" option in the bios.

f0cus10, 2019-12-13:
Running: Dell XPS 13 7390 (non-Developer Edition, non-2-in-1) Ubuntu 19.10 (Eoan Ermine). 

Reproduced this issue (1.2.0 -> 1.3.1). Disabling Secure Boot fixes the problem.

mitar, 2020-01-16:
Still happened to me just yesterday.

fredoche, 2020-01-17:
@superm1 Hi, you mentionned a possible "internal discussion about it." Were you able to get some pieces of information, is there anything we can do to help?

superm1, 2020-01-17:
I've passed the relevant information to the right people within Dell. For now I don't think there is anything else needed, thanks.

tsarajar, 2020-02-10:
The BIOS update software should mention about this secure boot thing. Took me an hour to figure out why the updates don't trigger even though the software gets uploaded and the updater requests a reboot

superm1, 2020-02-10:
Well this is firmly in a BIOS bug camp, but given the current situation, I do wonder if it's worth quirking the failure mode (IE if the update wasn't applied after reboot and it's on this particular system and secure boot is on notifying that there is a BIOS bug).

superm1, 2020-02-10:
@hughsie thoughts?

hughsie, 2020-02-10:
> and it's on this particular system

Sounds like it's what known issues was invented for. @tsarajar can you tell us what the report contains from `fwupdmgr report-history` please.Please don't upload the report yet, let us create the rule on the LVFS first.

superm1, 2020-02-10:
Now something extra confusing to me is that there are no failed reports on LVFS, and at least 6 that have secure boot enabled..

tymokvo, 2020-02-13:
Hi, experiencing this on Fedora 31 with secure boot disabled. Is there a known workaround that is documented somewhere?

superm1, 2020-02-14:
Different issue with fedora 31. You're running fwupd 1.3.7 right? There was a bug in UEFI capsule updates. It's fixed in fwupd 1.3.8

tymokvo, 2020-02-14:
> Different issue with fedora 31. You're running fwupd 1.3.7 right? There was a bug in UEFI capsule updates. It's fixed in fwupd 1.3.8

Yep, version 1.3.7. So nothing to do till 1.3.8 lands in package repos?

hughsie, 2020-02-14:
It's already in updates-testing. `sudo dnf update fwupd -y --enablerepo updates-testing`

tymokvo, 2020-02-15:
Worked! Thanks so much for the help and the work on this! :smile_cat:

superm1, 2020-03-25:
[This is still an issue w/ current Dell BIOS on XPS 7390]

closed 2020-04-01 by stale[bot]