fwupd/fwupd #1422: thunderbolt: Restrict NVM upgrades to Intel firmware

superm1, 2019-10-01:
The first RFC set for USB4 was posted to kernel mailing lists today.  One notable thing is in patch 17 (https://patchwork.kernel.org/patch/11168517/) that NVM upgrades are only allowed on Intel hardware for now until the format is documented for other vendors.

From a fwupd perspective we should decide how to message this and what to do when a non-Intel USB4 controller is encountered.

Of course the kernel will block us either way with the current, but we can at least build a nice `UpdateError` around it not yet being supported by the kernel.

superm1, 2019-10-01:
CC @YehezkelShB and @gicmo

superm1, 2019-10-01:
As I can tell, the udev attributes are not currently exported for the switch config vendor (the vendor attribute represents the vendor selling the device not the SOC vendor).

I've added some feedback into the RFC series along that line.  If that attribute is exported my thought process is:

1. Check sysfs attribute `switch_vendor != intel` then mark as not updatable with a reason that updates are only supported on Intel right now.
2. Later when other vendors are added, relax that check (probably a whitelist of supported vendors) and move checks into NVM parsing to parse the format of any other vendors that are supported.

superm1, 2019-10-06:
There has been some more discussion on the kernel thread and it's currently trending to have a generation attribute and potentially a hw_vendor and hw_product attribute that could read the switch data.

So the idea would be:
```
if (generation >= 4)
    // check the hw_vendor and hw_product attributes against a whitelist of known NVM formats
    // as new NVM formats are supported expand this whitelist
else
    // all code that exists today (read from native bit within the NVM during probe time, everything else udev)
```

hughsie, 2019-10-06:
Couldn't we do more in the kernel with current generations? the NVM read stuff is slooow right now.

superm1, 2019-10-06:
The only thing read at probe time in current generation by userland is the native bit.  And even then it's only reading enough of the NVM to get that bit.  I do think it's certainly "possible" to move that to kernel, just would need patches and discussion.

YehezkelShB, 2019-10-06:
Have we profiled it to see what exactly takes the time? The kernel reads the whole NVM in 2 reads and we read it to memory in one go (but maybe GTK does it in multiple reads inside). Is it the system call boundaries crossing that takes the time? I'd bet it isn't the in-memory jumps we do to find the relevant information after reading the NVM to memory, but of course I may be wrong.

superm1, 2019-10-23:
For those not following along; the non-RFC patch series was submitted today (https://patchwork.kernel.org/cover/11206421/)

Still no plans for attributes, quoting the cover letter:
>I did not yet add the hw_vendor_id and hw_product_id attributes because I
clocked that 'dd if=nvm_activeX/nvmem of=/dev/null bs=64 count=2' (that's
what fwupd does) takes typically less than 10 ms when the controller is
powered on. That should not slow down fwupd. We can add them later if it is
really a problem (I'm just trying to avoid adding too much attributes that
we need to maintain forever).

superm1, 2019-10-28:
I'm pretty sure that it's related to the controller going into RTD3 from inactivity making the whole thing slow.  I don't think we're going to get much better than this without attributes.

I think that basically means at coldplug not much we can do to check if we can really flash non-Intel.  
* A non-Intel controller will fail from the kernel (it won't export the attributes with the stuff going in initially) - so fwupd will show a missing NVM error in `UpdateError`
* If the kernel grows support for this NVM image format, fwupd will enumerate it.
* If fwupd tries to flash it, I expect the image parsing stuff will fall apart somewhere and it will fail trying to update.

I guess lets just revisit this when non-Intel controller flashing support lands in the kernel and we have a new USB4 image format to recognize.

closed 2019-10-28 by superm1