fwupd/fwupd #1427: XPS 9380 touchpad sometimes stops working when probing :/

superm1, 2019-10-02:
On some runs i'm noticing the touchpad probes properly and some it doesn't.

Good run:
```
$ sudo /usr/local/libexec/fwupd/fwupdtool get-devices  --plugin-whitelist=synaptics_rmi  --plugin-verbose=synaptics_rmi 
Loading…                 [-                                      ]16:08:49:0190 FuEngine             failed to find specified plugin dell: no plugin dell found
16:08:49:0191 FuEngine             failed to find specified plugin dell_dock: no plugin dell_dock found
16:08:49:0191 FuEngine             failed to find specified plugin dell_dock: no plugin dell_dock found
16:08:50:0257 FuEngine             device ID 6c24a747f97668873b761558e322398a91dbf394 was not found
Loading…                 [***************************************]
○
│
└─Touchpad:
      Device ID:           9040418c803f8614fb03864a70fa211fab88d2f1
      Current version:     1.5.14522
      Bootloader Version:  54.0
      Vendor:              Dell Inc. (HIDRAW:0x06CB)
      Flags:               updatable|registered
      GUIDs:               424bd00e-9789-5cdf-a12a-3c81bc4676d6 ← HIDRAW\VEN_06CB&DEV_76AF
                           140f4458-951b-5bb9-85e2-879bd5b02615 ← SYNAPTICS_RMI\TM3038-003
    
```
Bad Run:
```
$ sudo /usr/local/libexec/fwupd/fwupdtool get-devices  --plugin-whitelist=synaptics_rmi  --plugin-verbose=synaptics_rmi  --verbose
16:08:54:0346 FuDebug              Verbose debugging enabled (on console 1)
16:08:54:0346 FuDebug              setting FWUPD_SYNAPTICS_RMI_VERBOSE=1
16:08:54:0351 FuMain               Failed top stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
16:08:54:0351 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
16:08:54:0352 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
16:08:54:0352 Fwupd                looking for os-release at /etc/os-release
16:08:54:0353 FuConfig             loading /usr/local/share/fwupd/metainfo
16:08:54:0353 XbSilo               attempting to load /usr/local/var/cache/fwupd/metainfo.xmlb
16:08:54:0353 XbSilo               file: 443d59bd-a7a4-569b-bc0e-ef32f790e454, current:443d59bd-a7a4-569b-bc0e-ef32f790e454, cached: (null)
16:08:54:0353 XbSilo               loading silo with file contents
16:08:54:0353 FuConfig             using config path of /usr/local/etc/fwupd
16:08:54:0353 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
16:08:54:0353 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/lvfs.conf
16:08:54:0353 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
16:08:54:0353 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
16:08:54:0353 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/vendor.conf
16:08:54:0354 FuConfig             loading config from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
16:08:54:0354 FuConfig             using config path of /usr/lib/fwupd
16:08:54:0354 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
16:08:54:0354 FuConfig             ordering lvfs-testing=lvfs+1
16:08:54:0354 FuConfig             ignoring unfound remote fwupd
16:08:54:0354 FuConfig             ignoring unfound remote fwupd
16:08:54:0354 FuConfig             ignoring unfound remote fwupd
16:08:54:0354 FuConfig             ignoring unfound remote fwupd
16:08:54:0362 FuEngine             client certificate exists and working
16:08:54:0362 FuHistory            trying to open database '/usr/local/var/lib/fwupd/pending.db'
16:08:54:0362 FuHistory            got schema version of 5
16:08:54:0362 FuHwids              smbios property Manufacturer=Dell Inc.
16:08:54:0362 FuHwids              smbios property EnclosureKind=a
16:08:54:0362 FuHwids              smbios property Family=XPS
16:08:54:0362 FuHwids              smbios property ProductName=XPS 13 9380
16:08:54:0362 FuHwids              smbios property ProductSku=08AF
16:08:54:0362 FuHwids              smbios property BiosVendor=Dell Inc.
16:08:54:0362 FuHwids              smbios property BiosVersion=1.7.0
16:08:54:0362 FuHwids              smbios property BiosMajorRelease=01
16:08:54:0363 FuHwids              smbios property BiosMinorRelease=07
16:08:54:0363 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
16:08:54:0363 FuHwids              smbios property BaseboardProduct=
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/altos.quirk
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/ata.quirk
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/colorhug.quirk
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/csr-aiaiai.quirk
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/dell-dock.quirk
16:08:54:0363 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/dell.quirk
16:08:54:0364 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d/dfu.quirk
[... 561 lines cut ...]
16:08:54:0900 FuPluginSynapticsRmi ReportRead:     0b 06 00 00 00 00 00 00 01 e0 30 33 48 b6 70 09 88 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
16:08:54:0900 FuPluginSynapticsRmi DeviceRead:     00 00 00 00 00 00 
16:08:54:0900 FuPluginSynapticsRmi DeviceWrite:    09 01 ff 00 71 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
16:08:54:0903 FuPluginSynapticsRmi ReportWrite:    0a 00 e9 71 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
16:08:54:0903 FuPluginSynapticsRmi ReportRead:     0c 08 01 a4 02 66 02 31 04 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
16:08:54:0903 FuPluginSynapticsRmi DeviceRead:     01 a4 02 66 02 31 04 03 
16:08:54:0904 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0016/hidraw/hidraw0: failed to add device using on synaptics_rmi: PDT entry buffer invalid size 8 != 6
16:08:54:0906 FuEngine             no plugin specified for udev device /sys/devices/pci0000:00/0000:00:1d.0/0000:03:00.0/0000:04:04.0/0000:3a:00.0/0000:3b:02.0/0000:3c:00.0/usb5/5-2/5-2.3/5-2.3.5/5-2.3.5:1.0/0003:413C:B06F.0015/hidraw/hidraw2
16:08:54:0909 FuEngine             no plugin specified for udev device /sys/devices/pci0000:00/0000:00:1d.0/0000:03:00.0/0000:04:04.0/0000:3a:00.0/0000:3b:02.0/0000:3c:00.0/usb5/5-2/5-2.5/5-2.5:1.0/0003:413C:B06E.0014/hidraw/hidraw1
16:08:54:0910 FuCommon             not checking 2.0.6 as no version format set
16:08:54:0910 FuCommon             not checking 2.0.5 as no version format set
16:08:54:0910 FuCommon             not checking 2.0.7 as no version format set
16:08:54:0910 FuCommon             not checking 2.0.6 as no version format set
16:08:54:0910 FuCommon             not checking 0.1.7.0 as no version format set
16:08:54:0910 FuEngine             device ID 6c24a747f97668873b761558e322398a91dbf394 was not found
16:08:54:0910 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
No detected devices
```

After the bad run, the touchpad is left in a bad state - I need to `rmmod i2c-hid` / `modprobe i2c-hid` for it to come back.

I don't know much about what could cause that PDT entry buffer to change sizes between runs.

So maybe we do need quirks? :disappointed: Or maybe just need to reset it in the failure case.

superm1, 2019-10-02:
Ahh... it's only if i'm touching the touchpad while fwupdtool is probing it..

superm1, 2019-10-02:
This recovers it for me:

```
diff --git a/plugins/synaptics-rmi/fu-synaptics-rmi-device.c b/plugins/synaptics-rmi/fu-synaptics-rmi-device.c
index f44f8b4d..d1919876 100644
--- a/plugins/synaptics-rmi/fu-synaptics-rmi-device.c
+++ b/plugins/synaptics-rmi/fu-synaptics-rmi-device.c
@@ -487,8 +487,12 @@ fu_synaptics_rmi_device_setup (FuDevice *device, GError **error)
                return FALSE;
 
        /* read PDT */
-       if (!fu_synaptics_rmi_device_scan_pdt (self, error))
+       if (!fu_synaptics_rmi_device_scan_pdt (self, error)){
+               g_autoptr(GError) error_reset = NULL;
+               if (!fu_synaptics_rmi_device_rebind_driver (self, &error_reset))
+                       g_warning ("%s", error_reset->message);
                return FALSE;
+       }
        priv->f01 = fu_synaptics_rmi_device_get_function (self, 0x01, error);
        if (priv->f01 == NULL)
                return FALSE;
```

hughsie, 2019-10-02:
It's reading an 0xc event in place of a 0xb one. We should certainly check the cmd and if invalid retry the read, or re-request the chunk of it's been replaced. I'll have a go tomorrow, it should be easy to reproduce :)

closed 2019-10-03 by hughsie