fwupd/fwupd #1429: can not update fw

JeremyKim-ss, 2019-10-03:
Hello,
I executed "$ fwupdmgr install aaaa.cab" with fwupd ver 1.3.3
However, message was displayed below.

=============================================
#fwupdmgr install /home/updata/aaaa.cab
Invalid folder index
=============================================

I made cabinet file using makecab on windows10.
In xml file, I cannot find any location information.

Can you tell me what happens to these errors in some cases?
If you need more information, I'll add again.

Thank you!

FFY00, 2019-10-03:
What are the contents of the cab file? Which device are you trying to update?

hughsie, 2019-10-03:
Also, can you tell us the command line of what you used in windows 10 to construct the cab file please. If you can also attach the .cab file to this issue that would be helpful too.

JeremyKim-ss, 2019-10-04:
Thank you for your responses.

I tried to update NVMe SSD device and contents of cab file is firmware binary and xml.
When I make cab file in windows 10, I used below command.

c:\> makecab /F config.txt
Total files:              2
Bytes before:     2,098,010
Bytes after:      1,595,402
After/Before:            76.04% compression
Time:                     1.40 seconds ( 0 hr  0 min  1.40 sec)
Throughput:            1461.37 Kb/second

c:\>type config.txt
firmware.metainfo.xml
firmware.bin

c:\>type firmware.metainfo.xml

<?xml version="1.0" encoding="UTF-8"?>
<component type="firmware">
  <id>com.SEC.guid384e7e9a.firmware</id>
  <name>PM991 NVMe SSD</name>
  <summary>Dell PM991 NVMe SSD Firmware Update</summary>
  <description> Updating the SSD firmware for new version. </description>
  <provides>
    <firmware type="flashed">384e7e9a-43bd-4770-ba77-5d15e94811ac</firmware>
  </provides>
  <url type="homepage">http://support.dell.com</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>proprietary</project_license>
  <updatecontact>None</updatecontact>
  <releases>
    <release urgency="high" version="25304039" timestamp="1551856497.1822608">
      <description>
        Updating the SSD firmware for new version.
      </description>
    </release>
  </releases>
</component>

superm1, 2019-10-04:
in your config.txt please try adding:
```
.OPTION EXPLICIT
.set Cabinet=on
.set Compress=on
.set MaxDiskSize=0
.set DiskDirectoryTemplate=.
.set DestinationDir=DriverPackage
```

JeremyKim-ss, 2019-10-07:
@superm1  Thank you for your response.

Although I changed config.txt file, result was same as before, unfortunately.

`firmware.metainfo.xml
firmware.bin
.OPTION EXPLICIT
.set Cabinet=on
.set Compress=on
.set MaxDiskSize=0
.set DiskDirectoryTemplate=.
.set DestinationDir=DriverPackage`

`D:\>makecab /F config.txt
Cabinet Maker - Lossless Data Compression Tool

2,098,010 bytes in 2 files
Total files:              2
Bytes before:     2,098,010
Bytes after:      1,595,399
After/Before:            76.04% compression
Time:                     2.12 seconds ( 0 hr  0 min  2.12 sec)
Throughput:             968.26 Kb/second`

`hyun@KSHLINUX:~/sharedata/Jeremy$ fwupdmgr install 1.cab
Invalid folder index
`

`hyun@KSHLINUX:~/sharedata/Jeremy$ fwupdmgr --version
client version: 1.2.5
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.5
`
fwupd 1.3.3 also has same result.

I'll attach my cabinet file, soon. 
Thank you in advance.

JeremyKim-ss, 2019-10-07:
> in your config.txt please try adding:
> 
> ```
> .OPTION EXPLICIT
> .set Cabinet=on
> .set Compress=on
> .set MaxDiskSize=0
> .set DiskDirectoryTemplate=.
> .set DestinationDir=DriverPackage
> ```

JeremyKim-ss, 2019-10-07:
Sorry, I reopened issue (my mistake)

hughsie, 2019-10-07:
I've tried myself with makecab on Windows 10. Apart from the "version format unset and version 25304039 ambiguous" warning they seem to decompress fine. with and without the .OPTIONS header. The `Invalid folder index` error comes from inside libgcab, and seems to suggest the archive is somehow invalid. Can you upload it somewhere, or email it to me please?

superm1, 2019-10-07:
Just a thought - could it be you built `config.txt` on Linux but used on Windows, so maybe newline characters are jumbled causing the directives to not be adopted by `makecab.exe`?

JeremyKim-ss, 2019-10-08:
@hughsie @superm1 
Wow, you are right!!
I changed config.txt file like below

'''
C:\>type config.txt
.OPTION EXPLICIT
.set Cabinet=on
.set Compress=on
.set MaxDiskSize=0
.set DiskDirectoryTemplate=.
.set DestinationDir=DriverPackage

"C:\firmware.bin"
"C:\firmware.metainfo.xml"
C:\>
'''

makecab has 1.38MB limitation per a file (I didn't know thatT_T). So, error occurred inside as hughsie said.
And, now fwupd works well!!
I really appreciate your helps.

closed 2019-10-08 by JeremyKim-ss

hughsie, 2019-10-08:
I still think we should document this on the website somewhere. We mention `makecab.exe` in https://www.fwupd.org/lvfs/docs/vendors but don't have any detailed instructions for using it. Perhaps a new documentation page at https://www.fwupd.org/lvfs/docs/archive or something?

JeremyKim-ss, 2019-10-08:
If there is such a document, it will be very helpful. Actually, I was not used to making cabinet files  and there was not much information on website.