fwupd/fwupd #1438: Wrong ME firmware version detected on T480s

der-eismann, 2019-10-07:
Hey guys, I stumbled upon another issue today. I recently updated the ME firmware on my T480s with my Windows partition (since Lenovo seems to be a bit slow with this kind of update) to version `11.8.70.3626`, but `fwupdmgr` still shows the old version `184.65.3590`. BIOS and the Intel tool also show the newer version, so I'm quite sure it was updated correctly.
Could this be because the version format changed?

```
$ sudo ./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2019, Intel Corporation, All rights reserved.

Application Version: 1.0.41.0
Scan date: 2019-10-07 08:03:54 GMT

*** Host Computer Information ***
Name: ${HOSTNAME}
Manufacturer: LENOVO
Model: 20L7001LGE
Processor Name: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
OS Version: Fedora 30 Thirty (5.2.17-200.fc30.x86_64)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 11.8.70.3626
```
```
$ fwupdmgr get-updates
No upgrades for 20L7001LGE System Firmware, current is 0.1.34: 0.1.34=same, 0.1.33=older, 0.1.31=older, 0.1.30=older, 0.1.29=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.16=older, 0.1.13=older
```

Basic issue information:
```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             729dda7cd055f3947a1de93c8f2bbcabd7dfdf94
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              17.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-10-07

20L7001LGE System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.34
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-10-07

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        184.65.3590
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-10-07

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.17
  VersionLowest:        0.1.17
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-10-07

SAMSUNG MZVLB512HAJQ-000L7
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 79d6cfae-a5a2-5936-9248-5aebd23480f7
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              5L2QEXA7
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-10-07
```

Please answer the following questions:

- Operating system and version: Fedora 30
- How did you install fwupd: apt
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2019-10-14:
I think this is because we're treating the uint32 version number as a "triplet" e.g. `1.2.3` rather than one of the ME formats which have interesting encodings: https://github.com/fwupd/fwupd/blob/master/src/fu-common-version.c#L67

We don't get any indicator from the ESRT that the device is a "ME" device and so we fall back to the triplet format by default. Where we know about it beforehand we can tag it so that the device shows the correct value: https://github.com/fwupd/fwupd/blob/master/src/fu-common-version.c#L67

der-eismann, 2019-10-15:
Hey, thanks for your answer! Yes, the version format is an issue, but not the real problem. If I add a quirk by hand and set the version format to `intel-me2` then I get `11.8.65.3590` instead of `184.65.3590` which is fine.

However the real problem is the completely wrong version! According to the Intel tool and the BIOS of the ThinkPad I have `11.8.70.3626` installed and not `11.8.65.3590` as shown by fwupd. Maybe the new firmware has a bug?

hughsie, 2019-10-15:
Hah! That's no intentional. I assume the BIOS and windows tool are actually querying the ME version somehow, rather than trusting the value from the ESRT table.