fwupd/fwupd #1440: Uses SHA-1 for integrity

bk2204, 2019-10-09:
When looking at the hashes used to check the integrity of firmware, all of them appear to be using SHA-1.  In addition, the signatures over the firmware manifests downloaded appear to be using SHA-1 as well.

SHA-1 is considered dangerously weak, and other, better alternatives have been available for some time.  Fortunately, fwupd supports SHA-256 and SHA-512 as well, so it should be easy to switch over.  GnuPG has supported SHA-256 since the CentOS 4 timeframe and SHA-512 since CentOS 5, so updating the signatures is not a problem.

Much like apt, fwupd should stop using or accepting MD5 or SHA-1 both in the manifests and signatures and only accept strong alternatives, such as SHA-2, SHA-3, or BLAKE2.

You can verify the hash algorithms used via the following commands:
- `zcat /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz | grep checksum | perl -pe 's/^.*type="([^"]+)".*$/$1/g' | sort | uniq -c`
- `gpg -v --verify firmware.xml.gz.asc`

This is [Debian bug #942008](https://bugs.debian.org/942008).

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

(The error message is a known issue in Debian 1.3.2-1, but 1.3.2-2 is not available yet because it has not been signed.)  I've omitted items which are not relevant to this issue.

Please answer the following questions:

- Operating system and version: Debian sid/amd64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): From Debian
- Have you tried rebooting? I've rebooted since I first discovered this, but if this is a reboot-related issue, then it should still be fixed.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? Yes.

hughsie, 2019-10-09:
We're only really using the SHA1 checksums for download checking -- all the files from the LVFS are signed with the GPG key too.

bk2204, 2019-10-09:
As far as I can tell, you're using SHA-1 for signatures as well, though, which poses the same problem.

hughsie, 2019-10-09:
Isn't https://security.stackexchange.com/questions/68105/gpg-keys-sha-1 valid here?

bk2204, 2019-10-09:
That talks about key generation and fingerprints.  It's possible to generate two v3 keys with colliding fingerprints, and somewhat more difficult (but not theoretically impossible) to generate two v4 keys with colliding fingerprints (although it's likely one would be very weak).  The limitation there is that both keys must be valid RSA keys (or other types of keys) that are usable.

Here, we're talking about the ability of a user to upload arbitrary files with the same hash, which is much, much easier.  It's totally possible to create empty, unused areas in a firmware file that contain data solely for colliding hashes.  We saw this in the SHAttered attacks with PDFs, for example.

In addition, even if the attack is purely theoretical at this moment, attacks only get better (MD5 is now so weak that [collisions with significant freedom](https://news.ycombinator.com/item?id=13823704) can be generated quickly on a PC) and using weak algorithms poses compliance problems for folks.  For example, US government agencies aren't supposed to use MD5 or SHA-1 if it can be avoided and PCI DSS also requires that strong algorithms be used.

hughsie, 2019-10-09:
> we're talking about the ability of a user to upload arbitrary files with the same hash

I concede that's true. It's mitigated by only allowing authorised vendors to upload firmware, but also could be improved. The `checksum_upload` is currently a `Column(String(40))` but that could be trivially changed to SHA256 length.