fwupd/fwupd #1445: Unable to start systemd service

johansmitsnl, 2019-10-09:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
The command hangs and does not finish
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0005,0002
Boot0000* ubuntu	HD(1,GPT,186a5401-d858-4468-b46c-2143dff7da33,0x800,0x177000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* debian	HD(1,GPT,bdbbb709-6ec0-48ee-9349-ac45cca6f54f,0x800,0x100000)/File(\EFI\debian\grubx64.efi)
Boot0002* Linux Firmware Updater	HD(1,GPT,bdbbb709-6ec0-48ee-9349-ac45cca6f54f,0x800,0x100000)/File(\EFI\debian\fwupdx64.efi)
Boot0006* UEFI: PM981 NVMe Samsung 512GB, Partition 1	HD(1,GPT,186a5401-d858-4468-b46c-2143dff7da33,0x800,0x177000)/File(EFI\boot\bootx64.efi)..BO
```

```shell
$ efivar -l | grep fw
no output
```

```shell
$ tree /boot
/boot
├── config-5.2.0-2-amd64
├── config-5.2.0-3-amd64
├── efi
│   └── EFI
│       ├── debian
│       │   ├── fw
│       │   ├── fwupdx64.efi
│       │   └── grubx64.efi
│       └── Dell
│           ├── Bios
│           │   └── Recovery
│           │       ├── BIOS_CUR.RCV
│           │       └── BIOS_PRE.rcv
│           └── logs
│               ├── diags_current.xml
│               └── diags_previous.xml
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├── ast.mo
│   │   ├── ca.mo
│   │   ├── da.mo
│   │   ├── de_CH.mo
│   │   ├── de@hebrew.mo
│   │   ├── de.mo
│   │   ├── en@arabic.mo
│   │   ├── en@cyrillic.mo
│   │   ├── en@greek.mo
[... 332 lines cut ...]
 Main PID: 22491 (code=exited, status=239/CACHE_DIRECTORY)

okt 09 21:18:30 laptop systemd[1]: Starting Firmware update daemon...
okt 09 21:18:30 laptop systemd[22491]: fwupd.service: Failed to set up special execution directory in /var/cache: Not a directory
okt 09 21:18:30 laptop systemd[22491]: fwupd.service: Failed at step CACHE_DIRECTORY spawning /usr/lib/fwupd/fwupd: Not a directory
okt 09 21:18:30 laptop systemd[1]: fwupd.service: Main process exited, code=exited, status=239/CACHE_DIRECTORY
okt 09 21:18:30 laptop systemd[1]: fwupd.service: Failed with result 'exit-code'.
okt 09 21:18:30 laptop systemd[1]: Failed to start Firmware update daemon.

```

```
$ systemctl status fwupd-refresh.service 
● fwupd-refresh.service - Refresh fwupd metadata and update motd
   Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-10-09 21:16:54 CEST; 6min ago
     Docs: man:fwupdmgr(1)
  Process: 22305 ExecStart=/usr/bin/fwupdmgr refresh (code=exited, status=1/FAILURE)
 Main PID: 22305 (code=exited, status=1/FAILURE)

okt 09 21:16:54 laptop systemd[1]: Starting Refresh fwupd metadata and update motd...
okt 09 21:16:54 laptop systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
okt 09 21:16:54 laptop systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
okt 09 21:16:54 laptop systemd[1]: Failed to start Refresh fwupd metadata and update motd.
```

superm1, 2019-10-09:
This is fixed in 1.3.2-2 which hasn't yet migrated to testing. You can pull it from unstable for now.

closed 2019-10-09 by superm1

johansmitsnl, 2019-10-10:
Thanks for the quick reply, it indeed fixed the issue.

tsarajar, 2020-10-30:
I'm seeing this still in Ubuntu 20.04 with fwupdmgr 1.3.11:

> root@ready-koi:/var/log/libvirt/qemu# service fwupd-refresh status
● fwupd-refresh.service - Refresh fwupd metadata and update motd
     Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: disabled)
     Active: failed (Result: exit-code) since Fri 2020-10-30 06:43:12 UTC; 29min ago
TriggeredBy: ● fwupd-refresh.timer
       Docs: man:fwupdmgr(1)
    Process: 1110349 ExecStart=/usr/bin/fwupdmgr refresh --no-metadata-check (code=exited, status=1/FAILURE)
   Main PID: 1110349 (code=exited, status=1/FAILURE)
>
>Oct 30 06:43:10 ready-koi systemd[1]: Starting Refresh fwupd metadata and update motd...
Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
Oct 30 06:43:12 ready-koi systemd[1]: Failed to start Refresh fwupd metadata and update motd.
root@ready-koi:/var/log/libvirt/qemu# fwupdmgr --version
client version: 1.3.11
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.11

hughsie, 2020-10-30:
@tsarajar can you open a new issue please, this is probably a different bug.