fwupd/fwupd #1453: Undetected EC/BIOS with EFI+GRUB (Thinkpad X1C 7th)

in0ni, 2019-10-10:
I have tried to update my EC/BIOS using the following instructions to copy fwupdx64.efi to my ESP: https://wiki.archlinux.org/index.php/Fwupd#Setup_for_UEFI_BIOS_upgrade but constantly get "No supported devices found".

I have copied my fwupdx64.efi to both /efi/EFI and /efi/EFI/GRUB, without any luck. I never get *.cap files generated, or perhaps I need to manually do this?

From what I can tell I can never get a LVFS to show up in efibootmgr... I suspect I am missing something simple here. Any help would be greatly appreciated, 

```shell
$ fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.2
```

```shell
$ fwupdmgr get-devices
20QDCTO1WW
│
├─ThinkPad X1 Yoga 4th / Carbon 7th Thunderbolt Controller:
│     Device ID:           ee4cb94af47a4c1dba831e5080a63bf5d53fa1c7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|supported|registered
│     GUID:                c7920601-0cda-507f-851e-92229eb1d370
│   
├─Unifying Receiver:
│     Device ID:           82091cc2e67e8e248d8dad9ac847708118484457
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.10_B0032
│     Bootloader Version:  BOT01.02_B0015
│     Vendor:              Logitech (USB:0x046D)
│     Install Duration:    30 seconds
│     Flags:               updatable|supported|registered
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          279ed287-3607-549e-bacc-f873bb9838c4
│   
└─WDC PC SN720 SDAQNTW-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:             NVM Express Solid State Drive
      Current version:     10160101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               ff2112dc-038c-596d-90ca-d43c5077c6ec
                           63af1441-fb24-5368-b450-886fbc98610b
```

```shell
$ fwupdmgr get-devices --show-all-devices
20QDCTO1WW
│
├─ThinkPad X1 Yoga 4th / Carbon 7th Thunderbolt Controller:
│     Device ID:           ee4cb94af47a4c1dba831e5080a63bf5d53fa1c7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|supported|registered
│     GUID:                c7920601-0cda-507f-851e-92229eb1d370
│   
├─UHD Graphics 620 (Whiskey Lake):
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     Flags:               internal|registered
│     GUIDs:               6986c21a-0f3f-5538-bcef-fe4e8cda7ab7
│                          52aa8ac7-c837-561f-af32-b7979cd2f6ed
│   
├─Unifying Receiver:
│     Device ID:           82091cc2e67e8e248d8dad9ac847708118484457
[... 376 lines cut ...]
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux
```

```shell
$ tree /efi
/efi
└── EFI
    ├── fwupdx64.efi
    └── GRUB
        └── grubx64.efi
```

Please answer the following questions:

- Operating system and version: **Archlinux**
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): **pacman**
- Have you tried rebooting? **Yes, multiple times**
- Are you using an NVMe disk? **yes**
- Is secure boot enabled (only for the UEFI plugin)? **no**

I have also enabled "Flash BIOS Updating by End-User".

Thanks in advance.

Viginox, 2019-10-12:
I had the same issue on my **Lenovo Yoga X1 2019 (Gen 3).**

### What I've tried
I tried to uninstall **fwupd** and install it again. Nothing changed. Still not showing up all devices.
I tried to get information about an old update found in `~/.cache/fwupd`.
`fwupdmgr get-details ~/.cache/fwupd/<filename>.cap` resulted in following output
```
Decompressing…           [***************************************]
20LES09700
│
```
I downloaded the update from LVFS directly and tried to install it with `fwupdmgr install <filename>.cap` and just got the information that no supported device could be found.

### My Solution
The solution for me was to install snap and install fwupd from there. Then I had version 1.3.3 and everything worked like it did before.

[fwupd on Snap](https://snapcraft.io/fwupd)

output from `fwupdmgr --version`
```
client version:	1.3.3
checkout info:	1.3.2-5-g506ef63a
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.3
```
### My System
Running **Manjaro Linux 18.1.0 Juhraya** with Kernel **4.14.148-1-MANJARO**

superm1, 2019-10-13:
This sounds like a duplicate of https://github.com/fwupd/fwupd/issues/1460.
Install tpm2-tss and restart the daemon.

closed 2019-10-13 by superm1

in0ni, 2019-10-14:
@superm1 Thanks for the reference. Installing tpm2-tss only *partially* solved my issue. It seems that the system firmware is now showing up -- but I am still unable to update my [EC](https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN2HHT.firmware), still get "No supported devices found". This update is required before updating BIOS.

Here are both the output of `get-devices` and the verbose output of the daemon logs -- please let me know if you need anything else from me -- and thanks in advance!

```shell
fwupdmgr get-devices --show-all-devices
20QDCTO1WW
│
├─ThinkPad X1 Yoga 4th / Carbon 7th Thunderbolt Controller:
│     Device ID:           ee4cb94af47a4c1dba831e5080a63bf5d53fa1c7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|supported|registered
│     GUID:                c7920601-0cda-507f-851e-92229eb1d370
│   
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     N2HET30W (1.13 )
│     Vendor:              LENOVO
│     Update Error:        UEFI Capsule updates not available or enabled
│     Flags:               internal|require-ac|registered|needs-reboot
│     GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
│   
├─UHD Graphics 620 (Whiskey Lake):
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     Flags:               internal|registered
│     GUIDs:               6986c21a-0f3f-5538-bcef-fe4e8cda7ab7
│                          52aa8ac7-c837-561f-af32-b7979cd2f6ed
│   
├─Unifying Receiver:
│     Device ID:           82091cc2e67e8e248d8dad9ac847708118484457
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.10_B0032
│     Bootloader Version:  BOT01.02_B0015
│     Vendor:              Logitech (USB:0x046D)
│     Install Duration:    30 seconds
│     Flags:               updatable|supported|registered
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          279ed287-3607-549e-bacc-f873bb9838c4
│   
└─WDC PC SN720 SDAQNTW-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:             NVM Express Solid State Drive
      Current version:     10160101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               ff2112dc-038c-596d-90ca-d43c5077c6ec
                           63af1441-fb24-5368-b450-886fbc98610b
```
Daemon logs:

```shell
-- Logs begin at Tue 2019-08-20 22:39:14 CST, end at Sun 2019-10-13 21:21:07 CST. --
Oct 13 21:04:56 xXx systemd[1]: Starting Firmware update daemon...
Oct 13 21:04:58 xXx fwupd[1791]: 03:04:58:0467 FuPluginUefi         Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory
Oct 13 21:04:59 xXx systemd[1]: Started Firmware update daemon.
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0329 Fwupd                looking for os-release at /etc/os-release
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 FuConfig             loading /usr/share/fwupd/metainfo
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 XbSilo               file: cc14f2da-a69e-5ffd-af90-04c6986cf8ed, current:cc14f2da-a69e-5ffd-af90-04c6986cf8ed, cached: (null)
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 XbSilo               loading silo with file contents
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 FuConfig             using config path of /etc/fwupd
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd-tests.conf
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0330 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0331 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0331 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
[... 2 lines cut ...]
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0331 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0331 FuConfig             ordering lvfs-testing=lvfs+1
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0332 FuConfig             ignoring unfound remote fwupd
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0332 FuConfig             ignoring unfound remote fwupd
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0332 FuConfig             ::configuration changed
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0333 FuEngine             remote lvfs-testing not enabled, so skipping
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0333 FuEngine             remote fwupd-tests not enabled, so skipping
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0333 FuEngine             remote vendor not enabled, so skipping
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0333 FuEngine             remote vendor-directory not enabled, so skipping
[... cut ...]
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0339 XbSilo               1.28ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='ff2112dc-038c-596d-90ca-d43c5077c6ec']/../..|components/component/provides/firmware[@type='flashed'][text()='63af1441-fb24-5368-b450-886fbc98610b']/../..` limit=1 -> 0 results
Oct 13 21:21:05 xXx fwupd[1791]: 03:21:05:0340 XbSilo               0.65ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='9d131a0c-a606-580f-8eda-80587250b8d6']/../..|components/component/provides/firmware[@type='flashed'][text()='279ed287-3607-549e-bacc-f873bb9838c4']/../..` limit=1 -> 1 results
```

in0ni, 2019-10-14:
Thanks @superm1!

This and also enabling "Windows UEFI Firmware Update" (as shown in #856) in the BIOS did the trick