fwupd/fwupd #1454: Failed to open plugin libfu_plugin_uefi.so on Lenovo T590

MatTheCat, 2019-10-11:
The firmware doesn't appear with `get-devices` so `get-updates` always return `No updatable devices`.

Running `fwupd --verbose` I noticed the following line:
```
failed to open plugin /usr/lib/fwupd-plugins-3/libfu_plugin_uefi.so: failed to open plugin: libtss2-esys.so.0: Ne peut ouvrir le fichier d'objet partagé: Aucun fichier ou dossier de ce type
```

But

```shell
$ ls -l /usr/lib/fwupd-plugins-3/libfu_plugin_uefi.so
-rwxr-xr-x 1 root root 227232 26 sept. 17:25 /usr/lib/fwupd-plugins-3/libfu_plugin_uefi.so
```

I don't understand :confused: 

```shell
$ fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.2
```

```shell
$ fwupdmgr get-devices
20N4CTO1WW
│
├─T490/T590/P53s/P43s Thunderbolt Controller:
│     Device ID:           576d3cd0e45295481fe6d7ffc7db4e444e454df7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     18.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:                52265728-359a-574c-9a6a-a23d3b1f952d ← TBT-01091804-native
│   
└─SAMSUNG MZVLB512HBJQ-000L7:
      Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
      Summary:             NVM Express Solid State Drive
      Current version:     3L2QEXF7
      Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
      Serial Number:       S4ENNF1M714440
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
                           f4ceb2a8-09c4-55c0-af2a-13ccd63925b1 ← SAMSUNG MZVLB512HBJQ-000L7
```

```shell
$ efibootmgr -v
BootCurrent: 001B
Timeout: 0 seconds
BootOrder: 0010,0011,0012,0013,0014,0015,001B,0019,001A,001C,001D,001E,001F,0020,0021,0022,0023,0024
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0016  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0017  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0018  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0019* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001A* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001B* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001C* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001D* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001E* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001F* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PXE BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021* HTTPS BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri()
Boot0022* LENOVO CLOUD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0023  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0024  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0025* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0026* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0027* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0028* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw

```

```shell
$ tree /boot
/boot
├── 06f4c5a66b8d4436b223479290f4d9dd
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux
```

- Operating system and version: **Arch Linux 5.3.5-arch1-1-ARCH x86_64**
- How did you install fwupd: **pacman**
- Have you tried rebooting? **Yes**
- Are you using an NVMe disk? **Yes**
- Is secure boot enabled (only for the UEFI plugin)? **No**

hughsie, 2019-10-11:
Does libtss2-esys.so.0 exist on your system?

FFY00, 2019-10-11:
Perhaps I don't have any compatible device.

```
\20LBCTO1WW
│
├─P52s Thunderbolt Controller:
│     Device ID:           685059db1d7bba3ac49b1e7810636e1e64f7938b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     18.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:                4cbf4619-f34e-5bed-b1bc-de03d2f48afd
│
├─INTEL SSDPEKKF256G8L:
│     Device ID:           be29aadda98b637d33b6b740ff7b367f61886a31
│     Summary:             NVM Express Solid State Drive
│     Current version:     L08P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          cdc79848-b022-5a05-a77a-862f21df0c29
│
└─Unifying Receiver:
      Device ID:           e00407b12b98918824a3089a03c4a6a798378368
      Summary:             A miniaturised USB wireless receiver (bootloader)
      Current version:     RQR12.09_B0030
      Bootloader Version:  BOT01.04_B0016
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Flags:               updatable|supported|registered|is-bootloader
      GUIDs:               279ed287-3607-549e-bacc-f873bb9838c4
                           aa50f902-6989-5b40-ad1a-29955b5f1450
                           9d131a0c-a606-580f-8eda-80587250b8d6
```

If this is true could we softfail opening a plugin if dependencies are not met?

hughsie, 2019-10-11:
> because from my testing the plugin won't be loaded if it missing

Well, that's right, but I think people probably want UEFI updates to work by default..

FFY00, 2019-10-11:
From my POV if this is preventing fwupd from starting at all then I would make it a hard dependency. But the ideal would actually make fwupd softfail when it is unsuccessful on loading plugins.

MatTheCat, 2019-10-11:
Indeed /usr/lib/libtss2-esys.so does not exist on my system. Does that mean I have to install tpm2-tss?

hughsie, 2019-10-11:
> get-updates still says No updatable devices

Your system doesn't seem supported on the LVFS, yet. the T590 is already supported, but your T590 variant is not yet available for some reason.

MatTheCat, 2019-10-11:
You mean I wouldn't get  `UEFI Capsule updates not available or enabled` if it was supported?

superm1, 2019-10-11:
> From my POV if this is preventing fwupd from starting at all then I would make it a hard dependency. But the ideal would actually make fwupd softfail when it is unsuccessful on loading plugins.

From the description this is what happened. Fwupd started but UEFI plugin didn't load. It doesn't show plugin load errors unless in verbose mode, which is fully intentional because many plugins require very specific hardware at probe time.

> You mean I wouldn't get UEFI Capsule updates not available or enabled if it was supported?

Check BIOS setup and see if there is something to turn on for Uefi capsules.

FFY00, 2019-10-11:
> From the description this is what happened. Fwupd started but UEFI plugin didn't load. It doesn't show plugin load errors unless in verbose mode, which is fully intentional because many plugins require very specific hardware at probe time.

Oh, nvm then. It is working as I expected.

superm1, 2019-10-11:
Is there a concept of Recommends in arch like Debian and Ubuntu has? Recommends are installed by default but can be removed or when a package is installed flags can be set to not install Recommends on the command line.

FFY00, 2019-10-11:
Optional dependencies, yes. And `tpm2-tss` is already there.

https://www.archlinux.org/packages/community/x86_64/fwupd/

MatTheCat, 2019-10-11:
I found a disabled `Windows UEFI firmware update` option in the BIOS.

```shell
$ fwupdmgr get-devices
20N4CTO1WW
│
├─T490/T590/P53s/P43s Thunderbolt Controller:
│     Device ID:           576d3cd0e45295481fe6d7ffc7db4e444e454df7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     18.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:                52265728-359a-574c-9a6a-a23d3b1f952d ← TBT-01091804-native
│   
├─SAMSUNG MZVLB512HBJQ-000L7:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     3L2QEXF7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S4ENNF1M714440
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          f4ceb2a8-09c4-55c0-af2a-13ccd63925b1 ← SAMSUNG MZVLB512HBJQ-000L7
│   
├─System Firmware:
│     Device ID:           6150dd1f7291b0709289ab8a53cc85a17e117ef2
│     Current version:     0.1.51
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                603baf73-b997-45b5-86b4-2f981a008e18
│   
├─UEFI Device Firmware:
│     Device ID:           b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:     192.40.1433
│     Minimum Version:     0.0.1
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                bf2d0f8b-f9a9-400c-8914-36c225d16eb4
│   
├─UEFI Device Firmware:
│     Device ID:           489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:     0.1.12
│     Minimum Version:     0.1.12
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                38ea6335-29ca-417b-8cd4-6b4e5e866f92
│   
└─UEFI Device Firmware:
      Device ID:           88cf266a57697921241cc5f4b412c3f1b5a07780
      Current version:     1.1.7
      Minimum Version:     0.0.1
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUID:                a6634b3a-f446-42f0-a0b2-62c7d4dcb694

```

```shell
$ fwupdmgr get-updates
No upgrades for UEFI Device Firmware, current is 192.40.1433: 192.35.1427=older
20N4CTO1WW
│
├─System Firmware:
│ │   Device ID:           6150dd1f7291b0709289ab8a53cc85a17e117ef2
│ │   Current version:     0.1.51
│ │   Minimum Version:     0.0.1
│ │   Vendor:              LENOVO
│ │   Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│ │   GUID:                603baf73-b997-45b5-86b4-2f981a008e18
│ │ 
│ ├─ThinkPad T490/ThinkPad P43s/ThinkPad T590/ThinkPad P53s/ ThinkPad T490 HC (W-BIOS) System Update:
│ │     New version:       0.1.54
[... 51 lines cut ...]
  │     New version:       0.1.15
  │     Remote ID:         lvfs
  │     Summary:           Lenovo ThinkPad T490 Embedded Controller Firmware
  │     License:           proprietary
  │     Size:              767,1 ko
  │     Vendor:            Lenovo Ltd.
  │     Flags:             is-upgrade
  │     Description:       Lenovo ThinkPad T490/T590 Embedded Controller Firmware 1.15 The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress. Modified : Fixed an issue where system performance may slow on Linux.
  │   
  └─ThinkPad T490 Embedded Controller Update:
        New version:       0.1.13
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T490 Embedded Controller Firmware
        License:           proprietary
        Size:              767,1 ko
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad T490/T590 Embedded Controller Firmware 1.13  The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        This release includes following changes: System EC
        
        Modified : Fixed an issue where Thunderbolt dock will re-connect to system automatically when battery charge to 100%.
```

I guess it's working!

closed 2019-10-11 by MatTheCat

felagund, 2021-01-12:
I would like to note that it is rather inobvious that one needs  to have "Windows UEFI firmware update" enabled in order to update the firmware in _Linux_ - if you could reach out to Lenovo to change the name of that, that would be great. On fwupd side, googling led me here: https://github.com/rhboot/fwupdate/wiki/Debugging-UEFI-Capsule-updates#i-am-doing-multiple-updates-and-it-fails

and there is not mention of that there. Also the error currently displayed "UEFI Capsule updates not available or enabled" is not that helpfull - it is not called "Capsule updates" in the BIOS settings (though I understand this might be vendor-specific) at all and I did not understand I needed to go to BIOS setting to enable it.

superm1, 2021-01-12:
On Dell systems it is called "UEFI capsule firmware updates".  It's a more correct nomenclature because turning it off doesn't block UEFI updates from a traditional "EXE" file.  And if it's turned off fwupd will offer to turn it on via the `dell-esrt` plugin.

I agree, Lenovo probably should rename the string for this setting.  And until they offer a way to turn it on via software the best fwupd can do is a wiki page, feel free to write one.

felagund, 2021-01-12:
Ok, I added a note onto the wikipade I linked in my previous post. No idea if I am using the right terminology, Lenovo calls it BIOS but technically it is UEFI?

And anyway, I think that fwupd might offer a better error message on its own, isntead of `UEFI Capsule updates not available or enabled` `UEFI Capsule updates not available or enabled in BIOS` would make much more sense to me as it would point me to BIOS settinngs. But I have no idea if other states can cause this error message.

superm1, 2021-01-12:
I think if we're changing it, lets change it to `UEFI Capsule updates not available or enabled in firmware setup`

hughsie, 2021-01-12:
> I would like to note that it is rather inobvious that one needs to have "Windows UEFI firmware update" enabled in order to update the firmware in Linux 

@mrhpearson -- any chance we could get this fixed?

mrhpearson, 2021-01-13:
Yeah - that's just wrong.

It should be changeable - they're changing the sleep settings now from "Windows" vs "Linux" to be more correct with S0ix/modern standby vs S3/legacy. I'm not sure how I missed the firmware update setting :(

As a warning though - this kind of change takes quite a while to roll out, but I'll get the ball rolling (LO-944 for my internal reference)

Mark

hughsie, 2021-01-13:
Many thanks Mark!

PR #2751 (2021-01-12): trivial: adjust error for no UEFI capsules
Point people towards firwmare setup instead (Fixes: #1454)

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 797da4fc (2021-01-12): trivial: adjust error for no UEFI capsules

Point people towards firwmare setup instead (Fixes: #1454)