fwupd/fwupd #1455: Upgrade eMMC Storage Device

gwendalcr, 2019-10-12:
Is there any plan to add a plugin to allow eMMC firmware to be upgraded?

hughsie, 2019-10-12:
Are there any existing linux flashing tools, or any vendors who want to implement this code themselves?

superm1, 2019-10-12:
It looks like the firmware version attribute is advertised in sysfs: https://github.com/torvalds/linux/commit/0f762426769a517d5b278e4e5d579fcea6801734
and there is `mmc_utils` which includes "ffu":
https://git.kernel.org/pub/scm/linux/kernel/git/cjb/mmc-utils.git/tree/mmc_cmds.c

hughsie, 2019-10-12:
So we could add a fwupd plugin, but are any vendors going to be uploading firmware? I'm happy to contact vendors if someone can collect me some email addresses.

gwendalcr, 2019-10-14:
The mmc-utils commands 'ffu' updates eMMC implementing
MODE_OPERATION_CODES (Sandisk) in a single call. For those who don't
(Hynix, Samsung) a reboot/eMMC power cycle is required. Some eMMCs
require quirks. (See for instance
https://patchwork.kernel.org/patch/8829621/)

hughsie, 2019-10-15:
Okay, so it looks pretty straightforward. Now all we need is a vendor who wants to actually ship firmware updates :)

superm1, 2019-10-15:
@gwendalcr do you have some drives and payloads you would be able to test with?

I'll ask around for our stuff, but in general I don't _think_ we use eMMC much.

superm1, 2019-10-17:
For our stuff that we use eMMC we can leverage the exact same contacts.  I can drive a POC within those contacts once we have a plugin to mess with.

hughsie, 2019-10-17:
I don't have any MMC hardware here I think. What do I need to buy to be able to update it with firmware?

superm1, 2019-10-17:
I'd suspect the "easiest" way to interact is something like this: https://www.raspberrypi.org/products/compute-module-3-plus/?variant=compute-module-3plus-lite, but maybe a cheap chromebook that we know already ships an eMMC that has a FW update binary would make a better environment.

@gwendalcr any advice?  Could you lend some hardware for this development?

superm1, 2019-10-17:
Looking around my easily accessible hardware - I have an old "Google Coral" sample w/ a 32GB eMMC. @gwendalcr can you point me at a FW binaries for the eMMC?

superm1, 2019-10-17:
Specifically it looks like a DF4032 if that helps.

superm1, 2019-10-18:
@gwendalcr I ported `mmc`'s `ffu` function into a new `emmc` plugin.  I've verified that it enumerates on the Google Coral I have on hand, but I don't have any payloads to verify with.  So we'll have to merge it untested.  If you can please test, if you run into any problems, patches welcome.

closed 2019-10-18 by superm1

gwendalcr, 2019-10-21:
Thanks, I will ping manufacturers: eMMC firmware upgrade is not easy to test, rollback are not allowed on released firmware.

commit 13023ce7 (2019-10-18): Add a new plugin for working with eMMC devices (Fixes: #1455)
commit eb5d7a46 (2019-10-18): Add a new plugin for working with eMMC devices (Fixes: #1455)