fwupd/fwupd #1460: Lenovo T480 get-devices does not list System Firmware (2)

DarkRB, 2019-10-12:
My issue is very similar to 856: https://github.com/fwupd/fwupd/issues/856
Fwupd do not show entry for "System Firmware" when i use "fwupdmgr get-devices"
The issue 856 was resolved when the user activated capsule uefi update in the BIOS.
However my bios is already configured with both "UEFI only" and capsule update options are activated.

$ fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.2

$ fwupdmgr get-devices

T480 Thunderbolt Controller:
│     ID do dispositivo:   8dca27fd251f2e520773e357860a563c2b1873a7
│     Resumo:              Unmatched performance for high-speed I/O
│     Versão atual:        17.01
│     Fornecedor:          Lenovo (TBT:0x0109)
│     Opções:              internal|updatable|require-ac|registered
│     GUID:                fe328c2d-5300-597b-a21d-991116db8ab6
│   
└─LITEON CA3-8D256:
      ID do dispositivo:   f2759da7fe8e0388c5f3601cb072f837b1070b03
      Resumo:              NVM Express Solid State Drive
      Versão atual:        24879
      Fornecedor:          Lite-On Technology Corporation (NVME:0x14A4)
      Opções:              internal|updatable|require-ac|registered|needs-reboot
      GUIDs:               c8066dea-4bd3-50ad-9e08-d10bab80d671
                           2f3deb1f-5b77-555e-9940-2148efd76a0b
                           e3d85808-9c4b-55d9-b444-880e407e8759

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,001C,0000,001A,0019,001B,0017,0018,001D,001E,001F
Boot0000* Windows Boot Manager  HD(2,GPT,6578a901-832e-41a8-bdad-c4d63fc2b3dd,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...d................
Boot0001* Manjaro       HD(1,GPT,65bfb718-1b0d-4e34-8a3b-f2f9edba66de,0x1000,0x96000)/File(\EFI\Manjaro\grubx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
This command gives no outputs.
```

```shell
$ tree /boot
BASH does not recognize this command (maybe i need some sort of package?)
```

Please answer the following questions:

- Operating system and version: Manjaro (18.10)
- How did you install fwupd: pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? No.

superm1, 2019-10-12:
Can you show your daemon verbose log outputs?

DarkRB, 2019-10-12:
Im sorry for my ignorance, i dont know which command i should use to display these logs.

DarkRB, 2019-10-12:
After some research, here we go: 

fwupdmgr get-devices --show-all-devices --verbose
(fwupdmgr:1471): FuMain-DEBUG: 19:17:02.742: T480 Thunderbolt Controller
  DeviceId:             8dca27fd251f2e520773e357860a563c2b1873a7
  Guid:                 fe328c2d-5300-597b-a21d-991116db8ab6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              17.01
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-10-12

(fwupdmgr:1471): FuMain-DEBUG: 19:17:02.742: Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.65.3590
  VersionBootloader:    11.8.65.3590
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2019-10-12

(fwupdmgr:1471): FuMain-DEBUG: 19:17:02.742: LITEON CA3-8D256
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 c8066dea-4bd3-50ad-9e08-d10bab80d671
  Guid:                 2f3deb1f-5b77-555e-9940-2148efd76a0b
  Guid:                 e3d85808-9c4b-55d9-b444-880e407e8759
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Lite-On Technology Corporation
  VendorId:             NVME:0x14A4
  Version:              24879
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-10-12

(fwupdmgr:1471): FuMain-DEBUG: 19:17:02.742: UHD Graphics 620
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 fed2efa4-7045-55a1-b4fc-b29283d59fe5
  Guid:                 8b72a10c-1279-5f8e-a28a-34fb11a58240
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Version:              07
  VersionFormat:        plain
  Icon:                 audio-card
  Created:              2019-10-12

superm1, 2019-10-12:
That's for the frontend, we need the backend logs.

```
# fwupdmgr modify-config VerboseDomains "*"
```

It should restart the daemon, but if it fails or doesn't run:
` sudo systemctl restart fwupd.service`

Then share the journal output:
```journalctl -u fwupd.service -b```

DarkRB, 2019-10-12:
The output command is pretty big, here is a bit of it:

sudo journalctl -u fwupd.service -b
-- Logs begin at Fri 2019-10-11 00:01:07 -03, end at Sat 2019-10-12 19:47:59 -03. --
out 12 17:55:42 pablo-pc systemd[1]: Starting Firmware update daemon...
out 12 17:55:42 pablo-pc fwupd[21306]: 20:55:42:0416 FuEngine             failed to open plugin /usr/lib/fwupd-plugins-3/libfu_>
out 12 17:55:42 pablo-pc fwupd[21306]: 20:55:42:0419 FuEngine             failed to open plugin /usr/lib/fwupd-plugins-3/libfu_>
out 12 17:55:43 pablo-pc systemd[1]: Started Firmware update daemon.
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0471 Fwupd                looking for os-release at /etc/os-release
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0471 FuConfig             loading /usr/share/fwupd/metainfo
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0472 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0472 XbSilo               file: 0e7c8c95-45b0-578c-a91b-7b66d30aaa23, current:0>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0472 XbSilo               loading silo with file contents
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0472 FuConfig             using config path of /etc/fwupd
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0472 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0473 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd-tests.>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0474 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-direc>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0474 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0475 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.co>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             using config path of /usr/lib/fwupd
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             ignoring unfound remote fwupd
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             ordering lvfs-testing=lvfs+1
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             ignoring unfound remote fwupd
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0476 FuConfig             ::configuration changed
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 FuEngine             remote lvfs-testing not enabled, so skipping
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 FuEngine             remote fwupd-tests not enabled, so skipping
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 FuEngine             remote vendor not enabled, so skipping
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 FuEngine             remote vendor-directory not enabled, so skipping
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               0,05ms      parse blob
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               0,14ms      loaded file
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               file: be419237-53a8-55d3-8c1a-76f6fb375cb4, current:b>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               loading silo with file contents
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0478 XbSilo               0,02ms      parse blob
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0479 XbSilo               0,52ms      query on / with `components/component` li>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0479 FuEngine             307 components now in silo
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0480 XbSilo               0,61ms      query on / with `components/component/pro>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0480 XbSilo               0,33ms      query on / with `components/component/pro>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0482 XbSilo               1,19ms      query on / with `components/component/pro>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0483 XbSilo               1,23ms      query on / with `components/component/pro>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0486 XbSilo               2,37ms      query on / with `components/component/pro>
out 12 19:46:42 pablo-pc fwupd[21306]: 22:46:42:0487 XbSilo               1,74ms      query on / with `components/component/pro>
out 12 19:47:02 pablo-pc systemd[1]: Stopping Firmware update daemon...
out 12 19:47:02 pablo-pc fwupd[21306]: 22:47:02:0761 FuPlugin             performing destroy() on thunderbolt_power
out 12 19:47:02 pablo-pc fwupd[21306]: 22:47:02:0761 FuPlugin             performing destroy() on upower
out 12 19:47:02 pablo-pc fwupd[21306]: 22:47:02:0761 FuPlugin             performing destroy() on redfish
out 12 19:47:02 pablo-pc fwupd[21306]: 22:47:02:0761 FuPlugin             performing destroy() on thunderbolt
out 12 19:47:02 pablo-pc fwupd[21306]: 22:47:02:0761 FuPlugin             performing destroy() on synapticsmst
out 12 19:47:02 pablo-pc systemd[1]: fwupd.service: Succeeded.
out 12 19:47:02 pablo-pc systemd[1]: Stopped Firmware update daemon.

superm1, 2019-10-12:
Those first few lines are the important ones that got cutoff:
```
 out 12 17:55:42 pablo-pc fwupd[21306]: 20:55:42:0416 FuEngine failed to open plugin /usr/lib/fwupd-plugins-3/libfu_>
out 12 17:55:42 pablo-pc fwupd[21306]: 20:55:42:0419 FuEngine failed to open plugin /usr/lib/fwupd-plugins-3/libfu_>
```

Can you redirect it to a file and attach that file?

DarkRB, 2019-10-12:
[complete output command.txt](https://github.com/fwupd/fwupd/files/3721249/complete.output.command.txt)

DarkRB, 2019-10-12:
> Those first few lines are the important ones that got cutoff:
>
> ```
>  out 12 17:55:42 pablo-pc fwupd[21306]: 20:55:42:0416 FuEngine failed to open plugin /usr/lib/fwupd-plugins-3/libfu_>
> out 12 [...]

the command: kate /usr/lib/fwupd-plugins-3/libfu_> returns:
syntax error near unexpected token `newline '

trying without the '>' symbol at the end: kate /usr/lib/fwupd-plugins-3/libfu_ returns a blank file.

superm1, 2019-10-12:
You're still getting it cut off on the right.

Please try this:
```# sudo journalctl -u fwupd.service -b > file.txt```

And attach file.txt

DarkRB, 2019-10-12:
[file.txt](https://github.com/fwupd/fwupd/files/3721254/file.txt)

DarkRB, 2019-10-12:
Some of the output is in portuguese, but the output of this:
 failed to open plugin /usr/lib/fwupd-plugins-3/libfu_plugin_uefi.so: failed to open plugin: libtss2-esys.so.0: não é possível abrir arquivo compartilhado: Arquivo ou diretório inexistente

Means: **cannot open shared file: File or directory does not exist**

superm1, 2019-10-12:
So the failure is that you're missing the package that provides that library. Install that package and UEFI plugin should run.

This is a packager bug you should report to your distro, that they didn't make that a dependency.

DarkRB, 2019-10-12:
I see, indeed there is not such lib in my system: libtss2-esys
I dont know where to find this lib, pacman does not list it, neither aur.

superm1, 2019-10-12:
Try installing tpm2-tss

DarkRB, 2019-10-12:
After installing the package get-devices still remains with the same output, however i found this thread: https://bugs.archlinux.org/task/64095
Seems like the arch linux devs realized that the package is missing.

DarkRB, 2019-10-12:
i think the package tpm2-tss in arch linux while providing the lib, rename it, so fwupd looks for the lib and still doesnt find it.

DarkRB, 2019-10-12:
Maybe some other lib is missing?

superm1, 2019-10-12:
After install try to restart the daemon.

DarkRB, 2019-10-12:
Thanks for the help, now get-devices give the Firmware output :)

Sudo fwupdmgr get-devices                          

│
├─T480 Thunderbolt Controller:
│     ID do dispositivo:   8dca27fd251f2e520773e357860a563c2b1873a7
│     Resumo:              Unmatched performance for high-speed I/O
│     Versão atual:        17.01
│     Fornecedor:          Lenovo (TBT:0x0109)
│     Opções:              internal|updatable|require-ac|registered
│     GUID:                fe328c2d-5300-597b-a21d-991116db8ab6 ← TBT-01091707-native
│   
├─LITEON CA3-8D256:
│     ID do dispositivo:   f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Resumo:              NVM Express Solid State Drive
│     Versão atual:        24879
│     Fornecedor:          Lite-On Technology Corporation (NVME:0x14A4)
│     Número de série:     SS0Q94037L2BR97V50DC
│     Opções:              internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               c8066dea-4bd3-50ad-9e08-d10bab80d671 ← NVME\VEN_14A4&DEV_2300&REV_01
│                          2f3deb1f-5b77-555e-9940-2148efd76a0b ← NVME\VEN_14A4&DEV_2300
│                          e3d85808-9c4b-55d9-b444-880e407e8759 ← LITEON CA3-8D256
│   
├─System Firmware:
│     ID do dispositivo:   48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Versão atual:        0.1.26
│     Versão mínima:       0.0.1
│     Fornecedor:          LENOVO
│     Opções:              internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│   
├─UEFI Device Firmware:
│     ID do dispositivo:   5decfff65e3391259bca4003a090e59bf50e3bea
│     Versão atual:        184.65.3590
│     Versão mínima:       0.0.1
│     Opções:              internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                dd31d983-cf3b-4c84-acc9-70caa94e827b
│   
├─UEFI Device Firmware:
│     ID do dispositivo:   c19580db344e418cf2b23297bcf46d75c06e432a
│     Versão atual:        0.1.17
│     Versão mínima:       0.1.17
│     Opções:              internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                b87a926d-189c-49a1-b1ff-921099b1de89
│   
├─UEFI Device Firmware:
│     ID do dispositivo:   27314a74b0935338fbf788ab05043e745a318147
│     Versão atual:        2.1.9
│     Versão mínima:       0.0.1
│     Opções:              internal|updatable|require-ac|registered|needs-reboot
│     GUID:                2296f4c0-13a0-48f0-b5cb-83e7a1b868a2
│   
└─UEFI Device Firmware:
      ID do dispositivo:   66793ff4aaca476be9b106af22077b9fe8d0a313
      Versão atual:        1.3.1084
      Versão mínima:       0.0.1
      Opções:              internal|updatable|require-ac|registered|needs-reboot
      GUID:                77e3677e-dcc2-465c-b2a0-6c37aa6ba300

DarkRB, 2019-10-12:
Weird but my BIOS is actually up to date:

sudo fwupdmgr update
No upgrades for System Firmware, current is 0.1.26: 0.1.26=same, 0.1.25=older, 0.1.24=older, 0.1.23=older, 0.1.21=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.17: 0.1.17=same, 0.1.16=older, 0.1.14=older

closed 2019-10-12 by superm1

DarkRB, 2019-10-12:
Anyway i think we can close this issue now, thanks for the help again :)