fwupd/fwupd #1461: daemon crashing under systemd when replugging

superm1, 2019-10-12:
I am noticing some really "weird" behavior when the daemon is running under systemd if the client is gnome-firmware.  It will crash when a device replugs, but I can't really explain it.  Doesn't happen when using fwupdmgr, doesn't happen when daemon is run outside systemd.

The crash looks to be in glib, but the argument passed into glib was max tokens set to "1" not, "5" as it's supposed to be `fwupd-common.c`.

```
           PID: 2169 (fwupd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Sat 2019-10-12 16:19:13 CDT (13min ago)
  Command Line: /usr/local/libexec/fwupd/fwupd
    Executable: /usr/local/libexec/fwupd/fwupd
 Control Group: /system.slice/fwupd.service
          Unit: fwupd.service
         Slice: system.slice
       Boot ID: 566842b3a45b43ebb74cec1ed523a1c2
    Machine ID: b0dc656c975a4b2282bd8814dbfa7d99
      Hostname: supermario-XPS-13-7390
       Storage: /var/lib/systemd/coredump/core.fwupd.0.566842b3a45b43ebb74cec1ed523a1c2.2169.1570915153000000.lz4
       Message: Process 2169 (fwupd) of user 0 dumped core.
                
                Stack trace of thread 2169:
                #0  0x00007f5ce4b68c73 g_slice_alloc (libglib-2.0.so.0)
                #1  0x00007f5ce4b69b6a g_slist_prepend (libglib-2.0.so.0)
                #2  0x00007f5ce4b6be8d g_strsplit (libglib-2.0.so.0)
                #3  0x00007f5ce50bc6ec fwupd_guid_from_string (libfwupd.so.2)
                #4  0x00007f5ce50bcafa fwupd_guid_hash_data (libfwupd.so.2)
                #5  0x00007f5ce50bcc7f fwupd_guid_hash_string (libfwupd.so.2)
                #6  0x000055e766387bfa fu_device_add_instance_id_full (fwupd)
                #7  0x000055e766387ce3 fu_device_add_instance_id (fwupd)
                #8  0x000055e7663a1de4 fu_usb_device_probe (fwupd)
                #9  0x000055e76638a625 fu_device_probe (fwupd)
                #10 0x000055e76638a38c fu_device_open (fwupd)
                #11 0x000055e76638b991 fu_device_locker_new_full (fwupd)
                #12 0x000055e76638b82a fu_device_locker_new (fwupd)
                #13 0x000055e76639850d fu_plugin_usb_device_added (fwupd)
                #14 0x000055e766398829 fu_plugin_runner_usb_device_added (fwupd)
                #15 0x000055e7663b593d fu_engine_usb_device_added_cb (fwupd)
                #16 0x00007f5ce4c356e2 g_closure_invoke (libgobject-2.0.so.0)
                #17 0x00007f5ce4c49544 n/a (libgobject-2.0.so.0)
                #18 0x00007f5ce4c52327 g_signal_emit_valist (libgobject-2.0.so.0)
                #19 0x00007f5ce4c529b3 g_signal_emit (libgobject-2.0.so.0)
                #20 0x00007f5ce4ac5a41 n/a (libgusb.so.2)
                #21 0x00007f5ce4b4a72e g_main_context_dispatch (libglib-2.0.so.0)
                #22 0x00007f5ce4b4aae0 n/a (libglib-2.0.so.0)
                #23 0x00007f5ce4b4add3 g_main_loop_run (libglib-2.0.so.0)
                #24 0x000055e7663bf599 main (fwupd)
                #25 0x00007f5ce46cd1e3 __libc_start_main (libc.so.6)
                #26 0x000055e76637adbe _start (fwupd)
                
                Stack trace of thread 2180:
                #0  0x00007f5ce47bbc2f __GI___poll (libc.so.6)
                #1  0x00007f5ce40601a2 n/a (libusb-1.0.so.0)
                #2  0x00007f5ce48a0669 start_thread (libpthread.so.0)
                #3  0x00007f5ce47c8323 __clone (libc.so.6)
                
                Stack trace of thread 2179:
                #0  0x00007f5ce47bbc2f __GI___poll (libc.so.6)
                #1  0x00007f5ce4b4aa4e n/a (libglib-2.0.so.0)
                #2  0x00007f5ce4b4ab83 g_main_context_iteration (libglib-2.0.so.0)
                #3  0x00007f5ce4b4abd1 n/a (libglib-2.0.so.0)
                #4  0x00007f5ce4b74121 n/a (libglib-2.0.so.0)
                #5  0x00007f5ce48a0669 start_thread (libpthread.so.0)
                #6  0x00007f5ce47c8323 __clone (libc.so.6)
                
                Stack trace of thread 2182:
                #0  0x00007f5ce47bbc2f __GI___poll (libc.so.6)
                #1  0x00007f5ce4b4aa4e n/a (libglib-2.0.so.0)
                #2  0x00007f5ce4b4add3 g_main_loop_run (libglib-2.0.so.0)
                #3  0x00007f5ce4d9b4fa n/a (libgio-2.0.so.0)
[... 51 lines cut ...]
#7  0x00007f5ce50bcc7f in fwupd_guid_hash_string (str=0x55e76806eb10 "USB\\VID_2DC8&PID_5750") at ../libfwupd/fwupd-common.c:849
#8  0x000055e766387bfa in fu_device_add_instance_id_full (self=0x55e768060d80 [FuEbitdoDevice], instance_id=0x55e76806eb10 "USB\\VID_2DC8&PID_5750", flags=FU_DEVICE_INSTANCE_FLAG_NONE)
    at ../src/fu-device.c:918
#9  0x000055e766387ce3 in fu_device_add_instance_id (self=0x55e768060d80 [FuEbitdoDevice], instance_id=0x55e76806eb10 "USB\\VID_2DC8&PID_5750") at ../src/fu-device.c:939
#10 0x000055e7663a1de4 in fu_usb_device_probe (device=0x55e768060d80 [FuEbitdoDevice], error=0x7fff8cfc2288) at ../src/fu-usb-device.c:258
#11 0x000055e76638a625 in fu_device_probe (self=0x55e768060d80 [FuEbitdoDevice], error=0x7fff8cfc2288) at ../src/fu-device.c:2139
#12 0x000055e76638a38c in fu_device_open (self=0x55e768060d80 [FuEbitdoDevice], error=0x7fff8cfc2288) at ../src/fu-device.c:2040
#13 0x000055e76638b991 in fu_device_locker_new_full (device=0x55e768060d80, open_func=0x55e76638a2b3 <fu_device_open>, close_func=0x55e76638a410 <fu_device_close>, error=0x7fff8cfc2288)
    at ../src/fu-device-locker.c:154
#14 0x000055e76638b82a in fu_device_locker_new (device=0x55e768060d80, error=0x7fff8cfc2288) at ../src/fu-device-locker.c:104
#15 0x000055e76639850d in fu_plugin_usb_device_added (self=0x55e76800dfa0 [FuPlugin], device=0x55e768063360 [FuUsbDevice], error=0x7fff8cfc2288) at ../src/fu-plugin.c:1451
#16 0x000055e766398829 in fu_plugin_runner_usb_device_added (self=0x55e76800dfa0 [FuPlugin], device=0x55e768063360 [FuUsbDevice], error=0x7fff8cfc22f0) at ../src/fu-plugin.c:1515
#17 0x000055e7663b593d in fu_engine_usb_device_added_cb (ctx=0x55e767f63d60 [GUsbContext], usb_device=0x7f5ccc006740 [GUsbDevice], self=0x55e767f62830 [FuEngine]) at ../src/fu-engine.c:4337
#21 0x00007f5ce4c529b3 in <emit signal ??? on instance 0x55e767f63d60 [GUsbContext]> (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>) at ../../../gobject/gsignal.c:3453
    #18 0x00007f5ce4c356e2 in g_closure_invoke (closure=0x55e76805adf0, return_value=0x0, n_param_values=2, param_values=0x7fff8cfc24f0, invocation_hint=0x7fff8cfc2470) at ../../../gobject/gclosure.c:810
    #19 0x00007f5ce4c49544 in signal_emit_unlocked_R
    (node=node@entry=0x55e767fae4a0, detail=detail@entry=0, instance=instance@entry=0x55e767f63d60, emission_return=emission_return@entry=0x0, instance_and_params=instance_and_params@entry=0x7fff8cfc24f0) at ../../../gobject/gsignal.c:3641
    #20 0x00007f5ce4c52327 in g_signal_emit_valist (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>, var_args=var_args@entry=0x7fff8cfc26c0) at ../../../gobject/gsignal.c:3397
#22 0x00007f5ce4ac5a41 in  () at /lib/x86_64-linux-gnu/libgusb.so.2
#23 0x00007f5ce4b4a72e in g_main_dispatch (context=0x55e767f5d500) at ../../../glib/gmain.c:3179
#24 0x00007f5ce4b4a72e in g_main_context_dispatch (context=context@entry=0x55e767f5d500) at ../../../glib/gmain.c:3844
#25 0x00007f5ce4b4aae0 in g_main_context_iterate (context=0x55e767f5d500, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:3917
#26 0x00007f5ce4b4add3 in g_main_loop_run (loop=0x55e767f43060) at ../../../glib/gmain.c:4111
#27 0x000055e7663bf599 in main (argc=1, argv=0x7fff8cfc2a98) at ../src/fu-main.c:1621
```

hughsie, 2019-10-13:
That is weird. Does running it under valgrind give you some kind of invalid write?

superm1, 2019-10-14:
Doesn't want to crash for me under valgrind.  However I got it to crash outside systemd, the coredump looked different this time too.

```
#0  magazine_chain_pop_head (magazine_chunks=0x55d9cef13b90) at ../../../glib/gslice.c:538
538	../../../glib/gslice.c: No such file or directory.
[Current thread is 1 (Thread 0x7f90270e5180 (LWP 17740))]
(gdb) bt
#0  0x00007f902a242c7d in magazine_chain_pop_head (magazine_chunks=0x55d9cef13b90) at ../../../glib/gslice.c:538
#1  0x00007f902a242c7d in thread_memory_magazine1_alloc (tmem=0x55d9cef13b80, ix=0) at ../../../glib/gslice.c:841
#2  0x00007f902a242c7d in g_slice_alloc (mem_size=mem_size@entry=16) at ../../../glib/gslice.c:1015
#3  0x00007f902a243b6a in g_slist_prepend (list=0x0, data=data@entry=0x55d9cf02bda0) at ../../../glib/gslist.c:259
#4  0x00007f902a31457c in g_object_notify_queue_add (pspec=0x55d9cf02bda0 [GParamObject], nqueue=<optimized out>, nqueue=<optimized out>, object=<optimized out>) at ../../../gobject/gobject.c:333
#5  0x00007f902a3156d8 in object_set_property (nqueue=0x55d9cf039110, value=0x55d9cf02bd80, pspec=<optimized out>, object=0x55d9cf046ba0 [FuEbitdoDevice]) at ../../../gobject/gobject.c:1478
#6  0x00007f902a3156d8 in g_object_new_internal (class=class@entry=0x55d9cf0b5050, params=params@entry=0x0, n_params=n_params@entry=0) at ../../../gobject/gobject.c:1861
#7  0x00007f902a316a45 in g_object_new_with_properties (object_type=0x55d9ceffd6c0 [FuEbitdoDevice/FuUsbDevice/FuDevice/FwupdDevice], n_properties=0, names=names@entry=0x0, values=values@entry=0x0)
    at ../../../gobject/gobject.c:1995
#8  0x00007f902a3175f1 in g_object_new (object_type=<optimized out>, first_property_name=<optimized out>) at ../../../gobject/gobject.c:1667
#9  0x000055d9cee4642c in fu_plugin_usb_device_added (self=0x55d9ceff47a0 [FuPlugin], device=0x55d9cf04a1a0 [FuUsbDevice], error=0x7ffdf2741c18) at ../src/fu-plugin.c:1434
#10 0x000055d9cee46829 in fu_plugin_runner_usb_device_added (self=0x55d9ceff47a0 [FuPlugin], device=0x55d9cf04a1a0 [FuUsbDevice], error=0x7ffdf2741c80) at ../src/fu-plugin.c:1515
#11 0x000055d9cee63936 in fu_engine_usb_device_added_cb (ctx=0x55d9cef48d60 [GUsbContext], usb_device=0x55d9cf04de20 [GUsbDevice], self=0x55d9cef47830 [FuEngine]) at ../src/fu-engine.c:4337
#15 0x00007f902a32c9b3 in <emit signal ??? on instance 0x55d9cef48d60 [GUsbContext]> (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>) at ../../../gobject/gsignal.c:3453
    #12 0x00007f902a30f6e2 in g_closure_invoke (closure=0x55d9cf03d3d0, return_value=0x0, n_param_values=2, param_values=0x7ffdf2741e80, invocation_hint=0x7ffdf2741e00) at ../../../gobject/gclosure.c:810
    #13 0x00007f902a323544 in signal_emit_unlocked_R
    (node=node@entry=0x55d9cef91b20, detail=detail@entry=0, instance=instance@entry=0x55d9cef48d60, emission_return=emission_return@entry=0x0, instance_and_params=instance_and_params@entry=0x7ffdf2741e80) at ../../../gobject/gsignal.c:3641
    #14 0x00007f902a32c327 in g_signal_emit_valist (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>, var_args=var_args@entry=0x7ffdf2742050) at ../../../gobject/gsignal.c:3397
#16 0x00007f902a19fa41 in  () at /lib/x86_64-linux-gnu/libgusb.so.2
#17 0x00007f902a22472e in g_main_dispatch (context=0x55d9cef42af0) at ../../../glib/gmain.c:3179
#18 0x00007f902a22472e in g_main_context_dispatch (context=context@entry=0x55d9cef42af0) at ../../../glib/gmain.c:3844
#19 0x00007f902a224ae0 in g_main_context_iterate (context=0x55d9cef42af0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:3917
#20 0x00007f902a224dd3 in g_main_loop_run (loop=0x55d9cef28290) at ../../../glib/gmain.c:4111
#21 0x000055d9cee6d592 in main (argc=1, argv=0x7ffdf2742428) at ../src/fu-main.c:1621
```

hughsie, 2019-10-14:
Can you find out what device_gtype is before the object is created? A `g_type_from_name (value)` might be a good debugging statement to add anyway before the object is created.

closed 2019-10-14 by hughsie