fwupd/fwupd #1503: 8BitDo SN30 pro+ support

cmol, 2019-10-27:
```shell
$ fwupd.fwupdmgr --version  
client version:	1.3.3
checkout info:	1.3.2-5-g506ef63a
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.3
```

`fwupd` installed from latest stable snap.

I cannot seem to find anything confirming that the SN30 pro+ is supported with fwupd currently. Are we missing the firmware files, and is there a way I can help get them? Also, I have a feeling the USB device ID is different for this controller:

`Bus 003 Device 002: ID 2dc8:6002`

The closest I can find is `2dc8:6001` in this file: https://github.com/fwupd/8bitdo-firmware/blob/master/releases/SF30_Pro-SN30_Pro/1.30/sf30sn30pro.metainfo.xml

Thank you!

superm1, 2019-10-27:
Looks like a new ID to me. Does it take the same payloads as regular SN30 pro or has it's own firmware?

cmol, 2019-10-28:
It looks like it has its own version separate from the sn30pro with the sn30pro being on version 1.32, and the sn30pro+ being on version 3.01. So I'd expect the payload to be separate as well.

https://support.8bitdo.com/

superm1, 2019-10-28:
I opened a PR to add your device to the quirk list.  It still needs to have a firmware payload created, so file another issue here if you haven't already: https://github.com/fwupd/missing-firmware-8bitdo

closed 2019-10-28 by hughsie

cmol, 2019-10-28:
I will add the issue at that repo tonight. Thank you!

cmol, 2019-10-28:
Issue has been created in the other repo: https://github.com/fwupd/missing-firmware-8bitdo/issues/5

commit d2701b12 (2019-10-28): ebitdo: Add SN30Pro+ to recognized quirk list (Fixes: #1503)