fwupd/fwupd #1504: Cannot update Latitude 7480 System Firmware (probably SecureBoot)

exploide, 2019-10-28:
Today, I received an update for "Latitude 7480 System Firmware" version 0.1.16.1.

After agreeing to reboot, a blue error screen appeared, saying

> ERROR
Verification failed: (0x1A) Security Violation

Pressing OK opened a configuration screen allowing me to configure MOK keys (probably that is shim?!).

After reboot, the firmware was obviously not updated and I could run the procedure again.

I'm wondering why this happens, since I performed updates for this device multiple times with my setup. SecureBoot is obviously enabled but currently only with default keys. I have not tried updating with SecureBoot disabled yet, in order to perform debug steps in case you tell me to do so.

Besides, I noticed a strange thing: Running `fwupdmgr get-updates` again, prompts me to upload a report of a previous update (0.1.14.1 → 0.1.15.1). This happens after each failed try of updating to the new firmware 0.1.16.1. It is probably a bug that the same old report can be uploaded multiple times?!

```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

<details><summary>get-devices</summary><p>

```shell
$ fwupdmgr get-devices
Latitude 7480 Thunderbolt Controller
  DeviceId:             1378c419fdf36f48957f08a695808c6d0af69783
  Guid:                 afbdf062-a250-54d2-b49e-a3c1ebd9493c
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              23.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-10-28

Dell Thunderbolt Cable
  DeviceId:             5a2a3310fd580ed72006f45db746c12f1dc828aa
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 b4fd3cdf-4e3a-5090-a583-45367cfd6421
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-10-28

Dell Thunderbolt Dock
  DeviceId:             a64a610da5f1d8af05ca92f8bda944f24dcfe833
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 76cc74d4-f062-5b93-a11c-8d2a58a25848
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              27.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-10-28

Latitude 7480 System Firmware
  DeviceId:             07e9e86366cf9f3ea796c4d2f1f07ce837f8fa0d
  Guid:                 53f51f56-eff2-4516-9dda-ed501b3f0f8a
[... 231 lines cut ...]
│       ├── 71d0ee7dfd7546bd8e957dc1e4f3e004-0-rescue.conf
│       ├── 71d0ee7dfd7546bd8e957dc1e4f3e004-5.3.5-200.fc30.x86_64.conf
│       ├── 71d0ee7dfd7546bd8e957dc1e4f3e004-5.3.6-200.fc30.x86_64.conf
│       └── 71d0ee7dfd7546bd8e957dc1e4f3e004-5.3.7-200.fc30.x86_64.conf
├── lost+found
├── System.map-5.3.5-200.fc30.x86_64
├── System.map-5.3.6-200.fc30.x86_64
├── System.map-5.3.7-200.fc30.x86_64
├── vmlinuz-0-rescue-71d0ee7dfd7546bd8e957dc1e4f3e004
├── vmlinuz-5.3.5-200.fc30.x86_64
├── vmlinuz-5.3.6-200.fc30.x86_64
└── vmlinuz-5.3.7-200.fc30.x86_64

16 directories, 98 files
```

</p></details>

Please answer the following questions:

- Operating system and version: Fedora 30
- How did you install fwupd: dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? yes

superm1, 2019-10-28:
So this is the second instance suspected of secure boot causing the problem (see https://github.com/fwupd/fwupd/issues/1421).

I guess first off can you please confirm that if you turn off secure boot it helps fix the problem?

If it does, we need to figure out if this is a Fedora secure boot signing problem (given the other reporter was also a Fedora system) or a fwupd implementation problem with secure boot on newer systems.

exploide, 2019-10-28:
I can confirm it works after disabling SecureBoot.

superm1, 2019-10-28:
Thanks for confirming that.  Given the error message you saw; I'm pretty sure that's coming from shim:
https://github.com/rhboot/shim/blob/741c61abba7d5c74166f8d0c1b9ee8001ebcd186/shim.c#L1317

@hughsie any thoughts on next steps here?  Could something be up with the signing stuff in Fedora? 

CC @vathpela

superm1, 2019-10-28:
Looking at https://kojipkgs.fedoraproject.org//packages/fwupd/1.2.11/2.fc30/data/logs/x86_64/build.log

I think the build server signed with the wrong cert in Fedora 30.
```
+ /usr/bin/pesign -c 'Red Hat Test Certificate' --certdir /etc/pki/pesign-rh-test -i /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc30.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi -o /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc30.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi.signed -s
+ '[' '!' -s -o /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc30.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi.signed ']'
```

Compare that to Fedora 31: https://kojipkgs.fedoraproject.org//packages/fwupd/1.2.11/2.fc31/data/logs/x86_64/build.log

```
+ /usr/bin/pesign-client -t 'Fedora Signer (OpenSC Card)' -c '/CN=Fedora Secure Boot Signer' -i /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc31.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi -o /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc31.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi.signed -s
+ '[' '!' -s -o /builddir/build/BUILDROOT/fwupd-1.2.11-2.fc31.x86_64/usr/libexec/fwupd/efi/fwupdx64.efi.signed ']'
```

superm1, 2019-10-28:
CC @AdamWill  ^

superm1, 2019-10-30:
As this is an issue in Fedora, I'm going to close it upstream and move the issue into a bugzilla.  Please subscribe to this to follow up with the update:
https://bugzilla.redhat.com/show_bug.cgi?id=1767143

closed 2019-10-30 by superm1