fwupd/fwupd #1516: Unable to start systemd service

lgp171188, 2019-10-31:
After upgrading to the `1.3.2` version, the `fwupd` `systemd` service fails to start with the following error.

```
Oct 31 16:40:01 precious systemd[1]: Starting Firmware update daemon...
-- Defined-By: systemd
Oct 31 16:40:01 precious systemd[10854]: fwupd.service: Failed to set up special execution directory in /var/cache: Not a directory
Oct 31 16:40:01 precious systemd[10854]: fwupd.service: Failed at step CACHE_DIRECTORY spawning /usr/lib/fwupd/fwupd: Not a directory
-- Defined-By: systemd
Oct 31 16:40:01 precious systemd[1]: fwupd.service: Main process exited, code=exited, status=239/CACHE_DIRECTORY
-- Defined-By: systemd
Oct 31 16:40:01 precious systemd[1]: fwupd.service: Failed with result 'exit-code'.
-- Defined-By: systemd
Oct 31 16:40:01 precious systemd[1]: Failed to start Firmware update daemon
```
This appears to be the same issue as #1445, but that issue was on Debian and was closed with a fix and I am unable to re-open it to report the same issue. This is happening on Manjaro and it is not clear if it is a `fwupd` issue or a distribution package issue.

On downgrading to `fwupd` `1.2.10`, everything works okay. So this appears to be an issue with the `1.3.2` version.

```shell
$ fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37

$ fwupdmgr get-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* Linux Firmware Updater        HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x100000)/File(\EFI\manjaro\fwupdx64.efi)
Boot0001* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1    HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x100000)/File(\EFI\boot\bootx64.efi)..BO
Boot0005* rEFInd Boot Manager   HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x113000)/File(\EFI\refind\refind_x64.efi)

$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0-0

$ tree /boot
/boot
??? efi
??? ??? EFI
???     ??? boot
???     ??? ??? bootx64.efi
???     ??? Manjaro
???         ??? fw
???         ??? fwupdx64.efi
???         ??? grubx64.efi
??? grub
??? ??? fonts
??? ??? ??? unicode.pf2
??? ??? grub.cfg
??? ??? grubenv
??? ??? themes
??? ??? ??? starfield
??? ???     ??? blob_w.png
??? ???     ??? boot_menu_c.png
??? ???     ??? boot_menu_e.png
??? ???     ??? boot_menu_ne.png
??? ???     ??? boot_menu_n.png
??? ???     ??? boot_menu_nw.png
??? ???     ??? boot_menu_se.png
??? ???     ??? boot_menu_s.png
??? ???     ??? boot_menu_sw.png
??? ???     ??? boot_menu_w.png
??? ???     ??? COPYING.CC-BY-SA-3.0
??? ???     ??? dejavu_10.pf2
??? ???     ??? dejavu_12.pf2
??? ???     ??? dejavu_14.pf2
[... 287 lines cut ...]
???     ??? xnu_uuid_test.mod
???     ??? xzio.mod
???     ??? zfscrypt.mod
???     ??? zfsinfo.mod
???     ??? zfs.mod
???     ??? zstd.mod
??? initramfs-4.19-x86_64-fallback.img
??? initramfs-4.19-x86_64.img
??? intel-ucode.img
??? linux419-x86_64.kver
??? memtest86+
??? ??? memtest.bin
??? vmlinuz-4.19-x86_64

11 directories, 318 files

```

Please answer the following questions:

- Operating system and version: Manjaro, latest
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman`
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-10-31:
Can you please try master? There had been a bunch of activity here.

[EXTERNAL EMAIL]

superm1, 2019-10-31:
Try this commit: https://github.com/fwupd/fwupd/commit/dc7e7c3808d564d5769b2845dbd66a3651f72e07

You may need to manually clean up that directory too.

lgp171188, 2019-11-14:
@superm1, I tried the `1.3.3` release from the testing repos of Manjaro and still see the same error. Can you reopen this issue?

superm1, 2019-11-14:
You need to manually clear that directory if it was created from 1.3.2. arch doesn't have a postinst or preinst script to do it IIRC.

closed 2019-11-14 by lgp171188