fwupd/fwupd #1521: Firmware update issues on Lenovo Thinkpad 6th Gen X1

evilbulgarian, 2019-11-01:
Trying to update firmware results in a black screen, then just reboots back to linux without update.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.2.11
	efivar:	37
daemon version:	1.3.2
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20KHCTO1WW
│
├─Samsung SSD 970 PRO 1TB:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     1B2QEXP7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          21a2c5aa-cba2-5dcb-854b-ddeeb7c63413
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.40
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Update State:        success
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.60.3561
│     Minimum Version:     0.0.1
│     Update State:        success
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│   
└─UEFI Device Firmware:
      Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
      Current version:     0.1.15
      Minimum Version:     0.1.15
      Update State:        success
      Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
      GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0018,0019,001B,001C,001D,001E,001F,001A,0024,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,98b5654a-627f-4ede-9a22-bb5841ea8502,0x800,0x82000)/File(\EFI\gentoo\fwupdx64.efi)
Boot0001* Gentoo GNU/Linux	HD(1,GPT,98b5654a-627f-4ede-9a22-bb5841ea8502,0x800,0x82000)/File(\efi\gentoo\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
[... 518 lines cut ...]
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-genkernel-x86_64-5.3.4
├── initramfs-genkernel-x86_64-5.3.5
├── initramfs-genkernel-x86_64-5.3.7
├── vmlinuz -> /boot/vmlinuz-5.3.7
├── vmlinuz-5.3.4
├── vmlinuz-5.3.5
├── vmlinuz-5.3.7
└── vmlinuz.old -> /boot/vmlinuz-5.3.5

53 directories, 446 files

**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Gentoo GNU/Linux 
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): portage
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-11-01:
Does it work in Fedora, Arch or Ubuntu?

joakim-tjernlund, 2019-11-02:
I have seen this on Lenovo thinkpads (t480s) too, if the current BIOS is too old, then it fails to
upgrade itself.

evilbulgarian, 2019-11-03:
The bios is only one minor version behind what its wanting to update to. Its at 1.40 and update is 1.41.

joakim-tjernlund, 2019-11-04:
OK, that is odd then.

superm1, 2019-11-05:
dup of https://github.com/fwupd/fwupd/issues/1530

closed 2019-11-05 by superm1