fwupd/fwupd #1526: Give information about inability to load plugin to the user

FFY00, 2019-11-03:
Right now when a plugin can't be loaded (ex. missing dependency) we just fail silently. This works fine but causes some confusing among users. I think we should save the reason and during device enumeration we could say that we skip X device because the plugin could not be loaded and provide the reason the plugin wasn't loaded.

@hughsie @superm1 what do you think?

This would prevent issues such as #1454.

superm1, 2019-11-03:
Well the problem is that on Every single system some plugins are disabled. For example you would never have coreboot and UEFI on the same system. So you don't want to message to the user every single plugin that was disabled.

Additionally you don't know what kind or how many devices a plugin will create at the time it's loaded.

The UEFI case you linked above sounds like a packaging problem to me, that library should have been installed if the package was installed.

In Arch could you split fwupd into two binary packages maybe (fwupd and fwupd-efi)? Then make fwupd-efi optional but it requires a hard dependency on the TPM library that was missing.

FFY00, 2019-11-04:
But don't we have metadata to identify the possible compatible devices before loading the plugin? At some point we need to hand the device over to the correct plugin, how do we choose that plugin? From what I understand we do that based on the quirk files, can't just print a message saying that device X, Y and Z were skipped because the plugin couldn't be loaded?

> Well the problem is that on Every single system some plugins are disabled. For example you would never have coreboot and UEFI on the same system. So you don't want to message to the user every single plugin that was disabled.

It is fine if a plugin was disabled, we just print the message if we were trying to use the plugin but couldn't.

> Additionally you don't know what kind or how many devices a plugin will create at the time it's loaded.

So we have no clue about the supported devices before the plugin is loaded?

> The UEFI case you linked above sounds like a packaging problem to me, that library should have been installed if the package was installed.

Well, I don't think it's that clear. For me is kind of a gray area. Packaging every single plugin separately is not a great option.

> In Arch could you split fwupd into two binary packages maybe (fwupd and fwupd-efi)? Then make fwupd-efi optional but it requires a hard dependency on the TPM library that was missing.

True but that's not a long term solution. If we end up adding more plugins with external dependencies we wouldn't want for distributions to have the split all of them in different packages.

Detecting the dependencies at runtime is what most sense to me. I think for most cases it should be possible to identify the supported devices (or at least, potentially supported devices) without needing any external library.

hughsie, 2019-11-04:
> Then make fwupd-efi optional but it requires a hard dependency on the TPM library that was missing

How much space are we saving by not just hard depending on that specific library? If it's less than a few hundred Kb I really don't see what all the fuss is about...

FFY00, 2019-11-04:
> How much space are we saving by not just hard depending on that specific library? If it's less than a few hundred Kb I really don't see what all the fuss is about...

It pulls openssl, which pulls perl.

In this case this is actually fine because pacman pulls openssl as well but it might not be in future cases.

I am not saying this needs to be implemented now, I am just suggesting this would be a good thing to have.

hughsie, 2019-11-04:
> I am just suggesting this would be a good thing to have.

Right, but we have to balance "nice to have" against "long term maintainability and complexity".

FFY00, 2019-11-04:
> Right, but we have to balance "nice to have" against "long term maintainability and complexity".

That's what I was expecting to discuss in this issue.

superm1, 2019-11-04:
> In this case this is actually fine because pacman pulls openssl as well but it might not be in future cases. Also, other distributions might not pull openssl by default.

Doesn't turning on pkcs7 also cause the same thing from fwupd?

> Detecting the dependencies at runtime is what most sense to me. I think for most cases it should be possible to identify the supported devices (or at least, potentially supported devices) without needing any external library.

The problem here is that the TPM library isn't `ldopen`ed.   Even if it's optional to use at runtime (which it is right now) it needs to **physically be there** when the plugin is `ldopen`ed.

> How much space are we saving by not just hard depending on that specific library? If it's less than a few hundred Kb I really don't see what all the fuss is about...

There is a similar discussion happening in a Debian bug about splitting out the EFI application binary to a different package, but it is not based on space constraints.  It's based upon multiarch (ie co-installability of different arch support).

> Right, but we have to balance "nice to have" against "long term maintainability and complexity".

I suppose there is a difference here in terms of the plugin "failed to load" due to a missing dynamically loaded shared library and the plugin failed to startup or init due to something wrong on the system.  The former is more valuable generically, but I don't know really how you would message that generically to a client.  The simplest thought that comes to mind is to make a dummy device at that time with an `UpdateError` set to that string.

superm1, 2019-11-18:
I see at https://www.archlinux.org/packages/community/x86_64/fwupd/ that first option was taken, so will close this issue.
[image]

ArchangeGabriel, 2020-02-14:
Not exactly though, the commit message that changed this is:
`Fix FS#64095 (while waiting for upstream to enhance the error message).`

Also now we have requests to remove modemmanager from hard dependencies (https://bugs.archlinux.org/task/64875). So I guess this should be discussed again.

FFY00, 2020-02-14:
I think for arch a simple message saying `Plugin X has been disabled due to missing depencies` in the daemon log should suffice. People are expected to actively look at the optional dependencies.

For other distributions this might not be enough though.

ArchangeGabriel, 2020-02-14:
We also had a request about udisks2 recently: https://bugs.archlinux.org/task/65288.

I’ve figured out this is related to https://github.com/fwupd/fwupd/pull/1406. So I guess this means those people did not have their ESP mounted (or at a non-defaut location)? It likely means udisks2 should be a hard dep, but the dependency tree of that package is quite huge…

hughsie, 2020-02-14:
> We also had a request about udisks2 recently

fwupd really isn't designed to be sliced and diced; we only barely have enough resources to cope with people having all the deps installed, and really don't want to be fielding issues from people who can't get firmware updates because something like udisks isn't installed.

ArchangeGabriel, 2020-02-14:
@hughsie Then please at least mention it somewhere, because they are [almost no references to udisks2 anywhere](https://github.com/fwupd/fwupd/search?q=udisks2&unscoped_q=udisks2), especially in dependencies. ;)

But do I understand correctly that udisks2 is only required if the ESP is not mounted?

hughsie, 2020-02-14:
> udisks2 is only required if the ESP is not mounted

We've never tested it, sorry. I guess it's probably true.

superm1, 2020-02-14:
> We've never tested it, sorry. I guess it's probably true.

I'd guess the same.

> I’ve figured out this is related to #1406. So I guess this means those people did not have their ESP mounted (or at a non-defaut location)? It likely means udisks2 should be a hard dep, but the dependency tree of that package is quite huge…

Can you make the fwupd source package generate a handful of binary packages with varying dependencies?  For example a regular `fwupd` binary with all our regular deps and a `fwupd-uefi` that has `udisks2` and a `fwupd-modemmanager` that has MM stuff?  I think that's the best of both worlds then.  If someone wants to do UEFI updates they have to install `fwupd-uefi` (which would hopefully depend on `fwupd` and pull that in).

> @hughsie Then please at least mention it somewhere, because they are almost no references to udisks2 anywhere, especially in dependencies. ;)

Where would it make sense to document in your mind?  Compiling wiki page?  UEFI readme page?   Top readme?

superm1, 2020-02-14:
> Can you make the fwupd source package generate a handful of binary packages with varying dependencies? For example a regular fwupd binary with all our regular deps and a fwupd-uefi that has udisks2 [...]

@ArchangeGabriel  looking at https://jlk.fjfi.cvut.cz/arch/manpages/man/PKGBUILD.5#PACKAGE_SPLITTING

I do think this is entirely possible.
so make a `fwupd` with optional deps `fwupd-uefi` and `fwupd-modemmanager`, and then make those other deps hard requirements for those packages.

FFY00, 2020-02-14:
That achieves the same as making them optional dependencies, it just saves us the messages in the daemon logs. Package splitting for things like this goes against the arch philosophy, it would only make sense if the plugins were really big.

> Where would it make sense to document in your mind? Compiling wiki page? UEFI readme page? Top readme?

Yes, in the compiling wiki page. It should document all dependencies.

superm1, 2020-02-14:
> That achieves the same as making them optional dependencies, it just saves us the messages in the daemon logs.

I think of it being much more clear to someone who doesn't follow this stuff.  If you go to the wiki page and it says to install `fwupd-uefi` (which pulls in `fwupd`) then you'll get the right thing.  

> Package splitting for things like this goes against the arch philosophy, it would only make sense if the plugins were really big.

Isn't that exactly what the problem is here?  There are tons of huge dependency chains that not everyone wants, but it's confusing to know you need 'em.

superm1, 2020-02-14:
> Yes, in the compiling wiki page. It should document all dependencies.

I added a runtime dependencies section.  https://github.com/fwupd/fwupd/wiki/Compilation

FFY00, 2020-02-14:
> I think of it being much more clear to someone who doesn't follow this stuff. If you go to the wiki page and it says to install `fwupd-uefi` (which pulls in `fwupd`) then you'll get the right thing.

I understand and I agree, but that's not really how arch works. Arch users are expected to understand what they are doing, if they install a package they are expected to look at the optional dependencies and install what they need.

What you are describing would be the correct approach for a distribution which is focused on making the experience as easy as possible for users, like Ubuntu.

> Isn't that exactly what the problem is here? There are tons of huge dependency chains that not everyone wants, but it's confusing to know you need 'em.

Yes. Both making the dependencies optional or splitting the package would solve the issue, that's not in question here. I am just arguing that making the dependencies optional is the correct way to handle it on arch.

I find that making the dependencies optional instead of splitting the package also solves the problem in a more clean, synergetic way. If my laptop has a modem and I care for it, I will probably have `modemmanager` installed. If I then install `fwupd` I will automatically have updates for my modem, without having to install an extra package (`fwupd-modemmanager`).

Either way, I think we are drifting away from the issue that we supposed to solve, which is how to handle this as an upstream. From what it looks like we are in the same place as before: the upstream can't really justify changing the behavior for now as it would require a fair amount of refactoring and resources.

So I say we just do what we can -- make the error messages better.

Personally I am fine just replacing
```
FuEngine failed to open plugin /usr/lib/fwupd-plugins-3/libfu_plugin_dell.so: failed to open plugin: libtss2-esys.so.0: cannot open shared object file: No such file or directory
```
with something like
```
FuEngine Plugin libfu_plugin_dell.so was disabled due to missing dependencies (libtss2-esys.so.0: cannot open shared object file: No such file or directory)
```

@ArchangeGabriel is this fine with you?

@superm1 @hughsie does that seem reasonable?

I think we would still want to handle this use-case properly in the future, if it is viable.

FFY00, 2020-02-14:
Given how glib handles errors (errno gets overwritten) this was the best I could come up with:
```
Disabled plugin /usr/lib/fwupd-plugins-3/libfu_plugin_dell.so (libtss2-esys.so.0: cannot open shared object file: No such file or directory)
```
I cannot detect the actual error and have a custom message. Would matching the text (look for `cannot open shared object file`) be too janky? If I do that we could have a message like this:
```
Disabled plugin /usr/lib/fwupd-plugins-3/libfu_plugin_dell.so due to missing dependencies (libtss2-esys.so.0)
```
And for normal cases, we would have the message above.

ArchangeGabriel, 2020-02-16:
@superm1 As @FFY00 said, splitting in Arch should only occurs when the package itself is big, not when it brings huge dependencies —for this optdeps are the right way to go.

So regarding Arch we will move to optdeps indeed, and will mark udisks2 as required for UEFI support in order to not go into too much details.

Regarding upstream, I like @FFY00 suggestions for enhancing the error message.

hughsie, 2020-10-17:
@FFY00 can we use something like the recently added https://github.com/fwupd/fwupd/commit/7bcb8d4385665700f21b246747b69ab5aff0c136 to convey this to the user?

FFY00, 2020-10-17:
Seems fine to me, as long as we also export the reason why the plugin wasn't loaded.

superm1, 2020-10-17:
I think we need to confirm that the missing library condition works - I don't think it does. But the infrastructure seems right for it.

commit 029fc1cc (2020-10-20): Show an error when a plugin is missing dependencies (Fixes: #1526)

```
$ sudo mv /usr/lib/x86_64-linux-gnu/libtss2-esys.so.0.0.0 /usr/lib/x86_64-linux-gnu/libtss2-esys.so.0.0.0.renamed
$ sudo fwupdtool get-devices --plugins=uefi
14:15:48:0735 FuEngine             cannot load: failed to open plugin /usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi.so: libtss2-esys.so.0: cannot open shared object file: No such file or directory
Loading…                 [-                                      ]14:15:48:0753 FuEngine             failed to update history database: device ID b6c08fb [... cut]
commit c3a8173a (2020-10-20): Show an error when a plugin is missing dependencies (Fixes: #1526)

```
$ sudo mv /usr/lib/x86_64-linux-gnu/libtss2-esys.so.0.0.0 /usr/lib/x86_64-linux-gnu/libtss2-esys.so.0.0.0.renamed
$ sudo fwupdtool get-devices --plugins=uefi
14:15:48:0735 FuEngine             cannot load: failed to open plugin /usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi.so: libtss2-esys.so.0: cannot open shared object file: No such file or directory
Loading…                 [-                                      ]14:15:48:0753 FuEngine             failed to update history database: device ID b6c08fb [... cut]
closed 2020-10-20 by superm1