fwupd/fwupd #1530: fwupd-1.3.3:

okias, 2019-11-04:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.2.11
	efivar:	37
daemon version:	1.3.3
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
XPS 13 9350
│
├─Thunderbolt Controller:
│     Device ID:           d58bbd545872c866487ca6a5860528c575c0a274
│     Souhrn:              Unmatched performance for high-speed I/O
│     Current version:     16.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               dbbe0ed9-df91-529c-ac86-8976b78f726d ← TBT-00d40704
│                          8e9c55fc-e381-57ae-8f74-a3e5df340bd0 ← TBT-00d40704-0000:03:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Dell WD15:
│ │   Device ID:           84edf044a6b53037732cfc7591b65a8d36384cc9
│ │   Souhrn:              A USB type-C docking station
│ │   Current version:     0.0.0.0
│ │   Vendor:              Dell Inc.
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:         00e65ccfc5a875a15f79acf2081504990453810f
│ │     Souhrn:            A USB type-C docking station
│ │     Current version:   0.0.3.18
│ │     Vendor:            Dell Inc.
│ │     GUID:              fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │   
│ └─Dell WD15 Port Controller 1:
│       Device ID:         e91fa599260a0458fc2137a8352da6ed14c125d5
│       Souhrn:            A USB type-C docking station
│       Current version:   0.1.1.8
│       Vendor:            Dell Inc.
│       GUID:              8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│     
├─PM951 NVMe SAMSUNG 256GB:
│     Device ID:           e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Souhrn:              NVM Express Solid State Drive
│     Current version:     BXV77D0Q
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S29NNXAH214235
│     GUIDs:               cce2386f-6d68-5f85-9f3f-ebead9288696 ← NVME\VEN_144D&DEV_A802&REV_01
│                          77bb4022-9503-586c-99b7-c8a64502da30 ← NVME\VEN_144D&DEV_A802
│                          3a190a70-f325-53fc-9654-94fa36120f3e ← PM951 NVMe SAMSUNG 256GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           79f2996ff11608674c5d2e8db20e10e0b646f2cb
│     Current version:     0.1.7.0
│     Minimum Version:     0.1.7.0
│     Vendor:              Dell Inc.
│     GUID:                33773727-8ee7-4d81-9fa0-57e8d889e1fa
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─TPM 1.2:
  │   Device ID:           b719feb08bda534ee0cb9df33b8de5c5458e2132
  │   Souhrn:              Platform TPM device
  │   Current version:     5.81.0.0
  │   Vendor:              Dell Inc.
  │   Update Error:        Updating disabled due to TPM ownership
  │   GUID:                d433959e-03ca-524b-92b7-5022eff81a31 ← 0704-1.2
  │   Device Flags:        • Internal device
  │                        • Requires AC power
  │                        • Supported on remote server
  │ 
  └─TPM 2.0:
        Device ID:         eb4a4ec71c680a34c6248dc94ca017f81c772aa8
        Souhrn:            Alternate mode for platform TPM device
        Vendor:            Dell Inc.
        Update Error:      mode switch disabled due to TPM ownership
        GUID:              b62a2412-5ac4-5350-b16e-7e8f4655d096 ← 0704-2.0
        Device Flags:      • Internal device
                           • Requires AC power
                           • Device is locked
 
```

```shell
$ efibootmgr -v
BootNext: 0001
BootCurrent: 0005
Timeout: 2 seconds
BootOrder: 0005,0007,0001
Boot0000  Windows Boot Manager	HD(2,GPT,029b8a12-dede-477e-b375-d194c3d6a507,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...*................
Boot0001* Linux Firmware Updater	HD(1,GPT,806e1393-c00c-450c-b414-54e2fe5122ab,0x800,0x64000)/File(\EFI\gentoo\fwupdx64.efi)
Boot0005* Gentoo	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-42-61-B1-E3-67)/HD(1,GPT,806e1393-c00c-450c-b414-54e2fe5122ab,0x800,0x64000)/File(\Gentoo\bzImage)
Boot0007* Gentoo failsafe	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-42-61-B1-E3-67)/HD(1,GPT,806e1393-c00c-450c-b414-54e2fe5122ab,0x800,0x64000)/File(\Gentoo\old)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-33773727-8ee7-4d81-9fa0-57e8d889e1fa-0
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
drwxr-xr-x   - root  5 čec  2018 /boot/efi/EFI
drwxr-xr-x   - root  4 lis 11:54 └── gentoo
drwxr-xr-x   - root  4 lis 11:58    ├── fw
.rwxr-xr-x 11M root  4 lis 11:58    │  └── fwupd-33773727-8ee7-4d81-9fa0-57e8d889e1fa.cap
.rwxr-xr-x 57k root  4 lis 11:50    ├── fwupdx64.efi
.rwxr-xr-x 71k root  5 čec  2018    └── fwupx64.efi

**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Gentoo ~amd64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): emerge
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

BIOS updating shows only it found .cap and then nothing happens.

superm1, 2019-11-04:
Can you please:

1. turn on EFI application debugging:
`/usr/libexec/fwupd/fwupdate -d`
2. Do the update again 
3. Share the debug logging output:
`/usr/libexec/fwupd/fwupdate -L`

You can turn off debugging later with the `-D` argument to `fwupdate`

okias, 2019-11-04:
`../fwupd-1.3.3/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-33773727-8ee7-4d81-9fa0-57e8d889e1fa-0../fwupd-1.3.3/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule#`

superm1, 2019-11-04:
Well that all looks fine to me.  I'm a bit perplexed from your behavior.  To rule out an issue caused by a compilation problem in Gentoo would you please do the following:
1. grab the unsigned `fwupdx64.efi` from Ubuntu, Debian or Fedora.
2. Remove the `fwupdx64.efi` from your ESP
3. Replace the `fwupdx64.efi` installed by your gentoo package with the one from another distro
4. Re-run the updater.  It will copy the "new" `fwupdx64.efi` to the ESP to run the update.

See if that helps.  If so - we're looking at something Gentoo is doing to the packages or something in your `CFLAGS` leading to problems.

Interestingly enough there is someone else on Gentoo that reported a really similar behavior recently that isn't root caused yet.
https://github.com/fwupd/fwupd/issues/1521

okias, 2019-11-05:
@superm1 thank you, it helped.

My CFLAGS are set to CFLAGS="-march=native -O2 -flto -pipe". Maybe the LTO flag?

okias, 2019-11-05:
```
.rwxr-xr-x 57k root  5 lis 10:39 fwupdx64.efi # with binutils-config --linker ld.gold
.rwxr-xr-x 61k root  5 lis 10:47 fwupdx64.efi # with binutils-config --linker ld.bfd
.rwxr-xr-x 61k root  5 lis 10:38 fwupdx64.efi_ubnt # Ubuntu

```
can I retry somehow with "dummy update"? I have feeling that issue is code or gold linker.

okias, 2019-11-05:
```
meson --buildtype plain --libdir lib64 --localstatedir /var/lib --prefix /usr --sysconfdir /etc --wrap-mode nodownload --localstatedir /var -Dbuild=all -Dagent=false -Dplugin_amt=false -Dconsolekit=false -Dplugin_dell=true -Delogind=false -Dgpg=true -Dgtkdoc=false -Dman=false -Dplugin_nvme=true -Dpkcs7=false -Dplugin_redfish=false -Dplugin_synaptics=false -Dsystemd=true -Dtests=false -Dplugin_thunderbolt=true -Dplugin_uefi=true -Dplugin_flashrom=false -Dplugin_modem_manager=false /var/tmp/portage/sys-apps/fwupd-1.3.3/work/fwupd-1.3.3 /var/tmp/portage/sys-apps/fwupd-1.3.3/work/fwupd-1.3.3-build
The Meson build system
Version: 0.52.0
Source dir: /var/tmp/portage/sys-apps/fwupd-1.3.3/work/fwupd-1.3.3
Build dir: /var/tmp/portage/sys-apps/fwupd-1.3.3/work/fwupd-1.3.3-build
Build type: native build
Project name: fwupd
Project version: 1.3.3
Appending CFLAGS from environment: '-march=native -O2 -flto -fno-stack-protector -pipe'
Appending LDFLAGS from environment: ['-Wl,-O1', '-Wl,--as-needed']
Appending CFLAGS from environment: '-march=native -O2 -flto -fno-stack-protector -pipe'
Appending LDFLAGS from environment: ['-Wl,-O1', '-Wl,--as-needed']
C compiler for the host machine: x86_64-pc-linux-gnu-gcc (gcc 9.2.0 "x86_64-pc-linux-gnu-gcc (Gentoo 9.2.0 p1) 9.2.0")
C linker for the host machine: GNU ld.gold 2.32
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program git found: YES (/usr/bin/git)
Compiler for C supports arguments -fstack-protector-strong: YES 
Compiler for C supports arguments -Waggregate-return: YES 
Compiler for C supports arguments -Wunused: YES 
Compiler for C supports arguments -Warray-bounds: YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Wclobbered: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wduplicated-branches: YES 
Compiler for C supports arguments -Wduplicated-cond: YES 
Compiler for C supports arguments -Wempty-body: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Wformat-signedness: YES 
Compiler for C supports arguments -Wignored-qualifiers: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Wincompatible-pointer-types-discards-qualifiers: NO 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wlogical-op: YES 
Compiler for C supports arguments -Wmissing-declarations: YES 
Compiler for C supports arguments -Wmissing-format-attribute: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-noreturn: YES 
Compiler for C supports arguments -Wmissing-parameter-type: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wno-cast-function-type -Wcast-function-type: YES 
Compiler for C supports arguments -Wno-address-of-packed-member -Waddress-of-packed-member: YES 
Compiler for C supports arguments -Wno-unknown-pragmas -Wunknown-pragmas: YES 
Compiler for C supports arguments -Wno-deprecated-declarations -Wdeprecated-declarations: YES 
Compiler for C supports arguments -Wno-discarded-qualifiers -Wdiscarded-qualifiers: YES 
Compiler for C supports arguments -Wno-missing-field-initializers -Wmissing-field-initializers: YES 
Compiler for C supports arguments -Wno-strict-aliasing -Wstrict-aliasing: YES 
Compiler for C supports arguments -Wno-suggest-attribute=format -Wsuggest-attribute=format: YES 
Compiler for C supports arguments -Wno-unused-parameter -Wunused-parameter: YES 
Compiler for C supports arguments -Wnull-dereference: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Woverride-init: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wredundant-decls: YES 
Compiler for C supports arguments -Wreturn-type: YES 
Compiler for C supports arguments -Wshadow: YES 
Compiler for C supports arguments -Wsign-compare: YES 
Compiler for C supports arguments -Wstrict-aliasing: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wswitch-default: YES 
Compiler for C supports arguments -Wtype-limits: YES 
Compiler for C supports arguments -Wundef: YES 
Compiler for C supports arguments -Wuninitialized: YES 
Compiler for C supports arguments -Wunused-but-set-variable: YES 
Compiler for C supports arguments -Wunused-variable: YES 
Compiler for C supports arguments -Wvla: YES 
[... 120 lines cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: warning: wildcard match appears in both version 'LIBFWUPD_1.0.0' and 'LIBFWUPD_1.0.3' in script
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: warning: wildcard match appears in both version 'LIBFWUPD_1.0.3' and 'LIBFWUPD_1.0.4' in script
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: warning: wildcard match appears in both version 'LIBFWUPD_1.0.4' and 'LIBFWUPD_1.0.7' in script
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: warning: wildcard match appears in both version 'LIBFWUPD_1.0.7' and 'LIBFWUPD_1.0.8' in script
[... cut ...] able -Wvla -Wwrite-strings -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 -march=native -O2 -flto -fno-stack-protector -fPIC -pthread '-DFU_OFFLINE_DESTDIR=""' -MD -MQ 'src/25a6634@@fwupdprivate@sta/fu-srec-firmware.c.o' -MF 'src/25a6634@@fwupdprivate@sta/fu-srec-firmware.c.o.d' -o 'src/25a6634@@fwupdprivate@sta/fu-srec-firmware.c.o' -c ../fwupd-1.3.3/src/fu-srec-firmware.c
.........................
```

I think it could have something with these linking WARNINGs.

hughsie, 2019-11-05:
We do zero testing with Gentoo, for fairly obvious reasons... I think you will have to debug this yourself I'm afraid.

okias, 2019-11-05:
I understand, but generally have working code with binutils gold linker is nice thing to have, since gold is replacement for classic bfd. I think some distributions use it, but when compilation fails, they fallback to BFD.

superm1, 2019-11-05:
Are you sure it was told linker and not lto? We have had problems with lto reported in past.

https://github.com/fwupd/fwupd/issues/279

superm1, 2019-11-05:
Oh actually in 1.3.3 I recall we added a `reinstall` command.  So just run
`# fwupdmgr reinstall` and it will reinstall the same firmware again.

superm1, 2019-11-06:
@okias FYI, if it is indeed the linker and not LTO you can override the linker only for the EFI application by using the meson option `efi-ld` rather than for the whole project.

zhuyifei1999, 2019-11-13:
This is likely the exact same problem that I have debugged for systemd-boot a while ago: systemd/systemd#11541

The GOLD linker generation different section names than BFD, and some of the sections are critical for relocation; during the final EFI PE binary generation, objcopy does not copy some critical sections if they are named differently:
https://github.com/fwupd/fwupd/blob/e7b2ea02b96fe3b60aa41453ec37c2d01ea6bc4f/plugins/uefi/efi/generate_binary.sh#L6-L14

Shall I submit a patch?

superm1, 2019-11-13:
Sure, please do.

okias, 2019-11-13:
```
$ gdb fwupdx64.efi -ex 'info files' -batch
Symbols from "/usr/libexec/fwupd/efi/fwupdx64.efi".
Local exec file:
	`/usr/libexec/fwupd/efi/fwupdx64.efi', file type pei-x86-64.
	Entry point: 0x4000
	0x0000000000004000 - 0x000000000000bad0 is .text
	0x000000000000d000 - 0x000000000000d00a is .reloc
	0x000000000000e000 - 0x0000000000010e48 is .data
	0x0000000000011000 - 0x0000000000011140 is .dynamic
	0x0000000000012000 - 0x0000000000012270 is .dynsym
```

```
$ gdb fwupdx64.efi_ubnt -ex 'info files' -batch 
Symbols from "/usr/libexec/fwupd/efi/fwupdx64.efi_ubnt".
Local exec file:
	`/usr/libexec/fwupd/efi/fwupdx64.efi_ubnt', file type pei-x86-64.
	Entry point: 0x4000
	0x0000000000004000 - 0x000000000000ba70 is .text
	0x000000000000c000 - 0x000000000000c00a is .reloc
	0x000000000000d000 - 0x000000000000fea8 is .data
	0x0000000000010000 - 0x0000000000010150 is .dynamic
	0x0000000000011000 - 0x0000000000011e70 is .rela
	0x0000000000012000 - 0x0000000000012270 is .dynsym
```
@zhuyifei1999 very good catch! thank you! :)

PR #1560 (2019-11-14): efi: use a wildcard section copy for final EFI generation
Fixes #1530

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation

I have not been able to reproduce the hang / crash in QEMU without a .cap. @okias Would you mind testing this?
commit 8b044c6c (2019-11-14): efi: use a wildcard section copy for final EFI generation

The GNU gold linker uses the section name `.rela.dyn` instead of
`.rela` for containing the relocation information. If this section
is not copied the EFI executable can crash.

Fixes #1530
commit 1f579d82 (2020-01-10): efi: use a wildcard section copy for final EFI generation

The GNU gold linker uses the section name `.rela.dyn` instead of
`.rela` for containing the relocation information. If this section
is not copied the EFI executable can crash.

Fixes #1530

Cherry-picked from 8b044c6cb3d007803bd67ca5e6527ed89d5ecb62
commit d9bd4234 (2021-02-19): efi: use a wildcard section copy for final EFI generation

The GNU gold linker uses the section name `.rela.dyn` instead of
`.rela` for containing the relocation information. If this section
is not copied the EFI executable can crash.

Fixes #1530
closed 2019-11-14 by hughsie