fwupd/fwupd #1532: ModemManager plugin broken in 1.3.3

aleksander0m, 2019-11-04:
Since 1.3.3 the DW5821e upgrade doing the "fastboot+QMI/PDC" upgrade procedure is no longer working, and it looks like the culprit is 4b30380e78f864d722472bdebca010a60bb5b232.

In that commit, the fastboot `fu_plugin_update_attach()` method is removed in favor of a new generic `fu_plugin_device_attach()`  logic. Unfortunately, the new method checks for the `FWUPD_DEVICE_FLAG_IS_BOOTLOADER` flag in the device, and if it doesn't have the flag, the `fu_device_attach()` operation is completely ignored. This seems to be inline with the colorhug plugin, but there was no check for that flag in the ModemManager plugin.

```
16:25:48:0436 FuPlugin             running superclassed update_attach() on fastboot
16:25:48:0436 FuPlugin             already in runtime mode, skipping
```

In the DW5821e upgrade procedure, we require that attach operation because we have 2 cycles in the engine: first with the fastboot plugin, and then with the ModemManager plugin running the QMI-PDC logic (as per `FWUPD_DEVICE_FLAG_ANOTHER_WRITE_REQUIRED`). Without that attach happening, the modem doesn't reset from fastboot mode, and it never reboots in "modem mode" to be able to run the QMI-PDC bits, so we end up re-running the fastboot upgrade procedure in the 2nd cycle as well.

How can we avoid this?

hughsie, 2019-11-04:
Hmm, sorry about that. I guess maybe we move the IS_BOOTLOADER check back into the plugins?

aleksander0m, 2019-11-04:
What about the IS_BOOTLOADER check in `update_detach()`? The old generic code in `fu_plugin_update_detach()` had the check already, while the old generic code in `fu_plugin_update_attach()` didn't have it. If the IS_BOOTLOADER flag is something plugin-specific only, then maybe we shouldn't have it in either of them?

aleksander0m, 2019-11-05:
@hughsie want me to draft a patch to solve it, or will you get to do it?

hughsie, 2019-11-05:
> then maybe we shouldn't have it in either of them

I think it does make sense in most cases to check the bootloader flag if using attach or detach -- given the definitions of those functions are `Detach to bootloader mode` and `Attach to firmware mode`. Would it be a good idea to set `IS_BOOTLOADER` unconditionally when in fastboot mode? That seems logically what it is; a degraded mode that's able to update firmware without runtime functionality. I think that might fix this issue too.

aleksander0m, 2019-11-05:
That makes sense, yes.

closed 2019-11-05 by superm1