fwupd/fwupd #1534: lost thunderbolt controller after upgrade

Macbet, 2019-11-05:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**20KH006MRT
│
├─SAMSUNG MZVLB1T0HALR-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     5L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          08575ca8-c22c-540a-8ac9-e784c04f6e5a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.41
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.65.3590
│     Minimum Version:     0.0.1
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0003,0034,002E,002F,0030,0031,0032,0033,0027,0028,002D,0004
Boot0000* archlinux	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\archlinux\grubx64.efi)
Boot0003* EndeavourOS	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\EndeavourOS\grubx64.efi)
Boot0004* Linux-Firmware-Updater	HD(1,GPT,99b65f67-7006-c147-bcf4-b28a8f040837,0x1000,0x96000)/File(\EFI\endeavouros\fwupdx64.efi)
Boot000A  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot000B  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot000C  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot000D  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot000E  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot000F  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0025  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0026  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0027  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0028  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0029* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot002A* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot002B* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot002C* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot002D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot002E* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot002F* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0030* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot0031* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot0032* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0033* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0034* Linux Boot Manager	HD(1,GPT,5133c104-112f-b443-b52f-5f58071c44bb,0x800,0x1dd000)/File(\EFI\systemd\systemd-bootx64.efi)
**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman 
- Have you tried rebooting? - yes 
- Are you using an NVMe disk? - yes
- Is secure boot enabled (only for the UEFI plugin)?

yesterday came the update for thunderbolt controller, before installing it turned off Thunderbolt BIOS Assist Mode as indicated in the description, after the update was gone thunderbolt controller from the list of devices, tell me it immediately contact lenovo support or have a chance to return the controller to work without them ?

fwupdmgr get-history - 
├─Thunderbolt Controller:
│ │   Device ID:           9de9f2423b63f31e5a968d3781ecffb4b4ba6bca
│ │   Previous version:    43.00
│ │   Update State:        failed
│ │   Update Error:        device version not updated on success, 43.0.0 != 43.00
│ │   GUID:                2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Reported to remote server
│ │                        • Device stages updates
│ │ 
│ └─  New version:       43.0.0
│       Remote ID:         lvfs
│       License:           Unknown

hughsie, 2019-11-05:
>  │ │ Update Error: device version not updated on success, 43.0.0 != 43.00

Well, that's one failure, but the bigger is the thunderbolt controller that is probably stuck in recovery mode. What does `sudo lspci -vvv` say?

Macbet, 2019-11-05:
> sudo lspci -vvv

00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
	Subsystem: Lenovo Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ >SERR- <PERR- INTx-
	Latency: 0
	Capabilities: [e0] Vendor Specific Information: Len=10 <?>
	Kernel driver in use: skl_uncore

00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07) (prog-if 00 [VGA controller])
	Subsystem: Lenovo UHD Graphics 620
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 134
	Region 0: Memory at 2ffa000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at 2fa0000000 (64-bit, prefetchable) [size=256M]
	Region 4: I/O ports at e000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR-, OBFF Not Supported
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
			 AtomicOpsCtl: ReqEn-
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
		Address: fee00018  Data: 0000
	Capabilities: [d0] Power Management version 2
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Process Address Space ID (PASID)
		PASIDCap: Exec- Priv-, Max PASID Width: 14
		PASIDCtl: Enable- Exec- Priv-
	Capabilities: [200 v1] Address Translation Service (ATS)
		ATSCap:	Invalidate Queue Depth: 00
		ATSCtl:	Enable-, Smallest Translation Unit: 00
	Capabilities: [300 v1] Page Request Interface (PRI)
		PRICtl: Enable- Reset-
		PRISta: RF- UPRGI- Stopped+
		Page Request Capacity: 00008000, Page Request Allocation: 00000000
	Kernel driver in use: i915
	Kernel modules: i915

00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
	Subsystem: Lenovo Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 16
	Region 0: Memory at 2ffb020000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
		Address: 00000000  Data: 0000
	Capabilities: [d0] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [e0] Vendor Specific Information: Len=0c <?>
	Kernel driver in use: proc_thermal
	Kernel modules: processor_thermal_device

00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
	Subsystem: Lenovo Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
[... 418 lines cut ...]
	Capabilities: [b0] MSI-X: Enable+ Count=33 Masked-
		Vector table: BAR=0 offset=00003000
		PBA: BAR=0 offset=00002000
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap+ MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [158 v1] Power Budgeting <?>
	Capabilities: [168 v1] Secondary PCI Express <?>
	Capabilities: [188 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [190 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
			   T_CommonMode=0us LTR1.2_Threshold=163840ns
		L1SubCtl2: T_PwrOn=44us
	Kernel driver in use: nvme

superm1, 2019-11-05:
> > │ │ Update Error: device version not updated on success, 43.0.0 != 43.00

Isn't this a **metadata** problem? Thunderbolt versions are not triplets, they're pairs.  So the version string in the metadata is listing 3 groups of data and the comparison fails.

superm1, 2019-11-05:
As in this update shouldn't have even been pushed down to this machine - 43.0.0 > 43.00.  I think you should probably pull this update back to testing on LVFS.

hughsie, 2019-11-05:
> 43.0.0 > 43.00

I've unpushed the update and have sent email to the relevant people. I've asked them to help on this ticket to recover the hardware.

Macbet, 2019-11-05:
>I've unpushed the update and have sent email to the relevant people. I've asked them to help on this ticket to recover the hardware.

thanks !

superm1, 2019-11-05:
FYI, duplicate here: https://github.com/fwupd/missing-firmware-lenovo-thinkpad/issues/24
@hughsie I noticed a comment about not force upgrading.  Do you know more about what's going on here?  Why is the the same capsule even matching both native mode and BIOS assist mode?  If it requires native mode it shouldn't have GUIDs for BIOS assist mode.

hughsie, 2019-11-05:
> Why is the the same capsule even matching both native mode and BIOS assist mode

I have no idea. I've asked Lenovo and will escalate.

superm1, 2019-11-05:
This looks totally busted to me for Lenovo Thunderbolt controllers.
I looked at https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN2HTF.firmware (which I realize isn't necessarily the same as this reported issue).

It has two GUIDs listed:
```
-<provides>
<firmware type="flashed">c7920601-0cda-507f-851e-92229eb1d370</firmware>
<firmware type="flashed">0f11f23b-ccfc-5bac-9a41-405429734903</firmware>
</provides>
```

I can't tell what is used to build those GUIDs but they're either both the native and non-native GUIDs or it's two different systems using the exact same controller (_very very unlikely_).
If they only support the native GUID, only include the native GUID.  You're going to potentially brick controllers left and right if you allow people to flash on the non-native GUID.

sydneymeyer, 2019-11-06:
FWIW, i also had similar issues with this thunderbolt fw update from 43.00 to 43.0.0 on a thinkpad X1C6.

I had "Thunderbolt BIOS Assist Mode" already disabled long before the update, the update itself appeared to install just fine, but fwupd offered this same update over again (until it has been pulled). I have tried to install this update two times and have the following in "fwupdmgr get-history":

├─X1 Carbon Thunderbolt Controller:
│ │   Device ID:           93f29bac0d995f92032d0f0eb4bde946c3bcfe9f
│ │   Previous version:    33.00
│ │   Update State:        failed
│ │   Update Error:        device version not updated on success, 43.0.0 != 43.00
│ │   Flags:               internal|updatable|require-ac|reported|historical
│ │   GUID:                2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│ │ 
│ └─  New version:       43.0.0
│     
├─X1 Carbon Thunderbolt Controller:
│ │   Device ID:           93f29bac0d995f92032d0f0eb4bde946c3bcfe9f
│ │   Previous version:    43.00
│ │   Update State:        failed
│ │   Update Error:        device version not updated on success, 43.0.0 != 43.00
│ │   Flags:               internal|updatable|require-ac|reported|historical
│ │   GUID:                2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│ │ 
│ └─  New version:       43.0.0

Opposed to OP, i can still see the controller via lspci.

alexmurray, 2019-11-06:
fwiw I also had the same behaviour as @sydneymeyer on my X1C6 - I disabled BIOS Assist mode, rebooted, installed this update and then due to the version mismatch (43.00 vs 43.0.0) was still offered the upgrade again until this got pulled from lvfs - so I also tried installing it a second time but due to the version mismatch in the metadata fwupd would still offer the update again).

Unlike the OP, my thunderbolt controller still appears to be working as well. 

```
$ fwupdmgr get-devices 
X1 Carbon Thunderbolt Controller
  DeviceId:             6aed4d3ef9ec3f0b1b0cbe162d4ba2d063a9be3f
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              43.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-11-06
```

```
$ lspci | grep Thunderbolt
05:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
06:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
06:01.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
06:02.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
06:04.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
07:00.0 System peripheral: Intel Corporation JHL6540 Thunderbolt 3 NHI (C step) [Alpine Ridge 4C 2016] (rev 02)
3b:00.0 USB controller: Intel Corporation JHL6540 Thunderbolt 3 USB Controller (C step) [Alpine Ridge 4C 2016] (rev 02)
```

```
$ fwupmgr get-history
X1 Carbon Thunderbolt Controller
  DeviceId:             6aed4d3ef9ec3f0b1b0cbe162d4ba2d063a9be3f
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|reported
  Version:              30.00
  Created:              2019-11-01
  UpdateState:          failed
  UpdateError:          device version not updated on success, 43.0.0 != 43.00
  
  [Release]
  Version:              43.0.0
  Checksum:             SHA1(1e75dfb3e3b2b69c48c601101a4552033b4a775f)
  Flags:                none
  CompileVersion(org.freedesktop.gusb): 0.3.0
  RuntimeVersion(com.redhat.fwupdate): 12
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.10
  DistroVersion:        19.10
  DistroId:             ubuntu
  ThunderboltNative:    True
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  CompileVersion(org.freedesktop.fwupd): 1.2.10
  BootTime:             1572608623
  RuntimeVersion(com.dell.libsmbios): 2.4
  CompileVersion(com.redhat.efivar): 37
  KernelVersion:        5.3.0-19-generic
  CompileVersion(com.redhat.fwupdate): 12
...
X1 Carbon Thunderbolt Controller
  DeviceId:             6aed4d3ef9ec3f0b1b0cbe162d4ba2d063a9be3f
  Guid:                 2da42a33-cd30-5ef5-a8fb-2c800a4b760f
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|reported
  Version:              43.00
  Created:              2019-11-02
  Modified:             2019-11-02
  UpdateState:          failed
  UpdateError:          device version not updated on success, 43.0.0 != 43.00
  
  [Release]
  Version:              43.0.0
  Checksum:             SHA1(1e75dfb3e3b2b69c48c601101a4552033b4a775f)
  Flags:                none
  CompileVersion(org.freedesktop.gusb): 0.3.0
  RuntimeVersion(com.redhat.fwupdate): 12
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.10
  DistroVersion:        19.10
  DistroId:             ubuntu
  ThunderboltNative:    True
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  CompileVersion(org.freedesktop.fwupd): 1.2.10
  BootTime:             1572693605
  RuntimeVersion(com.dell.libsmbios): 2.4
  CompileVersion(com.redhat.efivar): 37
  KernelVersion:        5.3.0-19-generic
  CompileVersion(com.redhat.fwupdate): 12
```

```
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```