fwupd/fwupd #1538: Unable to update Lenovo T480s System Firmware on Fedora 31 - UpdateState: needs-reboot

maaattes, 2019-11-06:
I'm trying to update the System Firmware of my Lenovo T480s to 0.1.36. After `fwupdmgr get-updates` and `fwupdmgr update` machine reboots as expected. But the update could not be installed. After the Lenovo Splash Screen it says for a fraction of a second:
```
System BootOrder not found. Initializing defaults.
Creating boot entry "Boot0000" with label "Fedora" for file "\EFI\fedora\shimx64.efi"

Reset System
```
`fwupdmgr get-history` gives `UpdateState: needs-reboot` for System Firmware and UEFI Device Firmware

```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             24dd1466bf80723529c067bcfd0d35ab5bea2f38
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-11-06

20L8S02D00 System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(e6ea350dce7bebe362ca012efe094ed656431366)
  Checksum:             SHA256(99062fe4ad3ccb0643773dedda2bebe6ad5876772ddfa6974de8a4f251548897)
  Vendor:               LENOVO
  Version:              0.1.34
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-11-06
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-11-06

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.17
  VersionLowest:        0.1.17
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-11-06
  UpdateState:          success
[... 224 lines cut ...]
  CompileVersion(org.freedesktop.fwupd): 1.2.11
  SecureBoot:           Disabled
  CompileVersion(com.redhat.efivar): 37
  UEFIUXCapsule:        Disabled
  ESPMountPoint:        /boot/efi
  MissingCapsuleHeader: False
  CompileVersion(org.freedesktop.gusb): 0.3.0
  RuntimeVersion(com.dell.libsmbios): 2.4
  DistroId:             fedora
  CpuArchitecture:      x86_64
  BootTime:             1573064688
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  RuntimeVersion(com.redhat.fwupdate): 12
  DistroVariant:        workstation
  RuntimeVersion(org.freedesktop.fwupd): 1.2.11
  KernelVersion:        5.3.8-300.fc31.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 31, Kernel 5.3.8-300
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf install
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-11-07:
Can you schedule another update please, and then paste the output of this please `sudo efibootmgr -v` before rebooting. Thanks.

maaattes, 2019-11-07:
> Can you schedule another update please, and then paste the output of this please `sudo efibootmgr -v` before rebooting. Thanks.

```shell
BootNext: 0000
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,0001
Boot0000* Linux-Firmware-Updater	HD(1,GPT,dab0dd62-dae1-4628-95c9-883b06cd54c9,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* Linux-Firmware-Updater	HD(1,GPT,dab0dd62-dae1-4628-95c9-883b06cd54c9,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

hughsie, 2019-11-07:
Okay, that's certainly unexpected. The system GUID of `ebfe8df8-dee7-4692-a721-cbcf5095c5cf` isn't quirked in `uefi.quirk` so I'm really surprised to see `Linux-Firmware-Updater` with the dashes. This has to be an upgraded install that uses the old name we used until a while ago, and we kept using that. Quite why we're overwriting Boot0000 I have no idea.

hughsie, 2019-11-07:
I guess one thing it could be is if the existing `efi_loadopt` is invalid, and we never re-use the old entry. The code in https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-bootmgr.c#L182 is kinda scary, I don't really understand what that is exactly doing.

maaattes, 2019-11-07:
Okay, this is way beyond my knowledge. I don't know if this is relevant, but I've done a clean install of Fedora 31 and I'm using full disk encryption.
My first thought was that it had something to do with the changes made to [grub2](https://fedoraproject.org/wiki/Changes/Include_security_modules_in_efi_Grub2). But this was just a wild guess. I don't know anything about such things.

superm1, 2019-11-07:
@hughsie 
All Lenovo hardware is using that:
https://github.com/fwupd/fwupd/blob/51a869a01da445cd77dd9baddcf05d3bf0b4debf/plugins/uefi/fu-plugin-uefi.c#L536

superm1, 2019-11-07:
And also in above, notice:
```
  BootMgrDesc:          legacy
```

maaattes, 2019-11-07:
To my suprise, I just managed to update via gnome-software.
I've done a 'fwupdmgr report-history'. Let me know if you want me to provide any more logs.

syscll, 2019-11-21:
I am also experiencing this issue on the same hardware. Didn't want to open a new ticket, so here are some details. Hope this helps:

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
```

```shell
$ fwupdmgr get-devices
20L7CTO1WW
│
├─Samsung SSD 960 EVO 250GB:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B7QCXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
│                          e5b293c4-ce87-526e-b467-75fdec86360c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.34
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Update State:        success
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.65.3590
│     Minimum Version:     184.65.3590
│     Update State:        success
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     Update State:        success
│     GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─VMM0000:
      Device ID:           7c2dd454e6f29453896a4824e8234317d237f8c5
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.000
      Vendor:              Synaptics
      GUIDs:               42addef4-40f9-5e89-b925-d564e35ed368
                           cf8c03c5-18bf-53c4-971f-4a08f88932b5
                           e9427b6a-7389-5461-a592-1da5f8ec99fd
      Device Flags:        • Updatable
```

```shell
$ efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0001,0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater
Could not parse device path: Invalid argument
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-4d254d6e-cd67-477b-97d5-bc3048af45c4-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-676af093-2a5c-4238-9c29-db8063a33532-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0

```

```shell
$ tree /boot
/boot
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   ├── fwupd-4d254d6e-cd67-477b-97d5-bc3048af45c4.cap
│   │   │   ├── fwupd-676af093-2a5c-4238-9c29-db8063a33532.cap
│   │   │   └── fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf.cap
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   └── loader.conf
└── vmlinuz-linux

7 directories, 13 files
```

Please answer the following questions:

- Operating system and version: `Arch Linux 5.3.11-arch1-1`
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman`
- Have you tried rebooting? Yes, several times.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-11-21:
I'd say please test with master, this might have just been fixed (see that linked bug for details)

syscll, 2019-11-25:
I'm not running 1.3.4 and after following the steps in #1572 I still can't seem to update the system firmware.

`Linux-Firmware-Updater` is always assigned to `Boot0000`. I've manually deleted that entry, restarted, and ran `fwupdmgr get-updates`, and it's still assigned to `Boot000`:

```
$ sudo efibootmgr
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0001,0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater
Boot0001* Windows Boot Manager
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* USBR BOOT CDROM
Boot0020* USBR BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
Boot0023* PCI LAN
```

Is there something else I need to do?

Ingvin, 2019-11-27:
I tried 1.3.3 and 1.3.4 versions and have the same issue with Lenovo T470

```$ sudo efibootmgr
BootNext: 0000
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0017,0018,0019,001A,001B,001C,001D,001E,001F,0000
Boot0000* Linux-Firmware-Updater
Boot0002* Fedora
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD1
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* IDER BOOT CDROM
Boot0021* IDER BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
Boot0024* PCI LAN
```

syscll, 2019-11-30:
I just tried with 1.3.5 on my Lenovo ThinkPad T480s and was able to successfully update all firmware.

I removed any old `Linux-Firmware-Updater` boot entries using:
```
$ efibootmgr -B -b XXXX
```

Then proceeded to follow the standard `fwupdmgr` update path without any issues. Machine restarted ok and applied firmware as expected.

Ingvin, 2019-12-02:
I tried 1.3.5 and it still doesn't works on T470

```$ sudo efibootmgr 
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0017,0018,0019,001A,001B,001C,001D,001E,001F
Boot0002* Fedora
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD1
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* IDER BOOT CDROM
Boot0021* IDER BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
Boot0024* PCI LAN
```

```$ sudo fwupdmgr update 
No upgrades for System Firmware, current is 0.1.60: 0.1.60=same, 0.1.59=older, 0.1.58=older, 0.1.57=older, 0.1.56=older
Upgrade available for UEFI Device Firmware from 184.65.3590 to 184.70.3626
20HD0001RT must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 0.1.34: 0.1.34=same

An update requires a reboot to complete. Restart now? [Y|n]: n
```

```$ sudo efibootmgr -v
BootNext: 0000
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0017,0018,0019,001A,001B,001C,001D,001E,001F,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,9e3f5ec2-9cd9-4a90-8989-0dece08b646c,0x800,0x100000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0002* Fedora	HD(1,GPT,9e3f5ec2-9cd9-4a90-8989-0dece08b646c,0x800,0x100000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
[... 81 lines cut ...]
BootOrder: 0002,0017,0018,0019,001A,001B,001C,001D,001E,001F,0000
Boot0000* Linux-Firmware-Updater
Boot0002* Fedora
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD1
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* IDER BOOT CDROM
Boot0021* IDER BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
Boot0024* PCI LAN
```

Ingvin, 2019-12-05:
My colleague tried to upgrade from version 184.60.3561 to 184.70.3626 on T480 and it worked

Perhaps the problem is that the firmware itself has changed, in version 184.65.3590 there is a new splash screen during installation

closed 2020-01-11 by stale[bot]