fwupd/fwupd #1539: Clevis luks integration on UEFI fw updates?

superm1, 2019-11-06:
[Clevis](https://github.com/latchset/clevis) has support for binding LUKS encryption keys to TPM2 PCR register value(s).  The key gets stored in the metadata and the decryption credential is pulled out of TPM as needed.

The flow works by doing something like this:
```
clevis luks bind -f -d /dev/sdb1 tpm2 '{"pcr_bank":"sha1", "pcr_ids":"0"}'
clevis luks unlock -d /dev/sdb1 -n c1
```

This is great and I hope that it becomes prevalent, but of course if you bind to PCR0 you might accidentally lock yourself out of your disk from a firmware update coming down that causes PCR0 to change.

Even if we knew "all" the golden PCR0 values for a given system and enrolled those we can't possibly predict which one we're about to flash, especially since the firmware update could fail.

So I think we need some integration work here so that at least for one boot the system can boot into the new firmware and then refresh binding values.

I'm not entirely sure which aspects should belong to fwupd and which to clevis, but here's the parts I think are important that I wanted to open up for some discussion:

* A way for fwupd to detect that clevis has bound LUKS to the TPM PCRs.  I don't know if scanning disks and looking for `LUKSMeta` is enough or if clevis can help here.
* A way for fwupd to suspend LUKS for a single boot (Maybe the action suspend is actually enroll RNG secret that's written somewhere that is enabled only for one boot?) when doing a UEFI FW update
* A way for fwupd to request clevis to rebind the PCR values selected the first time after a successful UEFI FW update and remove that RNG secret)

cc @hughsie, @therealjuanmartinez, @diabonas, @sergio-correia feel free to loop some others for more ideas or discussion

hughsie, 2019-11-07:
Does clevis support us changing the PCR0 that it's expecting, or allow us to add another PCR0? It seems to me that we just want to prominently disable system firmware updates on any system using PCR0 to unlock any volume...

superm1, 2019-11-20:
I think this is a case that we want to recommend use of authorized policies when this is solved / merged for clevis (https://github.com/latchset/clevis/issues/121).

So basically fwupd's public key can be trusted into the TPM and when FW updates happen can sign a policy with private key that will allow the update to pass.

superm1, 2020-10-08:
So I did want to note that the tpm2-software group did put out a blog entry that explained how to use authorized policies.
https://tpm2-software.github.io/2020/04/13/Disk-Encryption.html

hughsie, 2020-10-09:
> Instead of using a rigid PCR policy tied to raw PCR values we now seal it to a PCR signature

I wasn't entirely sure what the difference was. Also, I don't think fwupd wants to be in the business of resealing/mounting disks -- I think fwupd should just tell *something* that the PCR0 is might change on the next boot.

superm1, 2020-10-09:
Yes exactly, I agree. It calling into an API of another piece of software like clevis is what I thought.

superm1, 2020-10-09:
(And also - firmware updates won't necessarily just change PCR0, others can change too and be part of policy)

hughsie, 2020-10-09:
> It calling into an API of another piece of software like clevis is what I thought.

Did you have any ideas on what for the "message" would have? e.g. executing everything in /etc/fwupd/update.d so that clevis could install something there? Or a dbus signal fwupd would emit and clevis would be listening for?

superm1, 2020-10-10:
Clevis doesn't have a daemon, so I don't think calling into anything makes sense right now.

What I'm thinking is offer a way to enroll policies that can be signed by a fwupd daemon key. And then let the new policies also get signed by the same thing on update. Specifics of it requires discussion with clevis team on what they want to offer.

hughsie, 2020-10-10:
> specifics of it requires discussion with clevis team on what they want to offer

Okay, if you can find the right person feel free to set up a chat and we can hash something out.

closed 2020-12-08 by superm1