fwupd/fwupd #1546: EC no longer visible in 1.3.4 on Star Labs hardware

Sean-StarLabs, 2019-11-08:
1.3.4 no longer displays the EC as a supported device. Source of the install doesn't seem to make a difference

```
client version:	1.3.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.
```

```
LabTop
│
├─Star Drive PCIe SSD:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     ECFM12.3
│     Vendor:              Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:               8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                          a44eb54c-5441-56f2-8cc0-5e48964c6457
│                          12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs shutdown after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           5984fe62b2de5dfa0c7980660e230707d696c8dd
│     Current version:     1.1.0
│     Minimum Version:     1.1.0
│     Vendor:              Star Labs
│     Update State:        failed
│     Update Error:        failed to update to 100: incorrect version
│     GUID:                8265d473-a6c2-42b4-897b-bc220faa2d32
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─VMM0000:
      Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.000
      Vendor:              Synaptics
      GUIDs:               42addef4-40f9-5e89-b925-d564e35ed368
                           cf8c03c5-18bf-53c4-971f-4a08f88932b5
                           e9427b6a-7389-5461-a592-1da5f8ec99fd
      Device Flags:        • Updatable

```

superm1, 2019-11-08:
> 1.3.4 no longer displays the EC as a supported device. Source of the install doesn't seem to make a difference

What kernel version?  And do you have kernel lock down enabled?

>VMM0000:
      Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.000
      Vendor:              Synaptics
      GUIDs:               42addef4-40f9-5e89-b925-d564e35ed368
                           cf8c03c5-18bf-53c4-971f-4a08f88932b5
                           e9427b6a-7389-5461-a592-1da5f8ec99fd
      Device Flags:        • Updatable

This doesn't look right.  is this in a dock or your machine?  Could you run fwupdmgr as root so we can see what instance IDs get built?

Sean-StarLabs, 2019-11-08:
```
5.3.0-21-generic
```
Bog standard Ubuntu 19.10

VMM0000 is an LG 38UC99-W (USB-C Monitor). Doesn't make any difference if it's connected or not.

```
$ sudo fwupdmgr get-devices 
LabTop
│
├─Star Drive PCIe SSD:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     ECFM12.3
│     Vendor:              Phison Electronics Corporation (NVME:0x1987)
│     Serial Number:       DB0A0789176600023454
│     GUIDs:               8cb1e2fe-eb01-5508-9fb3-98add4bb7c34 ← NVME\VEN_1987&amp;DEV_5012&amp;REV_01
│                          a44eb54c-5441-56f2-8cc0-5e48964c6457 ← NVME\VEN_1987&amp;DEV_5012
│                          12d61493-5ac4-57f7-a2c5-96878d115455 ← Star Drive PCIe SSD
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs shutdown after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           5984fe62b2de5dfa0c7980660e230707d696c8dd
│     Current version:     1.1.0
│     Minimum Version:     1.1.0
│     Vendor:              Star Labs
│     Update State:        failed
│     Update Error:        failed to update to 100: incorrect version
│     GUID:                8265d473-a6c2-42b4-897b-bc220faa2d32
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─VMM0000:
      Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.000
      Vendor:              Synaptics
      GUIDs:               42addef4-40f9-5e89-b925-d564e35ed368 ← MST-(null)-vmm0000-0
                           cf8c03c5-18bf-53c4-971f-4a08f88932b5 ← MST-(null)-0
                           e9427b6a-7389-5461-a592-1da5f8ec99fd ← MST-(null)
      Device Flags:        • Updatable
```

superm1, 2019-11-08:
> Bog standard Ubuntu 19.10

OK so it's probably not that kernel lockdown stuff.  I would say run daemon in verbose and share the logs.

> VMM0000 is an LG 38UC99-W (USB-C Monitor). Doesn't make any difference if it's connected or not.

@hughsie that looks like something we should be filtering, or at least marking not updatable.

Sean-StarLabs, 2019-11-08:
How does one change the daemon to verbose?

superm1, 2019-11-08:
`fwupdmgr modify-config VerboseDomains "*"`
Then re-run the command and look at your system journal.

Sean-StarLabs, 2019-11-08:
Not sure what to look for, there are references to it:
```
not checking 03.07 as no version format set
```

Journalctl:

```
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0176 FuMain               Called GetRemotes()
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0174 XbSilo               3.43ms      query on / with `components/component/releases/release/checksum[@target='container'][text()='f84161801b9b888878ce258ba6f38f6ad6aa70cc']/../../../../custom/value[@key='fwupd::RemoteId']` limit=1 -> 1 results
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0170 XbSilo               2.62ms      query on / with `components/component/releases/release/checksum[@target='container'][text()='20753e52673a4dd45d1a208418aaba4d8bbeec66']/../../../../custom/value[@key='fwupd::RemoteId']` limit=1 -> 1 results
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0168 XbSilo               3.14ms      query on / with `components/component/releases/release/checksum[@target='container'][text()='968e18a5207206f49794985eac23f81b890aa6ba']/../../../../custom/value[@key='fwupd::RemoteId']` limit=1 -> 1 results
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0165 FuCommon             not checking 1.1.0 as no version format set
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0165 FuCommon             not checking 03.07 as no version format set
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0164 FuCommon             not checking ECFM12.1 as no version format set
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0164 FuMain               Called GetHistory()
Nov 08 15:32:11 sean-LabTop fwupd[25898]: 15:32:11:0161 FuMain               Called GetDevices()
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0167 FuMain               FuMain: acquired name: org.freedesktop.fwupd
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0166 FuMain               Daemon ready for requests
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0166 FuMain               Started with locale en_GB.UTF-8
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuMain               Emitting PropertyChanged('Status'='idle')
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuEngine             Emitting PropertyChanged('Status'='idle')
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuHistory            modifying device System Firmware [5984fe62b2de5dfa0c7980660e230707d696c8dd]
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuPlugin             performing get_results() on uefi
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuEngine             device ID 1f664ad149f2151fb750b52751dbf9ba045a7846 was not found
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuCommon             not checking 1.1.0 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuCommon             not checking 03.07 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuCommon             not checking ECFM12.1 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 FuPlugin             1e5f0643-058f-55dc-b396-dc80368412d9,831e28e5-c719-527c-9f76-71da41b34fe3 has no updates, so ignoring device
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0164 XbSilo               0.24ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='831e28e5-c719-527c-9f76-71da41b34fe3']` limit=1 -> 0 results
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0163 XbSilo               0.27ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='1e5f0643-058f-55dc-b396-dc80368412d9']` limit=1 -> 0 results
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0163 FuDevice             using 279ed287-3607-549e-bacc-f873bb9838c4 for HIDRAW\VEN_046D&DEV_C52B
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0161 FuEngine             3 devices with subsystem hidraw
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0154 FuCommon             not checking ECFM12.3 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0154 FuCommon             not checking ECFM12.3 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0153 FuCommon             not checking ECFM12.3 as no version format set
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0153 FuEngine             21 devices with subsystem block
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0128 FuHistory            get device
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0128 XbSilo               1.45ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='fed2efa4-7045-55a1-b4fc-b29283d59fe5']/../..|components/component/provides/firmware[@type='flashed'][text()='8b72a10c-1279-5f8e-a28a-34fb11a58240']/../..` limit=1 -> 0 results
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0126 FuDeviceList         ::added bbbf1ce3d1cf15550c3760b354592040292415bb
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0126 FuPlugin             performing fu_plugin_device_registered() on uefi
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0126 FuPlugin             emit added from optionrom: bbbf1ce3d1cf15550c3760b354592040292415bb
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0126 FuDevice             using bbbf1ce3d1cf15550c3760b354592040292415bb for PCI_SLOT_NAME=0000:00:02.0
Nov 08 15:31:57 sean-LabTop fwupd[25898]: 15:31:57:0119 FuEngine             18 devices with subsystem pci
[... cut ...]
Nov 08 15:31:26 sean-LabTop fwupd[23806]: 15:31:26:0128 Fwupd                looking for os-release at /etc/os-release
Nov 08 15:31:26 sean-LabTop polkitd(authority=local)[785]: Operator of unix-session:2 successfully authenticated as unix-user:sean to gain TEMPORARY authorization for action org.freedesktop.fwupd.modify-config for system-bus-name::1.945 [fwupdmgr modify-config VerboseDomains *] (owned by unix-user:sean)
Nov 08 15:11:47 sean-LabTop sudo[24842]:     sean : TTY=pts/0 ; PWD=/home/sean ; USER=root ; COMMAND=/usr/bin/fwupdmgr get-devices
```

hughsie, 2019-11-08:
> Nov 08 15:31:56 sean-LabTop fwupd[25898]: 15:31:56:0805 GLib-GObject         invalid cast from 'FuSuperioIt89Device' to 'FuUdevDevice'

Eeek, this is my fault. I'll look now.

closed 2019-11-08 by hughsie

Sean-StarLabs, 2019-11-15:
Just noticed that the "Star Drive ATA" no longer display as supported devices either. Related by any change?

(PCIe ones work fine)

superm1, 2019-11-15:
That same log above, it's supposed to have an ATA drive?

Sean-StarLabs, 2019-11-15:
> That same log above, it's supposed to have an ATA drive?

No, different laptop. Will post log from the affected one in 2 secs

superm1, 2019-11-15:
Open new issue for it with logs if you can, please.  And use master if possible.